espressif/esp-idf #5945: Documentation bug with CONFIG_SECURE_FLASH_ENCRYPTION_MODE

jeremyherbert, 2020-10-06:
In the latest documentation for CONFIG_SECURE_FLASH_ENCRYPTION_MODE, it says:

```
By default Development mode is enabled which allows UART bootloader to perform flash encryption operations

Select Release mode only for production or manufacturing. Once enabled you can not reflash using UART bootloader
```

see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-secure-flash-encryption-mode

However, I believe that the release mode description is not correct. As far as I understand, you **can** still reflash using the UART bootloader, but your binary needs to be pre-encrypted on the host. The current wording implies you can never write to the flash using the UART bootloader. Which one of these two is correct?

projectgus, 2020-10-11:
Hi @jeremyherbert ,

Thanks for pointing this out. You understand correctly - if you have the key, you can still pre-encrypt on the host and flash. And unencrypted data can still be flashed.

The only time it's not possible to read/write the flash using the UART bootloader is if the config option "Permantly disable ROM download mode" is selected. This is only available on ESP32 V3, and only enabled by default if Secure Boot V2 is also enabled.

Will update the docs to reflect this.

jeremyherbert, 2020-10-12:
Thanks. Sorry if this is off topic, but is there a reason that this will be disabled by default on ESP32 V3? I am yet to use one, but I do make heavy use of the encrypted reflashing of the ESP32 in my tooling. 

Someone could always just attach wires to the SPI interface, but I suppose you are hedging your bets against another glitching attack?

projectgus, 2020-10-12:
No problems, happy to discuss.

We're erring towards "defend in depth" by default. Most production deployments with Flash Encryption & Secure Boot V2 enabled aren't intended for serial updates. Although we don't know of any attack vectors that work against ESP32 V3 via the UART download interface, more or less every successful attack against earlier ESP32 relies on it so having it disabled by default reduces the attack surface.

The ESP32-S2 has a new "secure download" mode option which keeps the UART downloader enabled but limits the range of operations to strictly flash access (more or less). But ESP32 only has disabling the entire mode.

It's relatively easy to override the default and keep the download interface enabled in the project configuration, though. And we don't know of any issues keeping it enabled, but for anyone who is not planning to use it then it's better to disable it.

Angus

jeremyherbert, 2020-10-12:
Thanks. I understand the ESP32 S3, etc is somewhat on the way, would they likely be similar to the ESP32S2 as far as having the secure download mode?

projectgus, 2020-10-12:
Yes, we don't have any plans to remove Secure Download Mode from the ROM.

closed 2021-04-15 by espressif-bot