espressif/esp-idf #5946: Not Receiving Correct ESPNOW Payload Value

physiii, 2020-10-06:
Hello, I am using the current master branch version of [espnow](https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c) and trying to transmit and receive a payload. 

However when I print the value I am receiving on the second esp32 in the `example_espnow_data_parse` function with `printf("PAYLOAD: %u\n", buf->payload[0]);` I get a repeating value of 251 when `esp_fill_random` should put random values in `buf->payload[0]` instead of 251:

```
PAYLOAD: 251
I (21815) espnow_example: Receive 0th broadcast data from: 30:ae:a4:8d:ee:2c, len: 10
I (22815) espnow_example: send data to ff:ff:ff:ff:ff:ff
PAYLOAD: 251
I (22815) espnow_example: Receive 1th broadcast data from: 30:ae:a4:8d:ee:2c, len: 10
I (23815) espnow_example: send data to ff:ff:ff:ff:ff:ff
PAYLOAD: 251
I (23815) espnow_example: Receive 2th broadcast data from: 30:ae:a4:8d:ee:2c, len: 10
PAYLOAD: 64
I (23815) espnow_example: Receive 0th unicast data from: 30:ae:a4:8d:ee:2c, len: 10
PAYLOAD: 251
I (24735) espnow_example: Receive 1th unicast data from: 30:ae:a4:8d:ee:2c, len: 10
PAYLOAD: 251
I (25735) espnow_example: Receive 2th unicast data from: 30:ae:a4:8d:ee:2c, len: 10
```

If I add `buf->payload[0] = 42;` to `example_espnow_data_prepare` and comment out `esp_fill_random`, I still see the same results as above with just getting a value of 251.

I have used espnow in the past without problems but that was over a year ago so not sure what has changed?

physiii, 2020-10-06:
The two lines I added for testing are in example_espnow_data_parse and example_espnow_data_prepare functions. See [line 120](https://gist.github.com/physiii/7033e9ec707b7658bb82ae0be2bfeb8f#file-espnow_example_main-c-L120) and [line 143](https://gist.github.com/physiii/7033e9ec707b7658bb82ae0be2bfeb8f#file-espnow_example_main-c-L143).

```
/* Parse received ESPNOW data. */
int example_espnow_data_parse(uint8_t *data, uint16_t data_len, uint8_t *state, uint16_t *seq, int *magic)
{
    example_espnow_data_t *buf = (example_espnow_data_t *)data;
    uint16_t crc, crc_cal = 0;

    if (data_len < sizeof(example_espnow_data_t)) {
        ESP_LOGE(TAG, "Receive ESPNOW data too short, len:%d", data_len);
        return -1;
    }

    *state = buf->state;
    *seq = buf->seq_num;
    *magic = buf->magic;
    crc = buf->crc;
    buf->crc = 0;
    crc_cal = esp_crc16_le(UINT16_MAX, (uint8_t const *)buf, data_len);
    printf("PAYLOAD: %u\n", buf->payload[0]);

    if (crc_cal == crc) {
        return buf->type;
    }

    return -1;
}

/* Prepare ESPNOW data to be sent. */
void example_espnow_data_prepare(example_espnow_send_param_t *send_param)
{
    example_espnow_data_t *buf = (example_espnow_data_t *)send_param->buffer;

    assert(send_param->len >= sizeof(example_espnow_data_t));

    buf->type = IS_BROADCAST_ADDR(send_param->dest_mac) ? EXAMPLE_ESPNOW_DATA_BROADCAST : EXAMPLE_ESPNOW_DATA_UNICAST;
    buf->state = send_param->state;
    buf->seq_num = s_example_espnow_seq[buf->type]++;
    buf->crc = 0;
    buf->magic = send_param->magic;
    /* Fill all remaining bytes after the data with random values */
    // esp_fill_random(buf->payload, send_param->len - sizeof(example_espnow_data_t));
    buf->payload[0] = 42;
    buf->crc = esp_crc16_le(UINT16_MAX, (uint8_t const *)buf, send_param->len);
}
```

boarchuz, 2020-10-06:
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/wifi/espnow/main/espnow_example.h#L66

physiii, 2020-10-06:
> https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/wifi/espnow/main/espnow_example.h#L66

Yes that was the first thing I checked but nothing I try gets me the value I send. See where I added `printf("PAYLOAD: %u\n", buf->payload[0]);`?

boarchuz, 2020-10-06:
That's a zero length array, there is no buf->payload[0].

It looks like it's intended as a variable length array but you'll have to implement that yourself (or treat payload[0] as a placeholder and simply replace it with your payload size/struct).

For the sake of your test, try changing it to uint8_t payload[1];

physiii, 2020-10-07:
Changing to `unint8_t payload[1]` in the struct definition causes it to crash. 

I tried preloading a uint8_t array with values and then store that in `buf->payload[i]`, see below.

```
/* Prepare ESPNOW data to be sent. */
void example_espnow_data_prepare(example_espnow_send_param_t *send_param)
{
    example_espnow_data_t *buf = (example_espnow_data_t *)send_param->buffer;

    assert(send_param->len >= sizeof(example_espnow_data_t));

    buf->type = IS_BROADCAST_ADDR(send_param->dest_mac) ? EXAMPLE_ESPNOW_DATA_BROADCAST : EXAMPLE_ESPNOW_DATA_UNICAST;
    buf->state = send_param->state;
    buf->seq_num = s_example_espnow_seq[buf->type]++;
    buf->crc = 0;
    buf->magic = send_param->magic;
    /* Fill all remaining bytes after the data with random values */
    // esp_fill_random(buf->payload, send_param->len - sizeof(example_espnow_data_t));

    uint8_t len = 10;
    uint8_t payload[len];
    for (uint8_t i = 0; i < len; i++)
      payload[i] = i;

    for (int i = 0; i < len; i++)
      buf->payload[i] = payload[i];

    buf->crc = esp_crc16_le(UINT16_MAX, (uint8_t const *)buf, send_param->len);
}
```
You said simply replace it with your payload size/struct; can it be an array or does it have to be a struct? Can you show an quick example?

Thank you for the help!

boarchuz, 2020-10-07:
Sorry @physiii I missed this line when I had a look earlier:
https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/wifi/espnow/main/espnow_example_main.c#L333

You need to set CONFIG_ESPNOW_SEND_LEN in menuconfig to the size of example_espnow_data_t. The default value is 10 (the size of the rest of the struct members), so with your 1 byte payload you will need to set it to 11. Your above changes should work then.

physiii, 2020-10-07:
> Sorry @physiii I missed this line when I had a look earlier (so it is already implemented as a flxible array member):
> https://github.com/espressif/esp-idf/blob/8bc19ba893e5544d571a753d82b44a84799b94b1/examples/wifi/espnow/main/espnow_example_main.c#L333
> 
> You need to set CONFIG_ESPNOW_SEND_LEN in menuconfig to the size of example_espnow_data_t. The default value is 10 (the size of the rest of the struct members), so with your 1 byte payload you will need to set it to 11. Your above changes should work then.

I think that did it! I will test more tomorrow and report back.

I do not understand why they would have the Send len start at 10 bytes. If I want to send 3 bytes I need to set that to 13. Why not just say that is a Send len of 3?

closed 2020-10-07 by physiii

zhangyanjiaoesp, 2020-10-12:
@physiii 

Please refer to the information in the README.

[image]