espressif/esp-idf #5947: Control frame sending and receiving(RTS/CTS)

gvivekchary, 2020-10-06:
I am not able to send Control frames using "esp_wifi_80211_tx". in the link https://hackmag.com/security/esp32-sniffer/ , i read that esp can capture only management and data frames but not control frames. Is there any update to send and receive control frames? or still need to include? i am trying to forcefully send control frames raw data using  "esp_wifi_80211_tx", iam getting unsupported frame format error. Please help me if i am doing wrong.

Alvin1Zhang, 2020-10-12:
Thanks for raising this feature request.

espxiehang, 2020-10-16:
Hi @gvivekchary ,I think you need to use "esp_err_t esp_wifi_get_promiscuous_ctrl_filter(wifi_promiscuous_filter_t *filter);".

closed 2020-12-09 by Alvin1Zhang