espressif/esp-idf #5950: Bootloader not setting ESP_OTA_IMG_NEW state to ESP_OTA_IMG_PENDING_VERIFY

aliraza1944, 2020-10-07:
## Environment

- Development Kit: None
- Kit version (for WroverKit/PicoKit/DevKitC): None
- Module or chip used:  ESP32-WROVER-B (8 MB version)
- IDF version (run ``git describe --tags`` to find it): Two versions

1. v3.3.2 (no tags): This version is the one that we are using to compile app binaries. Bootloader for this version is: ESP-IDF v3.3.2 2nd stage bootloader.

2. v3.3-dirty: This version was used to flash devices when we sent the devices to field. Bootloader for this version is: ESP-IDF v3.3-dirty 2nd stage bootloader

- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    crosstool-ng-1.22.0-80-g6c4433a5
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?: No
- Power Supply: USB

## Problem Description

We have a few devices in production. When the devices in production were flashed they were flashed using esp-idf v3.3-dirty, lets call this firmware version A. The new OTA update(firmware version B) that we are trying to push uses esp-idf v3.3.2 during compilation.

When we push firmware version B through OTA update, it reports its ota_state as 0, which is ESP_OTA_IMG_NEW. CONFIG_APP_ENABLE_ROLLBACK is set in firmware version B. The pseudo code process_ota_rollback() is

```
void process_ota_rollback(){

    if ota_state is ESP_OTA_IMG_PENDING_VERIFY or ESP_OTA_IMG_UNDEFINED
       then run ethernet connectivity test and mark app as valid if ethernet connectivity test passes.
    else if ota_state is not ESP_OTA_IMG_VALID
       then mark app as invalid  (this includes states ESP_OTA_IMG_ABORTED, ESP_OTA_IMG_INVALID and ESP_OTA_IMG_NEW)
}
```

What we have found on our side is that if we flash the bootloader v3.3.2 and then send the update (fw version B) to device it performs normally.

If we flash firmware version A, compiled on v3.3-dirty alongwith bootloader v3.3-dirty and then send firmware version B (compiled using v3.3.2) then on first boot of version B, it reports its ota_state as ESP_OTA_IMG_NEW. As per the esp-idf documentation, the bootloader is responsible for marking ota_state from ESP_OTA_IMG_NEW to ESP_OTA_IMG_PENDING_VERIFY.

### Expected Behavior

After first boot, the OTA update should report its state as ESP_OTA_IMG_PENDING_VERIFY.

### Actual Behavior

After first boot, the OTA update reports its state as ESP_OTA_IMG_NEW. Our assumption is that bootloader v3.3-dirty is unable to change state of fw version B compiled on v3.3.2.

- [x] sdkconfig files for fw version A and B.
[sdkconfigs.zip](https://github.com/espressif/esp-idf/files/5339608/sdkconfigs.zip)

mahavirj, 2020-10-08:
@aliraza1944 

Looks like you do not have `CONFIG_APP_ROLLBACK_ENABLE` enabled in your `sdkconfig` for firmware version A. Is that intentional? 

Also can you please check enabling `debug` logs in bootloader, specifically around following code lines:

https://github.com/espressif/esp-idf/blob/fb18b12efedee0fed1d7674e356b1ec82c5af39d/components/bootloader_support/src/bootloader_utility.c#L359-L365

aliraza1944, 2020-10-08:
@mahavirj It was we unintentionally didn't set CONFIG_APP_ROLLBACK_ENABLE in firwmare A. It is set in firmware B.

regarding the logs, I am assuming that we should enable debug logs and check the logs in firmware version B, right?

negativekelvin, 2020-10-08:
If the code is not compiled in to your older bootloader to support rollback then you have to work around it like

```
void process_ota_rollback(){

    if ota_state is ESP_OTA_IMG_PENDING_VERIFY or ESP_OTA_IMG_UNDEFINED or (ESP_OTA_IMG_NEW and OLD_BOOTLOADER_VERSION)
       then run ethernet connectivity test and mark app as valid if ethernet connectivity test passes.
    else if ota_state is not ESP_OTA_IMG_VALID
       then mark app as invalid  (this includes states ESP_OTA_IMG_ABORTED, ESP_OTA_IMG_INVALID and ESP_OTA_IMG_NEW)
}
```

but if your new app crashes before it can mark itself invalid it will be stuck in a boot loop. So don't roll out firmware updates to devices with old bootloaders until they are in wide stable release.

closed 2020-10-10 by aliraza1944