espressif/esp-idf #5951: Cant able to put ESP32-DevKit_v4 device into Debug mode

Hullesha, 2020-10-07:
With eclipse Version: 2019-09 R (4.13.0) and esp-idf version 4.1, we are able build and flash the code (run mode) for ESP32-DevKitc_v4.
But now we are stuck with debug mode, followed the steps provided in below link for configuring GDB Hardware Debugging using eclipse. (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-using-debugger-eclipse) and we are able to configure the steps in eclipse.

But after we click the debug button we are getting below message from debugger window.

Error in final launch sequence:

Failed to execute MI command:
-target-select remote localhost:3333

Error message from debugger back end:
localhost:3333: No connection could be made because the target machine actively refused it.
Failed to execute MI command:
-target-select remote localhost:3333

Error message from debugger back end:
localhost:3333: No connection could be made because the target machine actively refused it.
localhost:3333: No connection could be made because the target machine actively refused it.

**### After going through couple of forums they suggested to run gdb first, we tried running that command but there also we are seeing errors, please find below forum link, command and screen shot for reference**.

similar issue faced by other
https://esp32.com/viewtopic.php?t=9079

Start GDN first using openocd command 
openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wroom-32.cfg

### but here im getting this error

Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: type 'esp32' is missing virt2phys
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'

igrr, 2020-10-07:
@Hullesha ESP32 DevKit-C board doesn't include a USB-JTAG interface. If you would like to set up JTAG debugging with DevKit-C, you will need to connect an external JTAG adapter. There are many FT2232H based adapters available, for example [ESP-Prog](https://github.com/espressif/esp-iot-solution/blob/master/documents/evaluation_boards/ESP-Prog_guide_en.md).  Once you have a JTAG adapter, connect it to the DevKit-C as described in this document: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-other-jtag.html.

Alvin1Zhang, 2020-12-04:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates or details. Thanks.

closed 2020-12-04 by Alvin1Zhang