espressif/esp-idf #5954: Deep sleep wakeup from ULP?

fpgamaster, 2020-10-07:
Hi All,
Is there someone successfully using an external oscillator and running an ULP code which wakes up the main CPU and this works for more than few days?
It is a very annoying issue and we are out of ideas why it works stable with the internal RC oscillator but not with the external one.
Any hints are welcome.

Best regards

felmue, 2020-10-08:
Hi @fpgamaster 

just curious, have you tried using an external crystal (instead of an external oscillator)? If yes, did that work better and was it more stable? (I guess it's probably not that accurate though.)

Thanks
Felix

fpgamaster, 2020-10-08:
Hi @felmue ,
Initially we have started using an external 32768 crystal but there are known issues with deep sleep mode. Adding an additional current from gpio pads (option in the menuconfig) is not possible if an ULP code is running. Using external crystal has same issues as described. So we have switched to an external oscillator on XP pin with Vpp ~0.8V but the issues remain :(
Also, on different boards which have 3.3V from LTO and another with 3.3V DC-DC converter is same despite running different ULP code and using different wake-up sources, one from ULP and another from timer.
At the same time switching to the internal RC oscillator fixes the issue !?

negativekelvin, 2020-10-08:
Have you tried rev3 chips?

fpgamaster, 2020-10-08:
Hi @negativekelvin,

It is ESP32-D0WD (revision 1) on ESP32-WROOM-32D.
May be it is not a bad idea to try rev2 or rev3 chips but not sure whether it will help since there are no issues in the ECO document related to use of an external RTC clock.

negativekelvin, 2020-10-08:
I think there are changes to 32k clock circuits in rev3

fpgamaster, 2020-10-08:
@negativekelvin
Yes, but they are related to using an external crystal because the oscillator stops sometimes in deep sleep mode.
There is no way for an external clock to stop and to cause the same issue if it is correctly routed internally while switching between normal and sleep modes.
Actually it is a guess that the clock stops or not correctly routed based on the fact that using internal 8MD256 clock does not have such issue.
May be someone who has access to the internal chipset schematic can shed some light?

felmue, 2020-10-09:
Hello @fpgamaster 

in your setup, is the ULP the only wake up source (e.g. no timer wake up)? And if yes, how often does it wake up the system or asked differently, how long is the longest period of sleep?

I don't have an active project, but I am interested in this topic so I've setup two spare boards (FireBeetle with external crystal and LyraT with external oscillator). For now I am running the deep sleep example (touch or GPIO wakeup disabled) with only ULP and timer wakeup source. The timer wakes the board every 20 seconds. I've attached an LED which I briefly turn on / off every time the board wakes up so I can tell it's still working w/o the need to have a serial console attached all the time.

Which brings me to the following question: how can I tell the external clock source (crystal or oscillator) is actually being used all the time? It is my understanding that every time the system reboots (after deep sleep) it checks for the external clock source and if missing automatically switches to the internal one. (That is at least what I can see when I have a serial console attached and temporarily disconnect the external clock source.) Is there a way to disable this automatic fall-back?

Thanks
Felix

fpgamaster, 2020-10-09:
Hi @felmue,

We have two setups. The first one uses wake up from ULP only (the ULP code is started right before entering to deep sleep) and the second one uses wakeup from timer while the ULP code is running for a while.
Both systems are in deep sleep for 30 seconds.
The external RTC clock is detected every time at boot including when returning from deep sleep.
Before entering to deep sleep the external RTC clock is selected again just to be sure that it is properly configured but there is no difference.
May be I will add some dumps of configuration registers to see if there are any differences.

BTW the second setup is more stable but has the same issue, sometimes not returning from deep sleep

Best regards

felmue, 2020-10-10:
Hi @fpgamaster 

ok, I've changed my two test boards to wake up every 30 seconds and I also check that they are actually using the external clock. So far (about 12 hours) they still work fine.

I was wondering, when in your setup the ESP32 stops waking from deep sleep, did you check the power supply to the external oscillator is still at a level you'd expect and if yes is the output still generating the 32.7 kHz clock? I am just thinking that maybe the device providing power to the ESP32 and external oscillator might have some mechanism which would cause it to turn off its output under light load?

A while ago you wrote that you are using an external oscillator with 'Vpp ~0.8V'. What is the intention behind a Vpp at the lower end of the recommended amplitude? The ESP32 Hardware Design Guidelines documentation reads '0.6 < Vpp < VDD'. In my setup with the oscillator I use a Vpp very close to VDD of 3.3V. Not sure if that matters though.

Best Regards
Felix

fpgamaster, 2020-10-10:
Hi @felmue ,
The documentation of using an external RTC clock is really weird :(
You can see in the menuconfig:
 - "External 32kHz oscillator" allows using 32kHz clock generated by an
    external circuit. In this case, external clock signal must be connected
    to 32K_XP pin. Amplitude should be <1.2V in case of sine wave signal,
    and <1V in case of square wave signal. Common mode voltage should be
    0.1 < Vcm < 0.5Vamp, where Vamp is the signal amplitude.
    Additionally, 1nF capacitor must be connected between 32K_XN pin and
    ground. 32K_XN pin can not be used as a GPIO in this case.
But in esp32_hardware_design_guidelines_en.pdf (Version 3.0 from 2020) section 2.4.3 -> 'Sine wave or square wave 0.6 < Vpp < VDD' to XN pin !?

What is 'Common mode voltage' !?

AFIR somewhere in the docs were stated that in deep sleep the internal circuits operate at 0.9V

According to the power supply, it was the first suspicious point, so we have changed the DC-DC converter with a different one which is specially designed for low currents (based on TPS561201) but no difference :(

Best regards

fpgamaster, 2020-10-10:
Hi @felmue,

See this:
https://esp32.com/viewtopic.php?t=10625

felmue, 2020-10-10:
Hi @fpgamaster 

I have a feeling that the menuconfig help text might be obsolete as it was added in August 2018:

https://github.com/espressif/esp-idf/commit/96971e3e9fb5e1569e5d28549c86b6dc7ab0644d

BTW: In my setup I tried XN and XP and for me it only works when the external clock is connected to the XP pin and XN is floating.

Thanks
Felix

felmue, 2020-10-15:
Hi guys

Update: My test setup still works as expected and the board wakes every 30 seconds.

For completeness:
- the external oscillator is powered through a 10 Ohm resistor in series from VDD *)
- the clock output is connected through a 22 pF capacitor to 32K_XP **)
- 32K_XN is floating

*) I've harvested the oscillator from another board (https://www.wandboard.org) and there a 10 Ohm resistor was used.
**) I've seen the usage of a series capacitor in the ESP32-S2 Hardware Design Guidelines.

Note: I am aware that I am using a setup different from what is proposed for an ESP32, but then again the proposal for an ESP32 seems to differ depending on which document one's consulting.

Cheers
Felix