espressif/esp-idf #5955: Does esp-idf have tools to create project scaffolding? It allows me to quickly start a new project.

ICLXL, 2020-10-08:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

igrr, 2020-10-08:
Hi @ICLXL, `idf.py create-project` command has been recently implemented and will be available on Github soon. We'll update this issue when the feature is released.

igrr, 2020-10-14:
Added in https://github.com/espressif/esp-idf/commit/b599f127b5e6ee9af12f4de20ac41e66fdf60e1c.

closed 2020-10-14 by igrr