espressif/esp-idf #5956: Generate one pulse with a specific width after an interrupt occurs

walidamriou, 2020-10-08:
Hi, I want to generate a pulse after an interrupt occurs (from a signal of 50 Hz), I have developed a code under ESP-IDF to do that. I have implemented the interrupt part by Semaphore and the pulse generator by RMT.  Everything OK but the problem there is a pulse generated with the same width of my pulse after it  (not every time but at a random time, sometimes generate and some times not), I have made this code by three different methods but I have the same problem (Semaphore, TaskNotify, and mcpwm with Queues) and for the three implementations I didn't change the part of Pulse generator! because I don't know another method to generate a pulse ( I don't like the classic method by timer)

Any suggestions or help or idea can help me?

negativekelvin, 2020-10-08:
https://github.com/espressif/esp-idf/issues/2943

walidamriou, 2020-10-11:
Hi @negativekelvin , 
the issue #2943 that you mentioned it, it didn't help me,  I have read all the comments and I found some people talk about the same things that happened to me. I have tested to detect all the types (raising, falling, high) of a signal (like zero-crossing with a pulse in every zero-crossing) and my work is to generate one pulse after the detection, everything fine but there are ****my flickers** in the generated pulse**! 

[image]

closed 2020-10-11 by walidamriou