espressif/esp-idf #5957: Wrong disconnected reason when provisioning & connecting to WEP network

ChromaMaster, 2020-10-08:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.3.2-323-gbf0220609
- Build System: Make
- Compiler version: 1.22.0-96-g2852398
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Hi. 

I'm trying to implement a ble provisioning mechanism for my esp32 using the example provided by espressif: [here](https://github.com/espressif/esp-idf/tree/release/v3.2/examples/provisioning/ble_prov).

The thing is that I'm trying to get some feedback from the system when this fails to connect to the provisioned nerwork. Everything works fine when I'm using WPA-PSK and WPA2-PSK encryption. The problem comes when I'm trying to use a WEP network. I'm not able to connect to a WEP-64bits network although I've tried a couple of routers.... and I can connect to a WEP-128bits network if the password matches (obviously) but I can't get the auth error if the password is wrong...

### Expected Behavior

The expected behaviour is to get the wifi connected to the WEP-64bits when the right password is provided and if not, at least get the correct error messages. For the WEP-128bits I expect to get the "Auth error" if the password is wrong.

### Actual Behavior

When I'm using the WEP-64bits encryption I'm not able to connect no matter what I'm always getting "Auth error". I've noticed that the wifi debug sais `wifi:invaild status 13 in auth response, retry auth request` so I tried another router and it doesn't work either.

When I'm using the WEP-128bits encryption I am able to connect to the network if the password is right, but if I provide a wrong password, I always get a "AP Not found" error although you can see that the wifi debug sais `wifi:sta status is wrong password` just before the other error is reported.

### Steps to reproduce

1. Clone the example mentioned before.
2. Enable the wifi debug using `make menuconfig` (Component config -> Wi-Fi -> Enable WiFi debug log)
3. Compile & flash
4. Provision the esp32 with a WEP-64bits network and see the logs
5. Provision the esp32 with a WEP-128bits network and see the logs

## Debug Logs

### For WEP-64

```
D (30971) wifi:Start wifi connect
D (30975) wifi:connect status 0 -> 0
D (30975) wifi:connect chan=0
D (30979) wifi:first chan=1
D (30983) wifi:connect status 0 -> 1
D (30987) wifi:filter: set rx policy=3
D (30987) wifi:clear scan ap list
D (30991) wifi:start scan: type=0x50f, priority=2, cb=0x400e732c, arg=0x0, ss_state=0x1, time=29079764, index=0
0x400e732c: cnx_start_handoff_cb at ??:?

D (31003) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (31251) wifi:scan end: arg=0x0, status=0, ss_state=0x3
...
D (33443) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (33443) wifi:perform scan: ss_state=0x9, chan<11,0>, dur<0,120>
D (33447) wifi:profile match: ss_state=0x7
D (33531) wifi:profile match: ss_state=0x7
D (33683) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (33687) wifi:find first mathched ssid, scan done
D (33687) wifi:filter: set rx policy=4
D (33687) wifi:first chan=1
D (33691) wifi:handoff_cb: status=0
D (33691) wifi:ap found, mac=XX:XX:XX:XX:XX:XX
D (33695) wifi:new_bss=0x3ffc8410, cur_bss=0x0, new_chan=<11,0>, cur_chan=9
D (33703) wifi:filter: set rx policy=5
I (33707) wifi:new:<11,0>, old:<9,1>, ap:<255,255>, sta:<11,0>, prof:9
D (33715) wifi:connect_op: status=0, auth=1, cipher=7 
D (33719) wifi:connect_bss: auth=1, reconnect=0
I (33723) wifi:state: init -> auth (b0)
D (33727) wifi:start 1s AUTH timer
D (33727) wifi:clear scan ap list
D (33739) wifi:recv auth: seq=2, status=13
D (33739) wifi:invaild status 13 in auth response, retry auth request
D (33747) wifi:recv auth: seq=2, status=0
D (33747) wifi:auth_shared: sta send auth, seq=3
D (34727) wifi:auth timeout
I (34731) wifi:state: auth -> init (200)
D (34731) wifi:connect status 1 -> 4
D (34731) wifi:stop beacon/connect timer
D (34731) wifi:reason: auth expire(2)
D (34735) wifi:add bssid XX:XX:XX:XX:XX:XX to blacklist, cnt=0
D (34739) wifi:stop CSA timer
D (34743) wifi:remove XX:XX:XX:XX:XX:XX from rc list
I (34747) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:9
D (34755) wifi:filter: set rx policy=8
D (34759) wifi:Send disconnect event, reason=2, AP number=0
E (34763) app_prov: STA Disconnected
E (34767) app_prov: Disconnect reason : 2
I (110639) app_prov: STA Auth Error
```

### For WEP-128

```
D (44429) wifi:Start wifi connect
D (44429) wifi:connect status 0 -> 0
D (44433) wifi:connect chan=0
D (44437) wifi:first chan=1
D (44437) wifi:connect status 0 -> 1
D (44441) wifi:filter: set rx policy=3
D (44445) wifi:clear scan ap list
D (44449) wifi:start scan: type=0x50f, priority=2, cb=0x400e732c, arg=0x0, ss_state=0x1, time=42626886, index=0
0x400e732c: cnx_start_handoff_cb at ??:?

D (44457) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (44705) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (44709) wifi:perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
...
W (46909) wifi:Invalid WEP key password

D (46909) wifi:connect status 1 -> 2
D (47145) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (47145) wifi:perform scan: ss_state=0x9, chan<12,0>, dur<360,360>
D (47385) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (47389) wifi:perform scan: ss_state=0x9, chan<13,0>, dur<360,360>
D (47629) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (47629) wifi:filter: set rx policy=4
D (47629) wifi:first chan=1
D (47633) wifi:handoff_cb: status=0
D (47633) wifi:clear rc list
D (47637) wifi:clear blacklist
D (47637) wifi:send disconnect event
D (47641) wifi:sta status is wrong password
D (47645) wifi:disable connect timer
D (47649) wifi:clear scan ap list
E (47653) app_prov: STA Disconnected
E (47657) app_prov: Disconnect reason : 201
I (54133) app_prov: STA AP Not found
```

Alvin1Zhang, 2020-10-12:
Thanks for reporting, we will look into.

kapilkedawat, 2020-10-13:
Hi @ChromaMaster 

From the logs, it seems like WEP64 is not supported by AP therefore it is sending an auth reject from with reason code 13(auth algo not supported).

For WEP128(incorrect password) -- can you please check the password which you are giving is in compliance with WEP128 format since the device is not able to parse it as WEP password(Invalid WEP key password).

Also, auth error will only come if AP is using shared WEP(ie giving error code in auth2/auth4), for open WEP there is no such mechanism for detection since open WEP automatically authenticates any client regardless of whether it actually has the correct WEP keys.

ChromaMaster, 2020-10-13:
Hi @kapilkedawat 

Fow the WEP64 I'll try to use another AP to check if the AP is the problem, thanks.

For the WEP128, I provided a wrong password intentionally. The issue is here is: although the device is no able to parse the password, as you said it actually prints the message `Invalid WEP key password` when I try to catch the error in the code I check the disconnected reason in the provisioning event handler ([here](https://github.com/espressif/esp-idf/blob/release/v3.2/examples/provisioning/ble_prov/main/app_prov.c#L227-L246)). Well, this reason is always `WIFI_REASON_NO_AP_FOUND (201)` which is not true since the AP can be found.

AxelLin, 2020-10-13:
> Hi @ChromaMaster ,
> 
> During the scan phase, device searches for configured SSID APs. If it finds one AP with WEP mode enabled, it will check whether the device password is in compliance for WEP password, if not it will not consider it a candidate and will search for next AP(WPA/WPA2 mode).
> 
> So the reason code is correct since the device didn't find any AP with the configured SSID and password(WPA/WPA2 mode).

You explained why device shows reason WIFI_REASON_NO_AP_FOUND (201).
But the point is WIFI_REASON_NO_AP_FOUND is a quite surprised reason because people
will check the SSID again and confused why the SSID is there but is *not found*..
i.e. From user's point of view, the reason code is incorrect.

sagb2015, 2020-10-14:
@ChromaMaster @AxelLin The reason code WIFI_REASON_NO_AP_FOUND should be interpreted as AP(SSID) _with required configuration_ is not found.  The password provided (length) is one of the factor determining the mode in which the device connects to the AP. The same is true if auth-threshold in scan is set to higher security level. WIFI_REASON_NO_AP_FOUND will not be seen if the password format is as per WEP specification but there is a mismatch.

AxelLin, 2020-10-14:
> @ChromaMaster @AxelLin The reason code WIFI_REASON_NO_AP_FOUND should be interpreted as AP(SSID) _with required configuration_ is not found. 

@sagb2015 
I disagree. As I said most people won't check configuration if got WIFI_REASON_NO_AP_FOUND.
Take a look at wifi_disconnect_reason_to_str(WIFI_REASON_NO_AP_FOUND), it shows "no ap found".

jack0c, 2022-05-17:
Maybe we can add a new reason code, from the original WIFI_REASON_NO_AP_FOUND, there are more cases:
1. real WIFI_REASON_NO_AP_FOUND
2. WIFI_REASON_NO_AP_FOUND  because of rssi threshold check
3. WIFI_REASON_NO_AP_FOUND  because of security threshold check
4. other reasons

jack0c, 2023-11-16:
added three new reasons:
'''
    WIFI_REASON_NO_AP_FOUND_W_COMPATIBLE_SECURITY  = 210,
    WIFI_REASON_NO_AP_FOUND_IN_AUTHMODE_THRESHOLD  = 211,
    WIFI_REASON_NO_AP_FOUND_IN_RSSI_THRESHOLD      = 212,
'''
in

4f47c40542b52490c9f898ec294b7933f7eb9811

closed 2023-11-20 by espressif-bot