espressif/esp-idf #5958: Flash Encryption and debug on Eclipse with JTAG (JLink) on ESP-IDF v4.1

Nomuc, 2020-10-09:
Environment

- Development Kit: [ESP32-DevKitC]
- Kit version (DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : V4.1 tag
- Build System: [CMake and idf.py]
- Compiler version : xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r2) 8.2.0
- Operating System: [Windows]
- environment type: [ESP Command Prompt|PowerShell].
- Using an IDE: Eclipse 2019-12
- Power Supply: [USB]

## Problem Description
I tested the flash encryption feature on ESP-IDF v4.1 with a devkit-C. I am debugging the board with a JLink at 15000kHz and OpenOCD configured in Eclipse. 
I used the following configuration tutorial available on your website (https://docs.espressif.com/projects/esp-idf/en/v4.1/api-guides/jtag-debugging/using-debugger.html). 
The problem is that after encrypting the flash, the debugging process is not working properly. Even after deactivating the encryption, we still have the same problem. We followed the tutorial on your website to disable the encryption (https://docs.espressif.com/projects/esp-idf/en/v4.1/security/flash-encryption.html#disabling-flash-encryption).
We reproduced this problem on 3 chips.

When my chip is encrypted I added the following option into my debug configuration:
> -c 'set ESP_FLASH_SIZE 0'

Here is my sdkconfig when we activated the encryption (JTAG is allowed): 
[image]

I also checked the summary of my EFUSE and JTAG is not disabled on my chip:
[image]

When trying to debug the board, I get the following error:
[debug_log.txt](https://github.com/espressif/esp-idf/files/5353725/debug_log.txt)

Eclipse looks like this:
[image]

I also tried to take a new chip to debug without encryption and it works perfectly. 
Does anyone have idea/or the same problem ?

### Expected Behavior
My chip should be able to debug the firmware the same way it used to before encrypting it.

### Actual Behavior
My chip is unable to start a debugging session on Eclipse

igrr, 2020-10-09:
Hi @Nomuc, according to your screenshot, FLASH_CRYPT_CNT is 3, which means that flash encryption is still enabled. This likely happens if the bootloader still has flash encryption enabled. Please double-check that you have uploaded both the bootloader and the app after disabling flash encryption in menuconfig, and only then program another bit in FLASH_CRYPT_CNT efuse.

Nomuc, 2020-10-12:
Hi @igrr, 

Ok I understand on this point, here is what I did:

I modified my sdkconfig and burned my FLASH_CRYPT_CNT:
[image]
According to your last message, my chip is not encrypted at count 0b111.

I erased my flash and flashed a non-encrypted firmware.

I got the same result in debugging session as my message above.

I'm still unable to debug my chip after disabling flash encryption, any ideas ?

igrr, 2020-10-14:
FLASH_CRYPT_CNT is now 7, which is the maximum value (no more bits can be set), and flash encryption is still enabled (odd number of bits set). Unfortunately it is no longer possible to disable flash encryption for this specific chip. I think the issue is that when you were running the espefuse.py command to disable flash encryption, the board was still flashed with the bootloader that has been compiled with flash encryption enabled. So as soon as the board was reset after burning the efuse, flash encryption got enabled again.

Nomuc, 2020-10-20:
Hi @igrr, 

Sorry for the late answer. No problem, I burned the FLASH_CRYPT_CNT once again now it is 0xf (see image below) 
[image]

My JTAG adapter is a JLink (15000kHz speed) and my chip is an ESP32-WROOM-32D on a devkit-C. My ESP-IDF version is tag V4.1 (sha : 5ef1b390026270503634ac3ec9f1ec2e364e23b2). I use Windows 10 for debugging and it runs natively on my computer.

The simple example to reproduce the issue is : 
1. Start a firmware with encrypted flash and NVS encryption.
2. Disable encrypted flash and try to debug.

Find the open_ocd log here : 
[openocd_log.txt](https://github.com/espressif/esp-idf/files/5408613/openocd_log.txt)

I added the command you gave to me into my gdbinit file into C:\Users\Nomuc\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\share\gdb . But I don't see any output file coming after starting a debugging session on Eclipse.

I hope it is helpful.

Nomuc, 2020-11-02:
Hi @igrr,

I want to up this message, I still have no idea of what is going on my chip.

igrr, 2020-11-05:
@Nomuc sorry that I have missed your reply earlier. Could you please check UART output, does the program run successfully when you are trying to debug? From the logs I see that you are flashing project.bin only. You don't seem to be flashing the bootloader and the partition table. If the bootloader and the partition table are still encrypted, then the chip will not boot successfully into your program, and therefore the breakpoint will not work.

tom-borcin, 2021-07-19:
Hello @Nomuc, did you manage to resolve your problem following the steps provided by @igrr?

Nomuc, 2021-07-19:
Hello,

Yes the answer was totally fine for me.

I'll close the issue.

closed 2021-07-19 by Nomuc

tom-borcin, 2021-07-19:
Thank you! Feel free to report any issue in the future. We are always happy to help.