espressif/esp-idf #5960: mbedtls_ssl_read blocking when esp32 try to download a bunch of files with maximum speed through https with mbedtls API

NOA-Vernon, 2020-10-10:
Dears, 

I'm trying to download a bunch of files through https from the server, however, sometimes the mbedtls_ssl_read function block the task and keeps blocking always, I refer to the esp/example/protocols/https_mbedtls, also I have modified some parameter of LWIP and mbedtls in menuconfig, 

CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=32
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534

see the attachment, is this a bug for the esp32? 
The startup log is like this, hope it will help

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7036
load:0x40078000,len:14736
load:0x40080400,len:4112
entry 0x40080678
[0;32mI (29) boot: ESP-IDF v4.3-dev-907-g6c17e3a64-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 16:56:18[0m
[0;32mI (30) boot: chip revision: 3[0m
[0;32mI (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0[0m
[0;32mI (41) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (46) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (55) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (94) boot:  3 factory          factory app      00 00 00010000 00280000[0m
[0;32mI (101) boot:  4 ota_0            OTA app          00 10 00290000 00280000[0m
[0;32mI (109) boot:  5 ota_1            OTA app          00 11 00510000 00280000[0m
[0;32mI (116) boot: End of partition table[0m
[0;32mI (120) boot: Defaulting to factory image[0m
[0;32mI (125) boot_comm: chip revision: 3, min. application chip revision: 0[0m
[0;32mI (132) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6cb70 (445296) map[0m
[0;32mI (334) esp_image: segment 1: paddr=0x0007cb98 vaddr=0x3ffbdb60 size=0x03480 ( 13440) load[0m
[0;32mI (341) esp_image: segment 2: paddr=0x00080020 vaddr=0x400d0020 size=0x15e910 (1435920) map[0m
[0;32mI (962) esp_image: segment 3: paddr=0x001de938 vaddr=0x3ffc0fe0 size=0x0166c (  5740) load[0m
[0;32mI (965) esp_image: segment 4: paddr=0x001dffac vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (969) esp_image: segment 5: paddr=0x001e03b8 vaddr=0x40080404 size=0x1b6fc (112380) load[0m
[0;32mI (1048) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (1049) boot: Disabling RNG early entropy source...[0m
[0;32mI (1060) psram: This chip is ESP32-D0WD[0m
[0;32mI (1061) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (1061) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (1066) spiram: PSRAM initialized, cache is in normal (1-core) mode.[0m
[0;32mI (1073) cpu_start: cpu freq: 240[0m
[0;32mI (1077) cpu_start: Pro cpu up.[0m
[0;32mI (1081) cpu_start: Single core mode[0m
[0;32mI (1978) spiram: SPI SRAM memory test OK[0m
[0;32mI (2007) cpu_start: Pro cpu start user code[0m
[0;32mI (2007) cpu_start: Application information:[0m
[0;32mI (2007) cpu_start: Project name:     toky_phase_2[0m
[0;32mI (2011) cpu_start: App version:      4c9bfc3-dirty[0m
[0;32mI (2017) cpu_start: Compile time:     Oct 10 2020 15:43:51[0m
[0;32mI (2023) cpu_start: ELF file SHA256:  65d7fd56ae3c7c50...[0m
[0;32mI (2029) cpu_start: ESP-IDF:          v4.3-dev-907-g6c17e3a64-dirty[0m
[0;32mI (2037) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (2043) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM[0m
[0;32mI (2050) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (2056) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (2062) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (2068) heap_init: At 3FFC8828 len 000177D8 (93 KiB): DRAM[0m
[0;32mI (2075) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[0m
[0;32mI (2081) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[0m
[0;32mI (2087) heap_init: At 4009BB00 len 00004500 (17 KiB): IRAM[0m
[0;32mI (2094) spiram: Adding pool of 3933K of external SPI memory to heap allocator[0m
[0;32mI (2103) spi_flash: detected chip: gd[0m
[0;32mI (2106) spi_flash: flash io: dio[0m
[0;32mI (2111) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (2116) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m

Alvin1Zhang, 2020-10-12:
Thanks for reporting, we will look into.

AdityaHPatwardhan, 2020-11-20:
Hi @NOA-Vernon Thank you for reporting the issue.
I don't see any obvious issue with your modified configurations.( Considering you have enough RAM available ).
Can you please provide the full debug log ( enabling log output with debug level ) !, you have mentioned some attachment, but I am unable to find it.

NOA-Vernon, 2020-11-20:
@AdityaHPatwardhan  I run the task below, it happens sporadically, you can test the  /esp-idf/examples/protocols/https_mbedtls  example for downloading a bunch of files through HTTPS, one by one.

static void download_get_task(void *pvParameters){

	//char buf[1024];
	int ret, flags, len;

	int i = 0;

	uint8_t read_http_header = 0;
	int8_t download_flag = 0;
	char* p = NULL;

	float cur_per = 0.0;
	float pre_per = 0.0;
	int total_read = 0;
	uint8_t led_stat = 0;
	TickType_t update_tick = xTaskGetTickCount();
	TickType_t updateled_tick = xTaskGetTickCount();
	TickType_t readhttp_tick = xTaskGetTickCount();

	mbedtls_entropy_context entropy;
	mbedtls_ctr_drbg_context ctr_drbg;
	mbedtls_ssl_context ssl;
	mbedtls_x509_crt cacert;
	mbedtls_ssl_config conf;
	mbedtls_net_context server_fd;

	mbedtls_ssl_init(&ssl);
	mbedtls_x509_crt_init(&cacert);
	mbedtls_ctr_drbg_init(&ctr_drbg);
	ESP_LOGI(TAG, "Seeding the random number generator");

	mbedtls_ssl_config_init(&conf);

	mbedtls_entropy_init(&entropy);
	if((ret = mbedtls_ctr_drbg_seed(&ctr_drbg, mbedtls_entropy_func, &entropy,
									NULL, 0)) != 0)
	{
		ESP_LOGE(TAG, "mbedtls_ctr_drbg_seed returned %d", ret);
		download_flag = -1;
		//abort();
		goto exit;
	}

	ESP_LOGI(TAG, "Attaching the certificate bundle...");

	ret = esp_crt_bundle_attach(&conf);

	if(ret < 0)
	{
		ESP_LOGE(TAG, "esp_crt_bundle_attach returned -0x%x", -ret);
		download_flag = -1;
		//abort();
		goto exit;
	}

	ESP_LOGI(TAG, "Setting hostname for TLS session...");

	 /* Hostname set here should match CN in server certificate */
	//if((ret = mbedtls_ssl_set_hostname(&ssl, WEB_SERVER)) != 0)
	if((ret = mbedtls_ssl_set_hostname(&ssl, fodetail.file[0].server)) != 0)
	{
		ESP_LOGE(TAG, "mbedtls_ssl_set_hostname returned -0x%x", -ret);
		download_flag = -1;
		//abort();
		goto exit; //
	}

	ESP_LOGI(TAG, "Setting up the SSL/TLS structure...");
[... 174 lines cut ...]
			mbedtls_strerror(ret, buf, 100);
			ESP_LOGE(TAG, "Last error was: -0x%x - %s", -ret, buf);
			download_flag = -2;
		}

		if (folder_download_fp != NULL) {
			fclose(folder_download_fp);
		}

		if( download_flag < 0){
			break;
		}

		vTaskDelay(100/portTICK_RATE_MS);

	}
	if( download_flag >= 0){		
		ESP_LOGI(TAG,"download succ");

	}else{
		ESP_LOGI(TAG,"download failed");
	}

	vTaskDelete(NULL);
}

AdityaHPatwardhan, 2020-12-15:
Hi @NOA-Vernon Sorry for the late reply, I was unable to reproduce this on my side. I can see you are creating multiple ssl connections for each file to download. Can you please tell me what it the number of files. If I have read above code correctly I think multiple connections are present simultaneously.It is highly probable that the error your are facing is because of a large number simultaneous connections but I would still like to see a complete debug log where the actual error message is printed.
1 ) Can you please provide a complete code with which I could be able to reproduce.
2 ) Provide a complete debug log in which error message can be seen.
Thank you.

NOA-Vernon, 2020-12-16:
@AdityaHPatwardhan can you test to download a bunch of files through HTTPS,  by testing this example esp-idf\examples\protocols\https_mbedtls, the HTTPS server is from Azure from the client, sorry for that I can't provide you the URL, I just add a for loop in this example for downloading a bunch of different files.

AdityaHPatwardhan, 2020-12-21:
Hi @NOA-Vernon I am able to download file from AWS s3 bucket repeatedly with your configurations, I dont see any obvious problem on my side. I couldn't get an Azure URL to test for myself, but I dont think having an azure URL should cause any problem as far as handshake is succeeded.
Do you face such error every time you use that to download the files ?

NOA-Vernon, 2020-12-22:
@AdityaHPatwardhan No, it happened sporadically and stuck at the ret = mbedtls_ssl_read(&ssl, (unsigned char *)buf, len) without returning, so I don't know what's wrong happen during calling this function.

AdityaHPatwardhan, 2020-12-22:
Then I think that it might be some network issue which is unrelated to the ESP. What is the frequency of this error, like you face it frequently ( and randomly) or you have faced this very few times ! ( like once or twice ).

NOA-Vernon, 2020-12-22:
@AdityaHPatwardhan If this is the network issue, how to set the mbedtls_ssl_read expired timeout, is there an example regarding it?

AdityaHPatwardhan, 2020-12-22:
you can set the `timeout_ms`  value in the `esp_tls_cfg_t` at the start of the TLS connection establishment. You can get relevant info about the options in the `esp_tls_cfg_t` at this [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#_CPPv411esp_tls_cfg). You can also take a look at the [ESP-TLS documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#_CPPv411esp_tls_cfg) for more details.

AdityaHPatwardhan, 2020-12-22:
My Bad, I mistakenly suggested to use an `esp_tls` option.( which internally uses mbedtls ). In your case you can set the timeout for `mbedtls_ssl_read` by setting the value of  `read_timeout` in the `mbedtls` configuration (`mbedtls_ssl_config`).

AdityaHPatwardhan, 2021-01-05:
Hi @NOA-Vernon if the problem is resolved, can you please close the issue ? thank you.

closed 2021-01-08 by NOA-Vernon