espressif/esp-idf #5962: idf_monitor failed with exit code 1

exrector, 2020-10-11:
MACOS BIG SUR
Hello friends!
I went through all the steps again. Lots of changes. The settings are slightly different from previous versions. I overcame all the mistakes. But I stumbled on the last step. I probably wouldn't do anything. BUT idf monitor is required in esp-rainmaker project to show QR codes. Any thoughts on how to fix this? thanks

```
Exrector@Mac-mini-Andrei hello_world % idf.py -p /dev/cu.usbserial-0001 monitor
Executing action: monitor
Running idf_monitor in directory /Users/exrector/esp/hello_world
Executing "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python /Users/exrector/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- /Users/exrector/esp/hello_world/build/hello-world.elf -m '/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python' '/Users/exrector/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0001'"...
Traceback (most recent call last):
  File "/Users/exrector/esp/esp-idf/tools/idf_monitor.py", line 51, in <module>
    import serial.tools.list_ports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
idf_monitor failed with exit code 1
```

dobairoland, 2020-10-11:
Hi @exrector. This is a known issue with Big Sur. You can find the workaround [here](https://github.com/espressif/esptool/issues/540#issuecomment-650507813)

exrector, 2020-10-11:
> Привет @exrector . Это известная проблема Big Sur. Вы можете найти обходной путь [здесь](https://github.com/espressif/esptool/issues/540#issuecomment-650507813)

You are talking about a flashing error. I don't have it. The flashing is great. Also, I don't have the specified file /Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py. My Mac is set to Python 3 by default.

`Project build complete. To flash, run this command:
/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python ../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello-world.bin
or run 'idf.py -p (PORT) flash'
Exrector@MacExrector hello_world % idf.py -p /dev/cu.usbserial-0001 [-b BAUD] flash
zsh: bad pattern: [-b
Exrector@MacExrector hello_world % idf.py -p /dev/cu.usbserial-0001 flash      
Executing action: flash
Running ninja in directory /Users/exrector/esp/hello_world/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /Users/exrector/esp/esp-idf/c...omponents/esptool_py/run_esptool.cmake
esptool.py --chip esp32 -p /dev/cu.usbserial-0001 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v3.0-dev
Serial port /dev/cu.usbserial-0001
Connecting.....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:72:32:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 945.8 kbit/s)...
Hash of data verified.
Compressed 24896 bytes to 15313...
Writing at 0x00001000... (100 %)
Wrote 24896 bytes (15313 compressed) at 0x00001000 in 0.4 seconds (effective 540.9 kbit/s)...
Hash of data verified.
Compressed 153088 bytes to 80884...
Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 153088 bytes (80884 compressed) at 0x00010000 in 1.9 seconds (effective 633.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
Exrector@MacExrector hello_world % `

This error only occurs in the last step

`Exrector@MacExrector hello_world % idf.py -p /dev/cu.usbserial-0001 monitor
Executing action: monitor
Running idf_monitor in directory /Users/exrector/esp/hello_world
Executing "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python /Users/exrector/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- /Users/exrector/esp/hello_world/build/hello-world.elf -m '/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python' '/Users/exrector/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0001'"...
Traceback (most recent call last):
  **File "/Users/exrector/esp/esp-idf/tools/idf_monitor.py", line 51, in <module>**
    import serial.tools.list_ports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
idf_monitor failed with exit code 1`

OR  (if I use idf.py -p /dev/cu.usbserial-0001 flash monitor)

`Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /Users/exrector/esp/hello_world
Executing "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python /Users/exrector/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- /Users/exrector/esp/hello_world/build/hello-world.elf -m '/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/bin/python' '/Users/exrector/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0001'"...
Traceback (most recent call last):
  File "/Users/exrector/esp/esp-idf/tools/idf_monitor.py", line 51, in <module>
    import serial.tools.list_ports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
idf_monitor failed with exit code 1
Exrector@MacExrector hello_world % `

exrector, 2020-10-12:
ok. Done
idf.py -p /dev/cu.usbserial-0001 -b **115200**

dobairoland, 2020-10-12:
Hi @exrector. I'm sorry I missed that point and haven't explained it well. The root cause of the issue is the same.

If you remove line no. 51 `import serial.tools.list_ports` of `/Users/exrector/esp/esp-idf/tools/idf_monitor.py` and keep using `idf.py -p /dev/cu.usbserial-0001 monitor` then is your issue solved or you get another error?

exrector, 2020-10-12:
> Привет @exrector . Мне жаль, что я упустил этот момент и не объяснил его как следует. Основная причина проблемы та же.
> 
> Если удалить строку № 51 `import serial.tools.list_ports`из `/Users/exrector/esp/esp-idf/tools/idf_monitor.py`и продолжайте использовать, `idf.py -p /dev/cu.usbserial-0001 monitor`тогда ваша проблема решена или вы получите другую ошибку?

Thanks for your help @dobairoland . The error remained and was slightly modified. Perhaps there is a reason to delete (comment out) the rest of the points given in the error?
#import serial.tools.list_ports

bash-3.2$ idf.py -p /dev/cu.usbserial-0001 monitor
Executing action: monitor
Running idf_monitor in directory /Users/exrector/esp-homekit-sdk/examples/lightbulb
Executing "/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/bin/python /Users/exrector/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- /Users/exrector/esp-homekit-sdk/examples/lightbulb/build/lightbulb.elf -m '/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/bin/python' '/Users/exrector/esp/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0001'"...
Traceback (most recent call last):
  File "/Users/exrector/esp/esp-idf/tools/idf_monitor.py", line 50, in <module>
    import serial.tools.miniterm as miniterm
  File "/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/lib/python3.8/site-packages/serial/tools/miniterm.py", line 16, in <module>
    from serial.tools.list_ports import comports
  File "/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/.espressif/python_env/idf4.1_py3.8_env/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
idf_monitor failed with exit code 1
bash-3.2$

dobairoland, 2020-10-12:
Thank you @exrector for the followup. This means that miniterm is also broken which we cannot bypass without crippling IDF Monitor. I'm afraid that IDF Monitor cannot be used until they patch Big Sur. 

Until then, I recommend to use a terminal program to see the logs from ESP32, e.g.:

```sh
picocom -b 115200 /dev/cu.usbserial-0001
```

Note: any terminal program should work after specifying the port and the baudrate. You can find one with GUI as well.

exrector, 2020-11-17:
> ```shell
> picocom
> ```
Hello @dobairoland 
In both picocom and arduino ide monitor I get QR of very poor quality. Do you by chance know the reason?
[image]

dobairoland, 2020-11-17:
No, sorry @exrector, I don't. But you can check the recommendations of the QR code library (if there any). It may cause also the terminal window. Have you tried another?

exrector, 2020-11-17:
> Нет, прости @exrector , я не знаю. Но вы можете проверить рекомендации библиотеки QR-кодов (если есть). Это может вызвать также окно терминала. Вы пробовали другой?

Yes. I've tried a bunch of options. Alas, some projects I need do not generate a web link. In the process of finding out now

dobairoland, 2020-11-18:
Hi @exrector. There is one more option now but it requires a little bit work. 

There seems to be a work-in-progress fix for the original issue: https://github.com/pyserial/pyserial/pull/542

I have no Mac to test it. But if you would be willing to experiment and manually install pyserial from that branch then it might fix the issue and IDF Monitor would work for you. Otherwise, you have to wait until that fix get published. 

[Building a new version of Python from source with brew](https://github.com/espressif/esptool/issues/540#issuecomment-726970888) also should solve the original issue.

NevynUK, 2020-11-26:
FYI - `pyserial` 3.5 was released on 23 Nov.  I've just updated my Mac running Big Sur and I can run run `idf.py` to build, flash and monitor an application.

Hope this helps.

dobairoland, 2020-11-26:
Thank you @NevynUK for sharing the good news.

dobairoland, 2020-11-27:
I'm closing this now because this is essentially not and IDF issue. We have identified a couple of workarounds, but I believe that the final version of Big Sur and the new version of pyserial should be sufficient for everyone.

closed 2020-11-27 by dobairoland

Francois-Belanger, 2021-01-13:
For those with the issue:  (for me)
pip3 install --upgrade pyserial