espressif/esp-idf #5963: AES encryption support for softAP mode in ESP32

coolesthead, 2020-10-11:
**Is your feature request related to a problem? Please describe.**

I've updated my phone to iOS14 and I tried connecting to the soft AP of ESP32. I had chosen WPA2_PSK as AUTH mode.  When I connect my phone, iOS is giving the warning that, WiFi is using WPA2(TKIP) as encryption and suggesting to use WPA2(AES)/WPA3. Since WPA3 is not available for softAP, I'm requesting to support WPA2(AES) for softAP mode. 

**Describe the solution you'd like**

Adding support for AES for WPA2_PSK or adding WPA3 as auth mode for softAP

**Describe alternatives you've considered**

I don't think any alternatives available as far this is concerned.

**Additional context**

iOS13 didn't show warning if we use WPA2(TKIP). Only after the iOS14 update, they marked WPA2(TKIP) as unsafe. Only WPA2(AES) or WPA3 are considered as safe.

Alvin1Zhang, 2020-10-12:
Thanks for raising this feature request.

sagb2015, 2020-10-15:
@coolesthead The softAP already supports AES. By default, it uses mixed mode (CCMP+TKIP) when security is used. iOS13 maybe cribbing because of the mixed mode. We will add additional parameter in the config to choose the cipher-suite which is currently missing. WPA3 development for SoftAP is in progress.

coolesthead, 2020-10-15:
Thanks @sagb2015 this helps.

jcsbanks, 2020-11-08:
Is it possible to change something in release 3.3 to use CCMP only? I can find the enum WIFI_CIPHER_TYPE_CCMP but cannot see how to use it. Users are getting locked out on Windows 10 after its latest update.

AxelLin, 2020-12-14:
Just tried testing current master tree v4.3-dev-2136-gb0150615dff5 with iphone12, it shows warnings and does not allow join the esp32 softap.
I think this is a *bug*.

AxelLin, 2020-12-14:
> @coolesthead The softAP already supports AES. By default, it uses mixed mode (CCMP+TKIP) when security is used.

Hi @sagb2015 
Is there any workaround to make softAP use AES instead the default mixed mode?

lukasmalkmus, 2020-12-29:
Interested in this as well. Mainly to get rid of the >= iOS13 warnings.

zinke-ct-video, 2021-01-26:
+1 Our business is also interested in this feature.

AxelLin, 2021-01-29:
WPA3 for softap is a feature request.
But, IMHO, allow config softAP as AES should be consider as a bug fix.
When will this get fix?

kapilkedawat, 2021-01-30:
@AxelLin Please note mixed mode in WPA2 is a valid scenario and not a bug. 

We will provide a new option to set pairwise cipher in softAP configuration so that users can select TKIP/CCMP/TKIP+CCMP as they require(with the default option as TKIP+CCMP).

AxelLin, 2021-01-31:
> @AxelLin Please note mixed mode in WPA2 is a valid scenario and not a bug.

This is not what I complaint.
> We will provide a new option to set pairwise cipher in softAP configuration so that users can select TKIP/CCMP/TKIP+CCMP as they require(with the default option as TKIP+CCMP).

This is indeed what I complaint and I think it is a bug. (AFAIK, non-esp32 sdks can allow user to set this setting).

closed 2021-02-01 by igrr