espressif/esp-idf #5965: Ethernet PHY LAN8720a does not initialize correctly after warm reboot

eflukx, 2020-10-12:
## Environment

- Development Kit:  Olimex ESP32-Gateway
- Kit version Rev.G (GPIO17 as EMAC_CLK_OUT)
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-2182-gdddcc2ede
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: VSCode
- Power Supply: USB

## Problem Description

I'm working on a project based on the ESP32 gateway board from Olimex. Networking works fine, both Ethernet and WiFi. Only after remotely resetting the device (device is controlled over MQTT) the ethernet does not come up again. Internally `esp_restart()` is called to reboot the device. Only after power-cycling the device ethernet comes up again. 

When the phy is in the non-working state, both phy LED's (at the eth connector) blink at the same time, with a rate of about 1 second. (Just after reset, during initialization, I might see some faster blinks, it looks like its working for a brief moment, but immediately after the LED's get into the slow-blink (of death) again). The log shows something like this (logged in the ethernet event callback):

```
I (371226) ugw_eth: Ethernet Link Up
I (371226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (373226) ugw_eth: Ethernet Link Down
I (383226) ugw_eth: Ethernet Link Up
I (383226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (385226) ugw_eth: Ethernet Link Down
I (395226) ugw_eth: Ethernet Link Up
I (395226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (397226) ugw_eth: Ethernet Link Down
I (409226) ugw_eth: Ethernet Link Up
I (409226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (411226) ugw_eth: Ethernet Link Down
I (421226) ugw_eth: Ethernet Link Up
I (421226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (423226) ugw_eth: Ethernet Link Down
I (433226) ugw_eth: Ethernet Link Up
I (433226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (435226) ugw_eth: Ethernet Link Down
I (447226) ugw_eth: Ethernet Link Up
I (447226) ugw_eth: Ethernet HW Addr c8:2b:96:b5:2f:47
I (449226) ugw_eth: Ethernet Link Down
```
The logs show no errors whatsoever (logging level DEBUG in idf).

In [this revision of the Olimex board](https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/Hardware%20revision%20G/ESP32-GATEWAY_Rev_G.pdf) GPIO17 is used as EMAC clock out and GPIO5 is used for switching power to the PHY on and off. On the board the LAN8710A chip is used.

I have been playing with an initial pre-ethernet-init routine toggling the power pin with different wait times (up to 5 second poweroff before poweron). Sometimes there seemed to be some result, but all in all I can not seem to get this consistently and reliably working.
Also setting the `phy_config.reset_gpio_num = 5` for enabling reset via power pin or disabling this (set to `-1`) is not helping.

### Expected Behavior

Correct initialization of LAN8720 PHY after every startup, also when not power cycling.

### Actual Behavior

Blink of death as described above.

### Steps to reproduce

- Reboot device and wait 'till it connects to the MQTT broker.
- Send the `reset` command. See the described non-working behavior.

eflukx, 2020-10-12:
Some more info and a possible lead:

I also have an older revision of the Olimex board (Rev.B). In this hardware revision, the PHY has it's own oscillator and is providing the EMAC Clock externally through GPIO0. This version does not show the behavior as described and is working correctly.

Maybe the problem has something to do with correctly setting up the EMAC clock out?

suda-morris, 2020-10-13:
from your log, it looks like your PHY is in an unstable state after restart. I don't have such board, but a LAN8720A breakout board, clocking RMII from GPIO17, no PHY power control. Follow your steps, after got IP address then `esp_restart`, it works fine.
One thing might be that, your PHY power is not stable after restart, so the PHY is in a unstable state when esp32 reading the PHY register. Can you try read the BMSR register twice to make sure this register does reflect the latest state. 
https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_phy_lan8720.c#L181

DanKoloff, 2020-10-13:
We switched to more convetional Ethernet design in revision D. The major change that was introduced to the PHY clock circuit in hardware revision D is namely:

LAN8710's pin XTAL1 was disconnected from GPIO0 and connected to GPIO17/EMAC_CLK_OUT_180. GPIO17 is now source clock. GPIO5 is now used as Ethernet "power enable pin".

All changes can be found here:

https://github.com/OLIMEX/ESP32-GATEWAY/blob/master/HARDWARE/README.md

eflukx, 2020-10-13:
@suda-morris thanks for your feedback. Yes i guess it must have something to do with the LAN8710 getting into an unexpected state. The strange thing is that this happens also when I manually (in a separate init routine) power the phy off and on for a longer period (tried > 1 second) to make sure it is in a normal "powered on" state. Doing this (or not doing this) does not seem to have any effect. I will try how the device behaves when the PHY is really always on by bridging the 'L1' inductor (which is normally not populated). Also will test with the esp-idf ethernet example code to find out if maybe another part of my code is interfering somehow.

Another thought is that maybe the GPIO17 clk out is outputting unexpected waveform on a warm-reset. Will test this with my scope.

@DanKoloff thanks for your feedback. Yes I'm aware of the hardware changes between the revisions. I've implemented the sw changes in order to facilitate the newer board revisions (without the changes, the board would obviously not have a functioning ethernet at all). Pity Olimex decided to drop the external oscillator; probably saves a few bucks, but you loose GPIO17 (and at least in my case, causes some trouble too...)

DanKoloff, 2020-10-13:
I didn't understand if you tested with this:

https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.0

It is not about saving bucks - the spent time redesigning this part, making prototypes to test new revision, etc, etc costed much more than just keeping the original design and using the parts previously used.  It is that the original design perplexed people since no other board or design uses similar approach as we did. All information available online was recommending a power enable disable pin and the internal clock. We didn't even consider external generator for our newer designs like ESP32-PoE. We wanted to remove the external generator in ESP32-EVB also but it was impossible due to lack of a free pin.

eflukx, 2020-10-13:
@DanKoloff Ok.. thanks for the clarification, I will try the code your suggesting!

If I understand you correctly you say using the internal generator is the preferred way nowadays? (I know during the advent of the earlier Olimex gateway board the internal gen. was not available yet in the then-current IDF versions..)
I have a few other custom designs using the LAN8710  as PHY both with internal and external clk generation (both seem to work perfectly fine, although I haven't tested the warm-boot stuff back then..).

Only when an external PSRAM is required, the GPIO17 must be left open for that (or at least it was in older IDF versions, haven't checked if these pins can be remapped in more current versions...) 

Ps. you're right the "shaving some bucks"-part was just my assumption.

DanKoloff, 2020-10-13:
My bad, wrong link for ESP32-PoE, this is for ESP32-GATEWAY:

https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/SOFTWARE/Newest-board-revisions/ESP32_Gateway_Ethernet_v4.0

eflukx, 2020-10-13:
It's a Rev.F

DanKoloff, 2020-10-13:
That might cause the discrepancy, the C5 addition and the R24 bump to 10k was added in revision G.

eflukx, 2020-10-14:
> That might cause the discrepancy, the C5 addition and the R24 bump to 10k was added in revision G.

Hi Dan, what discrepancy are you referring to? C5/R24 have to do with the hardware reset of the ESP32 (holding the `EN` line low for some RC-time), which seems to be working fine on my Rev.F. The `EN` pin is not touched during a soft-reset (and initing the PHY), which is what we're having problems with.

Will test the other software (examples) today, keep ya posted!

eflukx, 2020-10-16:
Ok, I've done some more testing with the example code, unfortunately all tests with the same result: **ethernet phy reaches a non-working state after a soft reset. 

I left the example code untouched and emulated the soft-resetting by programming a second time using esptool (and resetting the ESP32 device). This left the ethernet (phy) in the problem state after the (esptool) soft-reset. I think we can safely say this rules out bugs in my application code, there must be some problem in the Olimex (Rev.F) hardware, a bug in ESP_IDF or a combination of the two.

### Testing approach

All compilation building of the firmware I have done by using the ESP-IDF Docker container to rule out the possibility of interference due to my local setup and to be able to easily switch IDF versions. The build command used (IDF v4.0 case shown):

    docker run --rm -ti -v `realpath .`:/firmware -w /firmware espressif/idf:release-v4.0 idf.py build

I've tested the following example(app) / ESP-IDF version combinations:

- [Olimex example](https://github.com/OLIMEX/ESP32-GATEWAY/tree/master/SOFTWARE/Newest-board-revisions/ESP32_Gateway_Ethernet_v4.0) with ESP-IDF version v4.2
- Olimex example with ESP-IDF version v4.0
- IDF ethernet example using ESP-IDF version v3.3 (the given Olimex example is not compatible with IDF v3.x)
- IDF ethernet example using ESP-IDF version v3.3 (with PHY power option enabled)

Some console output with more detailed version output of the three tests:

```
I (259) cpu_start: Application information:
I (264) cpu_start: Project name:     ethernet_basic
I (269) cpu_start: App version:      1
I (274) cpu_start: Compile time:     Oct 14 2020 20:34:13
I (280) cpu_start: ELF file SHA256:  14cdb997f7ed79c1...
I (286) cpu_start: ESP-IDF:          v4.2-dev-2182-gdddcc2ede
I (292) cpu_start: Starting app cpu, entry point is 0x4008183c
```
```
I (246) cpu_start: Application information:
I (251) cpu_start: Project name:     ethernet_basic
I (257) cpu_start: App version:      1
I (261) cpu_start: Compile time:     Oct 16 2020 10:21:44
I (267) cpu_start: ELF file SHA256:  3b2bb0080ec84c51...
I (273) cpu_start: ESP-IDF:          v4.0.1-441-g6093407d7
I (279) cpu_start: Starting app cpu, entry point is 0x40081170
```
```
I (240) cpu_start: Application information:
I (245) cpu_start: Project name:     esp-idf
I (250) cpu_start: App version:      1
I (254) cpu_start: Compile time:     Oct 16 2020 10:59:38
I (260) cpu_start: ELF file SHA256:  7c3240d097063e9c...
I (266) cpu_start: ESP-IDF:          v3.3.3-146-gfb18b12ef
I (272) cpu_start: Starting app cpu, entry point is 0x40080f8c
```

eflukx, 2020-10-16:
Some further testing showed that the issue also pops up when hard-reset is used (e.g. via `EN` line triggered by serial (`RTS`) or the button). So @DanKoloff maybe you're right on the reset circuitry not (R24/C5) holding `EN` low long enough for the ESP32 to properly initialize.

Still don't see how that would interfere with soft resetting the device though (don't think `EN` pin is 
pulled on soft reset, but have not tested that)...

**Update**

If have soldered a new RC circuit onto my Rev.F board: 1k5/10uF. Different values compared to the Rev.G board, but should be a similar RC-time to the Rev.G board (I'm at home only have a few parts lying around here, will try next week with the correct values) 
I've been playing around with randomly resetting the board using the `RST1` button on the board. It seems to work a little better then before the RC-mod: longer presses seem to result in a working PHY most of the time, pressing shorter _seems_ to result more often in the non-working state. 
Using `esptool.py` to reset the device (I just reprogram it) still result in a non-working PHY every time.

eflukx, 2020-11-05:
The problem confirmed to exist across different IDF versions with a correct (or at least defined) hardware implementation. I'd say this is a bug in esp-idf (?). Maybe @suda-morris has some input on this?

eflukx, 2020-11-27:
Ok guys, thanks for your support!!

The other day I was trying to reproduce this issue: I'm not able to reproduce this anymore. I have several Olimex ESP32-Gateway boards lying around. Possibly the issue was caused by a hardware issue with one of the boards (?).

Closing this issue.

closed 2020-11-27 by eflukx

Connect-and-Exchange, 2021-08-11:
Having the same issue and was intrigued reading the entire post. Hopefully my issue also magically disappears whereas I doubt it. 

Esp.restart() is always causing the Ethernet to remain down and you see occasional flicker of the leds on both the esp and switch side. A hard reset with the button always works for me … 

if anyone has a suggestion then I’m all ears …

eflukx, 2021-08-11:
@Connect-and-Exchange Hi Martin, it seems that in my case, in _most_ circumstances the problem does not surface. I _did_ seem to be able to reproduce this issue again using a certain cable/ethernet switch combination (switch in case, is a consumer grade TP-Link PoE device). I will try to reproduce this issue again (after vacation 🌊 😄) and see if I can find the exact reproduction parameters. I have a (gut) feeling it might differ if an external 50MHz clock or the internal ESP32 EMAC clock (CLKOUT) is used...

Will report back on this in some of weeks...

eflukx, 2021-08-11:
@Connect-and-Exchange Just to get a better grip on the findings: what kind of hardware are you using?

Board name/type (i was experiencing the issues using an 'Olimex ESP32-Gateway' Rev.G board)?
Using internal/external EMAC clock? Using a 8710a or 8720a PHY?
Make and model of switch used and length/quality of the cable used? 

Cheers!

Connect-and-Exchange, 2021-08-11:
I’m using an esp32-evb rev.I and an esp32-gateway rev.g. Both are experiencing the same problem. When reset through software or the esp crashes (because of some bad coding ;-) ) the Ethernet connection starts blinking and never comes online. 

Once the reset button is pressed the Ethernet connection is established without any problems. 

Using a cisco 2960l-pp 8 port and tested on different managed and unmanaged switches. Wiring is sftp cat6 with a length of 1.8 meters.

Connect-and-Exchange, 2021-08-11:
8270a seems to be on the chip.

Connect-and-Exchange, 2021-08-11:
The esp32-evb uses also the 8710a

[image]

SERIDJ, 2021-12-07:
hi @eflukx  iam working on a costume board with lan8720A and i have this problem LAN after sometimes cant connect and get ip adress, so i update my design pcb bcs i think its problem of  RMII trace impedance and clk too , so with my last pcb the board work 3 month without stop and after reboot its stops to work sometimes and now its work again . and this without hardware rest .nRST pin is connected to vcc for now ...

eflukx, 2021-12-08:
@SERIDJ I only have experience with the LAN8710a IC, and it has been very stable, apart from the warm-reset issue that is. When it is running it keeps running/online. I really don't know how the IP101GRI performs. 

btw. I doubt if the problems have something to do with impedance matching etc. as the frequencies are really not that high (50MHz, ~6m wavelength) and the traces (at least on my boards) are as short as possible (maybe 30mm max). When we have a signal integrity issue, I would expect the interface to be intermittently failing or giving a high error rate. That does not seem to be the case here. To me it really looks like the PHY is not initialized properly when it is in a certain state (that is typically only reached when a warm/soft system reset is issued).

SERIDJ, 2021-12-08:
@eflukx 

yes ur right, 

**i found this forum : https://www.eevblog.com/forum/microcontrollers/no-ethernet-phy-connectivity-with-esp-32-wroom/25/

 ```

Did the wESP32 developer already tell you that you need to drive NRST properly? Simply pulling it high permanently will not work and the Olimex hack with a cap does not work for a warm reset. I've been running into this issue myself in the last few days, to where the LAN8710A resets continuously.

I assume you don't care about using an extra pin, so you can make sure to drive NRST properly:

1. Pull NRST low
2. Wait for the 50mhz clock to come up (it might be already up)
3. Wait for a small period (100us worked in my case)
4. Pull NRST high

For me this makes the LAN8710a run rock solid and more importantly it can handle warm resets/debugging sessions. My code: https://github.com/tinic/lightguy/blob/master/ethernetif.cpp#L73 Code is for an STM32F107 clone, not the ESP32, so not really useful for you :-)

 ```

one think its about order , bcs with esp idf we cant pull nRst low and wait for clk , we can do this : 

1-wait first for clk to comme up
2-call reset hardware  :  ---->nRST low for 100us and pull it high **

SERIDJ, 2021-12-08:
@eflukx  i test now my board , with the hardware reset the problem is not fixed ... 

 ```
*****************************************
*****************************************
*****************************************
****************normal Eth***************
*****************************************
*****************************************
*****************************************
*****************************************
enbale oscillator
enbale done

*****************************************
*****************************************
*****************************************
****************eth dhcp***************
*****************************************
*****************************************
*****************************************
*****************************************
esp_rom_gpio_pad_select_gpio(lan8720->reset_gpio_num) is : 13
nrst LOW 100us
nrst HIGH
I (1087) system_api: Base MAC address is not set
I (1091) system_api: read default base MAC address from EFUSE
I (1118) esp_eth.netif.glue: 24:0a:c4:5f:82:57
I (1118) esp_eth.netif.glue: ethernet attached to netif
I (1130) eth_example: Ethernet Started
 ```

mihai-en, 2023-11-12:
The issue is reproducible for me always using an Ubiquiti POE passthrough switch but is not reproducible if I use a 24 port Tp Link switch .   Very annoying since the only way to fix this remotely I have to cut the switch power on&off.   Any workaround so I can avoid the issue only from the ESP32-POE-ISO board ?

mihai-en, 2023-11-12:
I think I managed to workaround it by adding the following code before restart . Somebody was saying that it might also work if done before Eth.begin().  

pinMode(ETH_PHY_POWER, INPUT);
digitalWrite(ETH_PHY_POWER, 0);
pinMode(ETH_PHY_POWER, OUTPUT);
delay(99);

hxmelab, 2026-01-11:
We encountered stability issues during a phased firmware rollout for a fleet of over 2000 devices in the field. The hardware used is the Olimex ESP32-POE-ISO-IND equipped with the LAN8710AiEZK PHY. In our first rollout batch, 2 out of 20 devices failed to come back online after an OTA update or even a soft reboot.

[image]

The logs showed an endlessly link flap with exactly 2000ms timeout, failing to obtain an IP. The failure affected the exact same two devices every time. The other 18 units worked flawlessly with the same firmware. We were unable to reproduce this behavior on other devices.

Our initial attempt to simply toggle ETH_PHY_POWER (set to 0, wait 100ms, set to 1, wait 2000ms) in the startup sequence did not solve the issue.

After deeper investigation, the issue was traced to a possible phantom power keeping the LAN8710AiEZK in a zombie state after a soft reboot. We suggest, pulling the PHY_POWER pin low to reset the chip, the ESP32 continues to drive the RMII data lines and especially the CLK pin. Current flows through the ESD protection diodes of the PHY inputs, may preventing the internal voltage rail from dropping to 0V. Hence, it could be possible this prevents the PHY from performing a clean internal reset.

However, our fix involves completely electrically isolating the PHY during the startup sequence. We actively force the clock pin to GND and set all data pins to INPUT/FLOATING before cycling the power. Here is the c++ implementation that solved the issue for us. We call this function at the very beginning of app_main(), before calling the **esp_eth_driver_install()**.  

```

#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// Adjust pins to your specific hardware config
#define PIN_PHY_POWER 12
#define PIN_RMII_CLK  17 

void force_phy_reset_clean() {
    ESP_LOGW("PHY_RST", "Starting clean PHY reset sequence (Isolation Mode)...");

    // Force the Clock to 0V immediately to stop oscillation/noise
    gpio_reset_pin((gpio_num_t)PIN_RMII_CLK);
    gpio_set_direction((gpio_num_t)PIN_RMII_CLK, GPIO_MODE_OUTPUT);
    gpio_set_level((gpio_num_t)PIN_RMII_CLK, 0);

    // Isolate all data pins to prevent phantom power back-feeding
    // Add all RMII pins used in your design here (MDC, MDIO, TX, RX, etc.)
    const gpio_num_t phy_pins[] = {
        (gpio_num_t)23, (gpio_num_t)18, (gpio_num_t)19, (gpio_num_t)22, 
        (gpio_num_t)21, (gpio_num_t)25, (gpio_num_t)26
    };

    for (auto pin : phy_pins) {
        gpio_reset_pin(pin); 
        gpio_set_direction(pin, GPIO_MODE_INPUT);
        gpio_set_pull_mode(pin, GPIO_FLOATING);
    }

    // Power Cycle
    gpio_reset_pin((gpio_num_t)PIN_PHY_POWER);
    gpio_set_direction((gpio_num_t)PIN_PHY_POWER, GPIO_MODE_OUTPUT);
    gpio_set_level((gpio_num_t)PIN_PHY_POWER, 0); // Power OFF

    // Wait for deep discharge
    vTaskDelay(pdMS_TO_TICKS(250)); 
	
	// Power ON
    gpio_set_level((gpio_num_t)PIN_PHY_POWER, 1); 
    ESP_LOGI("PHY_RST", "PHY Power restored. Waiting for stabilization...");
    
    // Wait for oscillator startup
    vTaskDelay(pdMS_TO_TICKS(2000)); 
}

```