espressif/esp-idf #5969: Modbus Slave Stops Working

MrCurio, 2020-10-13:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-907-g6c17e3a64
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): esp-2020r1
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [VsCode]
- Power Supply:         [USB]

## Problem Description
Hello everyone, 
I have been testing the Modbus software on my ESP32 and i have encountered an issue regarding the UART Communication. 

I am running a modbus slave program (similiar to the slave example) for communicating with a raspberry pi 4b using a **RS-232** protocol. I have connected all the pins straight to the RPI (Rx, Tx, RTS, CTS) and i was able to communicate over **modbus RS-232** with the ESP32. However, after some time (varies depending on reading frequency, usually less than a day with 5s), seems that the ESP32 cannot handle the communication anymore and it constantly gives timeouts. When i try to read the logs when failing, **NOTHING** appears in the console of the esp32...no errors or anything.

I have checked the menuconfig variables (including log levels) and everything seems to be correct, am i missing something?

I firstly assumed it could be a rpi problem so i reset it and tried again, no success. Then, i tried connecting with a PC over FTDI using Modbus and nothing. The only solution that i could find was reseting the ESP32.

I read in #4182 that the issue with the RTS pin was solved, can that be my problem besides i am using the v4.3 and it should be fixed?

Can anyone help me understand this issue?

PD: I don't know know if it is relevant, but i am using ESP-BLE_MESH while using Modbus.

Thanks!

MrCurio

### Expected Behavior
Stable Modbus RS-232 communication that can last days/months

### Actual Behavior
Randomly stops working without advertising

### Steps to reproduce

1.  Get a rpi and a ESP32
2. Connect them using UART pins (Rx, Tx, RTS, CTS)
3. Flash ESP32 modbus serial slave example (i have a modified version of it, but should also fail)
4. Read from RPI using modbus RS-232 with a frequency of 5 seconds for hours
5. Wait until you can't keep on reading from rpi

### Code to reproduce this issue
Modbus Serial Slave example.

I use a modified version that just use a Queue for receiving data and transfer it to the registers of Modbus.

## Debug Logs

NONE

## Other items if possible

Attached sdskconfig as Txt File:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5369942/sdkconfig.txt)

alisitsyn, 2020-10-20:
Hello @MrCurio,

Thank you for your issue and sorry for delay with answer. The existing modbus example stops communication once the input value reach the alarm value but as I understand this is not your case. The #4182 was solving issue with UART driver however there is other issue in Modbus driver which may keep the RTS line in TX state in some conditions and prevent to receive other input packets. This is known issue and fix is ready but not merged yet. However let us try to check my guess of the reason. It usually happens when in your app you have other tasks (WiFi, BT) with higher priority than Modbus tasks. 
The function uart_wait_tx_done() can disable the `UART_TX_DONE_INT_ENA_M` interrupt and the driver does not set the RTS line back in the interrupt.
```
BOOL xMBMasterPortSerialTxPoll(void)
{
    USHORT usCount = 0;
    BOOL bNeedPoll = TRUE;

    if( bTxStateEnabled ) {
        // Continue while all response bytes put in buffer or out of buffer
        while(bNeedPoll && (usCount++ < MB_SERIAL_BUF_SIZE)) {
            // Calls the modbus stack callback function to let it fill the UART transmit buffer.
            bNeedPoll = pxMBMasterFrameCBTransmitterEmpty( ); // callback to transmit FSM
        }
        ESP_LOGD(TAG, "MB_TX_buffer sent: (%d) bytes.", (uint16_t)(usCount - 1));
        // Waits while UART sending the packet
-       esp_err_t xTxStatus = uart_wait_tx_done(ucUartNumber, MB_SERIAL_TX_TOUT_TICKS); 
+      esp_err_t xTxStatus = uart_wait_tx_idle_polling(ucUartNumber); // <<<<<<< change the function call
        vMBMasterPortSerialEnable( TRUE, FALSE );
        MB_PORT_CHECK((xTxStatus == ESP_OK), FALSE, "mb serial sent buffer failure.");
        return TRUE;
    }
    return FALSE;
}
```
The change above should help to solve the issue but since the Modbus driver uses polling approach you may need to configure the FMB_PORT_TASK_PRIO to be higher than max priority of your tasks.

Unfortunately I do not have your application and can not reproduce the issue directly.

Please try the above suggestions and let me know the result.

Thanks.

MrCurio, 2020-10-21:
Hello @alisitsyn, 
Thank you for answering!

I have deeply reviewed my code and I can ensure I have no other tasks with a higher priority than the Modbus Stack. 
I will make the change you recommend me for solving the issue and i will continue testing my application. When i get the new results (may take one week or more) i will let you know. 

I'm sorry, but I cannot share my application since it is not open source. However, if problems persists, I will prepare a modified example for this issue.

Is there any other bug in the Modbus Stack that i should be aware of? If so, may i get a temporary patch before merging?

Thanks.

alisitsyn, 2020-10-22:
Hello @MrCurio,

> I'm sorry, but I cannot share my application since it is not open source. However, if problems persists, I will prepare a modified example for this issue.

Yes, if you still have an issue then send me some example to be able to reproduce the issue.
Also please try to disable the CONFIG_FMB_TIMER_PORT_ENABLED option in kconfig.

This patch fixes the event processing issue in modbus master:
[fix_mbm_event_processing.zip](https://github.com/espressif/esp-idf/files/5421300/fix_mbm_event_processing.zip)

Please let me know the result. Thanks.

MrCurio, 2020-10-22:
Hello @alisitsyn ,
I will continue testing the application. 
Thank you for the patch and suggestions. I will continue updating this issue as i move forward.

alisitsyn, 2020-10-22:
MrCurio,

Ahh, I just observed that the changes related to CONFIG_FMB_TIMER_PORT_ENABLED in port serial are also not merged into master. Please apply the changes attached and check.
[modbus_fix_portserial_issues.zip](https://github.com/espressif/esp-idf/files/5422802/modbus_fix_portserial_issues.zip)

Thanks.

MrCurio, 2020-10-22:
Hi @alisitsyn ,
I will only apply the last patch you sent me to master because it seems it gathers all the previous patches. If I am wrong, please correct me.

Thank you!

alisitsyn, 2020-10-26:
@MrCurio,

The last patch has the fix of xMBMasterPortSerialTxPoll(), so the manual fix above should be reverted before apply. The fix_mbm_event_processing.diff can be applied to v4.3 together with modbus_fix_portserial_issues.diff.

MrCurio, 2020-11-03:
Hello, its been a week i applied the patches you gave me and the problem seems to be resolved. I will close this issue and reopen if further problems appear.

Thanks!

closed 2020-11-03 by MrCurio

alisitsyn, 2020-11-03:
Hello @MrCurio,

Yes, feel free to reopen issue if you have further problems. 

Thanks.