espressif/esp-idf #5970: On release 4.1 branch a lot more code is pulled in from libcoexist

erikcorry, 2020-10-13:
## Environment

- Development Kit: Custom
- Kit version: Custom
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1-357-gc249ed646 - based off 7b7c641076ef4c5dff7b994170d445bbb46f81ab
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: No

## Problem Description
As I understand it, libcoexist is used used to control the hardware
that the Wifi and the Bluetooth implementation have in common.
I am using Wifi, but not Bluetooth.  On moving to the 4.1 branch
I have a lot of new symbols in my executable, which is causing
and increase in binary size:

I get this list with objdump -t on the elf file. 

+coex_schm_all_default
+coex_schm_ble_default_bt_a2dp_wifi_conn
+coex_schm_ble_default_bt_a2dp_wifi_connecting
+coex_schm_ble_default_bt_a2dp_wifi_scan
+coex_schm_ble_default_bt_default_wifi_conn
+coex_schm_ble_default_bt_default_wifi_connecting
+coex_schm_ble_default_bt_default_wifi_scan
+coex_schm_ble_default_bt_idle_wifi_conn
+coex_schm_ble_default_bt_idle_wifi_connecting
+coex_schm_ble_default_bt_idle_wifi_scan
+coex_schm_ble_mesh_config_bt_a2dp_paused_wifi_conn
+coex_schm_ble_mesh_config_bt_a2dp_paused_wifi_connecting
+coex_schm_ble_mesh_config_bt_a2dp_paused_wifi_scan
+coex_schm_ble_mesh_config_bt_a2dp_wifi_conn
+coex_schm_ble_mesh_config_bt_a2dp_wifi_connecting
+coex_schm_ble_mesh_config_bt_a2dp_wifi_scan
+coex_schm_ble_mesh_config_bt_conn_wifi_conn
+coex_schm_ble_mesh_config_bt_conn_wifi_connecting
+coex_schm_ble_mesh_config_bt_conn_wifi_scan
+coex_schm_ble_mesh_config_bt_default_wifi_conn
+coex_schm_ble_mesh_config_bt_default_wifi_connecting
+coex_schm_ble_mesh_config_bt_default_wifi_scan
+coex_schm_ble_mesh_config_bt_iscan_wifi_conn
+coex_schm_ble_mesh_config_bt_iscan_wifi_connecting
+coex_schm_ble_mesh_config_bt_iscan_wifi_scan
+coex_schm_ble_mesh_config_bt_sniff_wifi_conn
+coex_schm_ble_mesh_config_bt_sniff_wifi_connecting
+coex_schm_ble_mesh_config_bt_sniff_wifi_scan
+coex_schm_ble_mesh_config_wifi_conn
+coex_schm_ble_mesh_config_wifi_connecting
+coex_schm_ble_mesh_config_wifi_scan
+coex_schm_ble_mesh_standby_bt_a2dp_paused_wifi_conn
+coex_schm_ble_mesh_standby_bt_a2dp_paused_wifi_connecting
+coex_schm_ble_mesh_standby_bt_a2dp_paused_wifi_scan
+coex_schm_ble_mesh_standby_bt_a2dp_wifi_conn
+coex_schm_ble_mesh_standby_bt_a2dp_wifi_connecting
+coex_schm_ble_mesh_standby_bt_a2dp_wifi_scan
+coex_schm_ble_mesh_standby_bt_conn_wifi_conn
+coex_schm_ble_mesh_standby_bt_conn_wifi_connecting
+coex_schm_ble_mesh_standby_bt_conn_wifi_scan
+coex_schm_ble_mesh_standby_bt_default_wifi_conn
+coex_schm_ble_mesh_standby_bt_default_wifi_connecting
+coex_schm_ble_mesh_standby_bt_default_wifi_scan
+coex_schm_ble_mesh_standby_bt_iscan_wifi_conn
+coex_schm_ble_mesh_standby_bt_iscan_wifi_connecting
+coex_schm_ble_mesh_standby_bt_iscan_wifi_scan
+coex_schm_ble_mesh_standby_bt_sniff_wifi_conn
+coex_schm_ble_mesh_standby_bt_sniff_wifi_connecting
+coex_schm_ble_mesh_standby_bt_sniff_wifi_scan
+coex_schm_ble_mesh_standby_wifi_conn
+coex_schm_ble_mesh_standby_wifi_connecting
+coex_schm_ble_mesh_standby_wifi_scan
+coex_schm_ble_mesh_traffic_bt_a2dp_paused_wifi_conn
+coex_schm_ble_mesh_traffic_bt_a2dp_paused_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_a2dp_paused_wifi_scan
+coex_schm_ble_mesh_traffic_bt_a2dp_wifi_conn
+coex_schm_ble_mesh_traffic_bt_a2dp_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_a2dp_wifi_scan
+coex_schm_ble_mesh_traffic_bt_conn_wifi_conn
+coex_schm_ble_mesh_traffic_bt_conn_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_conn_wifi_scan
+coex_schm_ble_mesh_traffic_bt_default_wifi_conn
+coex_schm_ble_mesh_traffic_bt_default_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_default_wifi_scan
+coex_schm_ble_mesh_traffic_bt_iscan_wifi_conn
+coex_schm_ble_mesh_traffic_bt_iscan_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_iscan_wifi_scan
+coex_schm_ble_mesh_traffic_bt_sniff_wifi_conn
+coex_schm_ble_mesh_traffic_bt_sniff_wifi_connecting
+coex_schm_ble_mesh_traffic_bt_sniff_wifi_scan
+coex_schm_ble_mesh_traffic_wifi_conn
+coex_schm_ble_mesh_traffic_wifi_connecting
+coex_schm_ble_mesh_traffic_wifi_scan
+coex_schm_bt_a2dp_paused_wifi_conn
+coex_schm_bt_a2dp_paused_wifi_connecting
+coex_schm_bt_a2dp_paused_wifi_scan
+coex_schm_bt_a2dp_wifi_conn
+coex_schm_bt_a2dp_wifi_connecting
+coex_schm_bt_a2dp_wifi_scan
+coex_schm_bt_conn_wifi_conn
+coex_schm_bt_conn_wifi_connecting
+coex_schm_bt_conn_wifi_scan
+coex_schm_bt_default_wifi_conn
+coex_schm_bt_default_wifi_connecting
+coex_schm_bt_default_wifi_scan
+coex_schm_bt_idle_wifi_conn
+coex_schm_bt_idle_wifi_connecting
+coex_schm_bt_idle_wifi_scan
+coex_schm_bt_iscan_wifi_conn
+coex_schm_bt_iscan_wifi_connecting

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

erikcorry, 2020-10-13:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5371786/sdkconfig.zip)

MaxwellAlan, 2020-11-09:
hi @erikcorry thanks for reporting this issue.
We have already optimize our ld script in master branch to only link used symbols, and will backport this optimization to release/v4.1 lately.
Thanks.

jack0c, 2020-12-10:
Have an MR for this in IDFv4.1 in internal gitlab, just need to wait the CI push it to GitHub.

Alvin1Zhang, 2020-12-16:
Reopen until the fix is available on GitHub.

projectgus, 2021-05-20:
Closed in 649d834

closed 2021-05-20 by projectgus