espressif/esp-idf #5972: Intermitten WiFi Connection Issue

spadespwnz, 2020-10-13:
## Environment

- Development Kit:      [ESP32-DevKitC|ESP32-PICO-Kit|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v3]
- Module or chip used:  [ESP32-PICO-D4]
- IDF version (run ``git describe --tags`` to find it):
    v4.2-beta1-86-gdddcc2ede
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (please give details)]
- Power Supply:         [USB|external 3.3V]

## Problem Description

I am experiencing intermittent WiFi connection errors on some networks. I can successfully connect to the WiFi network, then after some amount of time will be disconnected. Reconnecting will fail for about an hour before successfully reconnecting. Restarting does not solve the issue.

I have had this issue using esp-idf version 3.3, 4.0 and 4.2.
This issue only occurs on some networks, while working fine on others.
I have 2 D-Link DIR-890L Routers, the issue always occurs on one yet never on the other.

### Expected Behavior

A WiFi connection will be maintained, and will successfully reconnect when dropped.

### Actual Behavior

The WiFi connection is dropped and unable to reconnect for ~1 hour.

### Steps to reproduce

Connect to WiFI network and wait.

## Debug Logs

### At first the connection is successful:

connect status 0 -> 0
connect status 0 -> 1
profile match: ss_state=0x7
profile match: ss_state=0x7
ap found, mac=90:8d:78:4d:8c:0c
new_bss=0x3ffca5b0, cur_bss=0, new_chan=<11,0>, cur_chan=1
connect_op: status=0, auth=5, cipher=3 
auth mode is not none
connect_bss: auth=1, reconnect=0
state: init -> auth (b0)
recv auth: seq=2, status=0
state: auth -> assoc (0)
restart connect 1s timer for assoc
recv assoc: type=0x10
state: assoc -> run (10)
start 10s connect timer for 4 way handshake
connected with dlink-8C0C, aid = 15, channel 11, BW20, bssid = 90:8d:78:4d:8c:0c
security: WPA2-PSK, phy: bgn, rssi: -63
Send sta connected event
connect status 1 -> 5
D (6428) esp_netif_lwip: check: remote, if=0x3ffd8bcc fn=0x401439a8

D (6428) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffd8bcc
D (6438) esp_netif_lwip: check: local, if=0x3ffd8bcc fn=0x40144550

D (6438) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd8bcc
D (6438) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (6448) esp_netif_lwip: check: remote, if=0x3ffd8bcc fn=0x401436e0

D (6448) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffd8bcc
D (6448) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd8bcc
D (6448) esp_netif_lwip: if0x3ffd8bcc start ip lost tmr: no need start because netif=0x3ffd8c50 interval=120 ip=0
D (6458) esp_netif_lwip: starting dhcp client
D (6458) esp_netif_lwip: call api in lwip: ret=0x0, give sem
AP's beacon interval = 102400 us, DTIM period = 1
D (7918) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffd8c50
D (7928) esp_netif_lwip: if0x3ffd8bcc ip changed=1
D (7928) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffd8bcc
D (7928) esp_netif_lwip: esp-netif has been successfully initialized

### Then after some amount of time it will get into a loop of attempting to reconnect but failing:

recv deauth, reason=0xe
state: run -> init (ec0)
recv deauth/disassoc, stop beacon/connect timer
connect status 5 -> 4
pm stop, total sleep time: lu us / lu us
Send disconnect event, reason=14, AP number=0
D (103424488) esp_netif_lwip: check: remote, if=0x3ffd8bcc fn=0x40143b24

D (103424498) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffd8bcc
D (103424498) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffd8c50
D (103424498) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd8bcc
D (103424508) esp_netif_lwip: if0x3ffd8bcc start ip lost tmr: interval=120
D (103424508) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd8bcc
D (103424508) esp_netif_lwip: if0x3ffd8bcc start ip lost tmr: already started
D (103424518) esp_netif_lwip: check: local, if=0x3ffd8bcc fn=0x40144550

D (103424518) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd8bcc
D (103424518) esp_netif_lwip: call api in lwip: ret=0x0, give sem
connect status 4 -> 0
connect status 0 -> 1
E (103424538) aws_iot: failed! mbedtls_ssl_write returned -0x4e
profile match: ss_state=0x7
profile match: ss_state=0x7
ap found, mac=90:8d:78:4d:8c:0c
new_bss=0x3ffca5b0, cur_bss=0, new_chan=<11,0>, cur_chan=11
connect_op: status=0, auth=5, cipher=3 
auth mode is not none
connect_bss: auth=1, reconnect=0
state: init -> auth (b0)
recv auth: seq=2, status=0
state: auth -> assoc (0)
restart connect 1s timer for assoc
recv auth: seq=2, status=0
not auth state, ignore
recv assoc: type=0x10
state: assoc -> run (10)
start 10s connect timer for 4 way handshake
recv deauth, reason=0xe
state: run -> init (ec0)
recv deauth/disassoc, stop beacon/connect timer
connect status 1 -> 2
Send disconnect event, reason=14, AP number=0
D (103426308) esp_netif_lwip: check: remote, if=0x3ffd8bcc fn=0x40143b24

D (103426308) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffd8bcc
D (103426308) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd8bcc
D (103426318) esp_netif_lwip: if0x3ffd8bcc start ip lost tmr: already started
D (103426318) esp_netif_lwip: check: local, if=0x3ffd8bcc fn=0x40144550

D (103426318) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd8bcc
D (103426328) esp_netif_lwip: call api in lwip: ret=0x0, give sem
connect status 2 -> 0
connect status 0 -> 1
send disconnect event
connect status 1 -> 3
D (103428468) esp_netif_lwip: check: remote, if=0x3ffd8bcc fn=0x40143b24

D (103428468) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffd8bcc
D (103428468) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffd8bcc
D (103428478) esp_netif_lwip: if0x3ffd8bcc start ip lost tmr: already started
D (103428478) esp_netif_lwip: check: local, if=0x3ffd8bcc fn=0x40144550

D (103428478) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd8bcc
D (103428488) esp_netif_lwip: call api in lwip: ret=0x0, give sem
connect status 3 -> 0
connect status 0 -> 1
profile match: ss_state=0x7
ap found, mac=90:8d:78:4d:8c:0c
new_bss=0x3ffca5b0, cur_bss=0, new_chan=<11,0>, cur_chan=11
connect_op: status=0, auth=5, cipher=3 
auth mode is not none
connect_bss: auth=1, reconnect=0
state: init -> auth (b0)
recv auth: seq=2, status=0
state: auth -> assoc (0)
restart connect 1s timer for assoc
recv assoc: type=0x10
state: assoc -> run (10)
start 10s connect timer for 4 way handshake
recv deauth, reason=0xe
state: run -> init (ec0)
recv deauth/disassoc, stop beacon/connect timer
connect status 1 -> 2
Send disconnect event, reason=14, AP number=0

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5371952/sdkconfig.txt)

spadespwnz, 2020-10-19:
Bump, any hint as to why this is occurring?

negativekelvin, 2020-10-20:
`recv deauth, reason=0xe` suggests TKIP MIC error and could result in AP temporary blacklist (for 1 hour).  One possible cause of MIC failure is from bad signal or interference.

YouDONG-ESP, 2020-10-27:
Hi @spadespwnz, Could you check the two routers D-Link DIR-890L have the same software version. I think it's not a problem from our side considering one of your routers works. In addition, the log shows that is the router who send a deauth frame.

spadespwnz, 2020-10-27:
I do believe the issue is caused by the router, but what's concerning is that it only seems to happen when my ESP32-based devices are connected. It's possible that the issue is coming from the increased number of devices connected, and not the ESP32 specifically. I have read that setting the router to 20Mhz instead of 20/40 has fixed network issues for some people using the router, and will test that.
Thanks for the ideas, from your input it sounds like its not an issue caused by the ESP32 but simply an issue on the router.

closed 2020-11-06 by Alvin1Zhang