espressif/esp-idf #5974: DM90501 ethernet can't run example

giorgio80, 2020-10-13:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version v4.0

- Operating System:     [macOS]
- (Windows only) environment type: [PowerShell].
- Using an IDE?:  Platformio
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

new to specific ESP-IDF programming I'm trying to run DM90501 spi ethernet example. when trying to compile I'm getting various error of undefined variables and I suppose it depends from menuconfig configuration. 
actually running menuconfig, btw, I inserted support for dm90501 

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

plain Ethernet code in esp-idf example , unmodified

## Debug Logs

`src/main.c:83:27: error: 'CONFIG_EXAMPLE_ETH_PHY_ADDR' undeclared (first use in this function)
     phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:83:27: note: each undeclared identifier is reported only once for each function it appears in
src/main.c:84:33: error: 'CONFIG_EXAMPLE_ETH_PHY_RST_GPIO' undeclared (first use in this function); did you mean 'CONFIG_ETH_RMII_CLK_OUT_GPIO'?
     phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                 CONFIG_ETH_RMII_CLK_OUT_GPIO
src/main.c:90:24: error: 'CONFIG_EXAMPLE_DM9051_MISO_GPIO' undeclared (first use in this function)
         .miso_io_num = CONFIG_EXAMPLE_DM9051_MISO_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:91:24: error: 'CONFIG_EXAMPLE_DM9051_MOSI_GPIO' undeclared (first use in this function)
         .mosi_io_num = CONFIG_EXAMPLE_DM9051_MOSI_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:92:24: error: 'CONFIG_EXAMPLE_DM9051_SCLK_GPIO' undeclared (first use in this function)
         .sclk_io_num = CONFIG_EXAMPLE_DM9051_SCLK_GPIO,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:96:40: error: 'CONFIG_EXAMPLE_DM9051_SPI_HOST' undeclared (first use in this function)
     ESP_ERROR_CHECK(spi_bus_initialize(CONFIG_EXAMPLE_DM9051_SPI_HOST, &buscfg, 1));
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
src/main.c:101:27: error: 'CONFIG_EXAMPLE_DM9051_SPI_CLOCK_MHZ' undeclared (first use in this function)
         .clock_speed_hz = CONFIG_EXAMPLE_DM9051_SPI_CLOCK_MHZ * 1000 * 1000,`

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

suda-morris, 2020-10-14:
Have you enabled the `CONFIG_EXAMPLE_USE_DM9051` properly in the menuconfig?

Alvin1Zhang, 2020-10-28:
Thanks for reporting, feel free to reopen. Thanks.

closed 2020-10-28 by Alvin1Zhang