espressif/esp-idf #5975: idf.py build on Windows 10 goes into infinite loop

tozz88, 2020-10-13:
Describe the bug
When I execute

idf.py build

the build gets stuck in an infinite loop it reruns the config portion over and over. I'm including the com window output. My environment is esp-idf with btstack installed on Windows 10 machine. I'm trying to build a btstack demo that has worked fine
for many days. Not sure what changed to cause this issue. Stale file somewhere perhaps? I have tried issuing "fullclean" and
"reconfig" idf.py commands but they don't seem to help. If I just do an "idf.py reconfig" this completes but then following it
with a "idf.py build" puts it back in infinite loop.

To Reproduce
Here is what I do to reproduce the behavior.

Open "ESP-IF Command Prompt" window shortcut created during installation
cd to previously working project ....\esp\btstack\port\esp\hid_mouse_demo
idf.py build
get infinite loop. Have to ctl-C to stop it.

Here is output:
Using Python in C:\Users\tozz\AppData\Local\Programs\Python\Python36
Python 3.6.8
Using Git in C:\Program Files\Git\cmd
git version 2.21.0.windows.1
Setting IDF_PATH: C:\Users\tozz\Desktop\esp-idf-2

Adding ESP-IDF tools to PATH...
C:\Users\tozz.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin
C:\Users\tozz.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
C:\Users\tozz.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
C:\Users\tozz.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
C:\Users\tozz.espressif\tools\cmake\3.13.4\bin
C:\Users\tozz.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin
C:\Users\tozz.espressif\tools\ninja\1.9.0
C:\Users\tozz.espressif\tools\idf-exe\1.0.1
C:\Users\tozz.espressif\tools\ccache\3.7
C:\Users\tozz.espressif\python_env\idf4.1_py3.6_env\Scripts
C:\Users\tozz\Desktop\esp-idf-2\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\tozz\Desktop\esp-idf-2\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

idf.py build

C:\Users\tozz\Desktop\esp-idf-2>cd ....\esp\btstack\port\esp32\example\hid_mouse_demo

C:\Users\tozz\esp\btstack\port\esp32\example\hid_mouse_demo>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\tozz\esp\btstack\port\esp32\example\hid_mouse_demo\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project version: f21bd20e5-dirty
-- Building ESP-IDF components for target esp32
-- Adding linker script C:/Users/tozz/esp/btstack/port/esp32/example/hid_mouse_demo/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt btstack cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/tozz/Desktop/esp-idf-2/components/app_trace C:/Users/tozz/Desktop/esp-idf-2/components/app_update C:/Users/tozz/Desktop/esp-idf-2/components/asio C:/Users/tozz/Desktop/esp-idf-2/components/bootloader C:/Users/tozz/Desktop/esp-idf-2/components/bootloader_support C:/Users/tozz/Desktop/esp-idf-2/components/bt C:/Users/tozz/Desktop/esp-idf-2/components/btstack C:/Users/tozz/Desktop/esp-idf-2/components/cbor C:/Users/tozz/Desktop/esp-idf-2/components/coap C:/Users/tozz/Desktop/esp-idf-2/components/console C:/Users/tozz/Desktop/esp-idf-2/components/cxx C:/Users/tozz/Desktop/esp-idf-2/components/driver C:/Users/tozz/Desktop/esp-idf-2/components/efuse C:/Users/tozz/Desktop/esp-idf-2/components/esp-tls C:/Users/tozz/Desktop/esp-idf-2/components/esp32 C:/Users/tozz/Desktop/esp-idf-2/components/esp_adc_cal C:/Users/tozz/Desktop/esp-idf-2/components/esp_common C:/Users/tozz/Desktop/esp-idf-2/components/esp_eth C:/Users/tozz/Desktop/esp-idf-2/components/esp_event C:/Users/tozz/Desktop/esp-idf-2/components/esp_gdbstub C:/Users/tozz/Desktop/esp-idf-2/components/esp_http_client C:/Users/tozz/Desktop/esp-idf-2/components/esp_http_server C:/Users/tozz/Desktop/esp-idf-2/components/esp_https_ota C:/Users/tozz/Desktop/esp-idf-2/components/esp_https_server C:/Users/tozz/Desktop/esp-idf-2/components/esp_local_ctrl C:/Users/tozz/Desktop/esp-idf-2/components/esp_netif C:/Users/tozz/Desktop/esp-idf-2/components/esp_ringbuf C:/Users/tozz/Desktop/esp-idf-2/components/esp_rom C:/Users/tozz/Desktop/esp-idf-2/components/esp_serial_slave_link C:/Users/tozz/Desktop/esp-idf-2/components/esp_websocket_client C:/Users/tozz/Desktop/esp-idf-2/components/esp_wifi C:/Users/tozz/Desktop/esp-idf-2/components/espcoredump C:/Users/tozz/Desktop/esp-idf-2/components/esptool_py C:/Users/tozz/Desktop/esp-idf-2/components/expat C:/Users/tozz/Desktop/esp-idf-2/components/fatfs C:/Users/tozz/Desktop/esp-idf-2/components/freemodbus C:/Users/tozz/Desktop/esp-idf-2/components/freertos C:/Users/tozz/Desktop/esp-idf-2/components/heap C:/Users/tozz/Desktop/esp-idf-2/components/idf_test C:/Users/tozz/Desktop/esp-idf-2/components/jsmn C:/Users/tozz/Desktop/esp-idf-2/components/json C:/Users/tozz/Desktop/esp-idf-2/components/libsodium C:/Users/tozz/Desktop/esp-idf-2/components/log C:/Users/tozz/Desktop/esp-idf-2/components/lwip C:/Users/tozz/esp/btstack/port/esp32/example/hid_mouse_demo/main C:/Users/tozz/Desktop/esp-idf-2/components/mbedtls C:/Users/tozz/Desktop/esp-idf-2/components/mdns C:/Users/tozz/Desktop/esp-idf-2/components/mqtt C:/Users/tozz/Desktop/esp-idf-2/components/newlib C:/Users/tozz/Desktop/esp-idf-2/components/nghttp C:/Users/tozz/Desktop/esp-idf-2/components/nvs_flash C:/Users/tozz/Desktop/esp-idf-2/components/openssl C:/Users/tozz/Desktop/esp-idf-2/components/partition_table C:/Users/tozz/Desktop/esp-idf-2/components/perfmon C:/Users/tozz/Desktop/esp-idf-2/components/protobuf-c C:/Users/tozz/Desktop/esp-idf-2/components/protocomm C:/Users/tozz/Desktop/esp-idf-2/components/pthread C:/Users/tozz/Desktop/esp-idf-2/components/sdmmc C:/Users/tozz/Desktop/esp-idf-2/components/soc C:/Users/tozz/Desktop/esp-idf-2/components/spi_flash C:/Users/tozz/Desktop/esp-idf-2/components/spiffs C:/Users/tozz/Desktop/esp-idf-2/components/tcp_transport C:/Users/tozz/Desktop/esp-idf-2/components/tcpip_adapter C:/Users/tozz/Desktop/esp-idf-2/components/ulp C:/Users/tozz/Desktop/esp-idf-2/components/unity C:/Users/tozz/Desktop/esp-idf-2/components/vfs C:/Users/tozz/Desktop/esp-idf-2/components/wear_levelling C:/Users/tozz/Desktop/esp-idf-2/components/wifi_provisioning C:/Users/tozz/Desktop/esp-idf-2/components/wpa_supplicant C:/Users/tozz/Desktop/esp-idf-2/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/tozz/esp/btstack/port/esp32/example/hid_mouse_demo/build
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project version: f21bd20e5-dirty
-- Building ESP-IDF components for target esp32
-- Adding linker script C:/Users/tozz/esp/btstack/port/esp32/example/hid_mouse_demo/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/tozz/Desktop/esp-idf-2/components/esp32/ld/esp32.peripherals.ld
[... cut ...] sp-idf-2/components/wear_levelling C:/Users/tozz/Desktop/esp-idf-2/components/wifi_provisioning C:/Users/tozz/Desktop/esp-idf-2/components/wpa_supplicant C:/Users/tozz/Desktop/esp-idf-2/components/xtensa
-- Configuring done
^C
C:\Users\tozz\esp\btstack\port\esp32\example\hid_mouse_demo>ninja: error: rebuilding 'build.ninja': interrupted by user

Aborted!

C:\Users\tozz\esp\btstack\port\esp32\example\hid_mouse_demo>
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5374573/sdkconfig.zip)

dobairoland, 2020-10-14:
Hi @tozz88. You are experiencing an issue building an example project included in btstack. I don't even see support for Cmake in the master branch of btstack nor could I find the project you are using.

Please report this issue for btstack. I'm closing this but I can reopen if you can reproduce it with an example provided for ESP-IDF.

closed 2020-10-14 by dobairoland

georgik, 2022-06-07:
This is old issue, but it's good to have root cause analysis and solution:

The problem with Infinite CMake loop is caused by timestamp in future. 
Solutions: 
- touch all files to reset timestamp 
- change the time on local machine
This may happen in CI, when it has incorrect timezone or incorrect time, so the code is cloned from the "future"

taherrera, 2023-02-08:
This happened to me because my pc had an incorrect time, probably a timestamp issue. Reconnecting to internet and updating the system time solved this for me.

ivan-ca97, 2024-09-25:
Old issue but... I still get this and my PC has the time set 100% correctly, what could the problem be?

fhrbata, 2024-09-25:
> Old issue but... I still get this and my PC has the time set 100% correctly, what could the problem be?

Hello, if you are using the default `ninja` generator, you can try running `ninja -d explain` in the `build` directory. It should explain why each build statement is being rebuilt. This may help you understand what's happening.

ivan-ca97, 2024-09-25:
Actually... it seems like he issue was simpler than that... I placed the project in another path and it worked... I had it in a Google Drive path which started with "G:\Mi Unidad\..." (Mi Unidad is spanish for My Drive) and that space in the path seemed to be screwing everything up. After I realised that I remembered I was told not to have any spaces in the path but I didn't think it would cause an infinite loop like that, rather, I thought it would just fail