espressif/esp-idf #5976: WiFi sometimes disconnects immediately on startup, and never reconnects

jeremyherbert, 2020-10-14:
## Environment

- Module or chip used: ESP32-WROVER-B
- IDF version: v4.1
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0

- Operating System: Linux
- Using an IDE?: No
- Power Supply: External 3.3V

## Problem Description

I am having an issue where my devices will often not connect to WiFi immediately after being programmed (log is attached below). After this, no matter how many times I call `esp_wifi_connect()` in my **main** loop/thread, it will never actually connect (it does return ESP_OK though). If I reset the device, it will usually connect first time. 

However, if I change the code to instead call `esp_wifi_connect()` in the wifi event handler immediately after receiving `WIFI_EVENT_STA_DISCONNECTED`, it will reconnect fine.

Is there some restriction about which thread this function can be called in?

## Debug Logs

Log when device does not connect:
```
D (7631) esp_netif_objects: esp_netif_add_to_list 0x3ffe605c
D (7632) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (7635) nvs: nvs_open_from_partition misc 1
D (7636) nvs: nvs_get_str_or_blob log
I (7645) wifi:wifi driver task: 3ffcd298, prio:23, stack:6656, core=0
I (7646) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (7648) efuse: coding scheme 0
D (7649) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (7655) efuse: coding scheme 0
D (7655) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (7661) efuse: coding scheme 0
D (7661) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (7668) efuse: coding scheme 0
D (7668) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (7669) efuse: coding scheme 0
D (7669) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (7675) efuse: coding scheme 0
D (7675) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (7682) efuse: coding scheme 0
D (7682) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
I (7684) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (7691) efuse: coding scheme 0
D (7691) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (7697) efuse: coding scheme 0
D (7698) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (7699) efuse: coding scheme 0
D (7699) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (7706) efuse: coding scheme 0
D (7706) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (7712) efuse: coding scheme 0
D (7712) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (7718) efuse: coding scheme 0
D (7718) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (7719) efuse: coding scheme 0
D (7720) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (7726) nvs: nvs_open_from_partition nvs.net80211 1
D (7728) nvs: nvs_get opmode 1
D (7729) nvs: nvs_get_str_or_blob sta.ssid
D (7731) nvs: nvs_get_str_or_blob sta.mac
D (7733) nvs: nvs_get sta.authmode 1
D (7734) nvs: nvs_get_str_or_blob sta.pswd
D (7740) nvs: nvs_get_str_or_blob sta.pmk
D (7742) nvs: nvs_get sta.chan 1
D (7743) nvs: nvs_get auto.conn 1
D (7744) nvs: nvs_get bssid.set 1
D (7744) nvs: nvs_get_str_or_blob sta.bssid
D (7746) nvs: nvs_get sta.lis_intval 2
[... 272 lines cut ...]
I (8404) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
D (8405) nvs: nvs_set sta.chan 1 6
D (8406) nvs: nvs_set_blob sta.apinfo 700
I (8410) wifi:pm start, type: 1
D (8411) event: running post WIFI_EVENT:4 with handler 0x401b0b44 on loop 0x3ffba808
D (8413) esp_netif_handlers: esp_netif action connected with netif0x3ffe5fac from event_id=4
D (8419) esp_netif_lwip: check: remote, if=0x3ffe5fac fn=0x401a0f4c
D (8421) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffe5fac
D (8422) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (8428) esp_netif_lwip: check: remote, if=0x3ffe5fac fn=0x401a01bc
D (8429) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffe5fac
D (8435) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffe5fac
D (8436) esp_netif_lwip: if0x3ffe5fac start ip lost tmr: no need start because netif=0x3ffe6394 interval=120 ip=0
D (8443) esp_netif_lwip: starting dhcp client
D (8446) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (8451) event: running post WIFI_EVENT:4 with handler 0x400e71a4 on loop 0x3ffba808
I (8473) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (10178) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffe6394
D (10179) esp_netif_lwip: if0x3ffe5fac ip changed=1
D (10180) event: running post IP_EVENT:0 with handler 0x401b0ac8 on loop 0x3ffba808
D (10187) wifi_init_default: Got IP wifi default handler entered
D (10188) esp_netif_handlers: esp_netif action got_ip with netif0x3ffe5fac from event_id=0
I (10194) esp_netif_handlers: sta ip: 192.168.3.65, mask: 255.255.255.0, gw: 192.168.3.1
D (10196) event: running post IP_EVENT:0 with handler 0x400e71a4 on loop 0x3ffba808
```

HarveyRong-Esp, 2020-10-15:
Hi jeremyherbert, 

`I (8201) wifi:state: assoc -> init (2c0)`

Check the log, the WiFi connection failed because it has received deauth frames. 
Can you try to change the router or provide WiFi packets?

The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Maybe you can provide relevant code for `esp_wifi_connect()`?  It should be noted that after the code is released here, the code will be open source.

jeremyherbert, 2020-10-15:
Hi Harvey,

Yes, the current code is difficult to share, I am going to try to reproduce tomorrow in a new application.

Router is a Ubiquiti nanoHD, but it’s not something I can change unfortunately. I will also try to capture some packets.

HarveyRong-Esp, 2020-11-09:
Hi @jeremyherbert , Is there any update?

jeremyherbert, 2020-11-09:
Hi Harvey, I haven't been able to find the time to try to reproduce this unfortunately. I'm happy if you want to close this issue.

HarveyRong-Esp, 2020-11-10:
@jeremyherbert, OK,  please open at any time once it reappears

HarveyRong-Esp, 2020-11-17:
Hi @jeremyherbert ,
We purchased the Ubiquiti nanoHD router and tried to reproduce it locally.  Unfortunately, we were unable to reproduce it. 
If you have time, please help to try to reproduce and provide data packets, thank you very much.

ryankurte, 2020-11-17:
we're observing a _similar_ issue with `release-4.2`, though are just working on debugging from our side. we might be misusing the API, or it could be the network but, just in case this is related the debug output follows.

```
I (12909657) wifi:bcn_timout,ap_probe_send_start
I (12912167) wifi:ap_probe_send over, resett wifi status to disassoc
I (12912227) wifi:state: run -> init (c800)
I (12912227) wifi:pm stop, total sleep time: 10245962013 us / 12819866086 us
I (12912237) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (12912237) WIFI_MGR: WiFi disconnected, attempting to reconnect (attempt 0)
I (12912497) wifi:new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (12912587) wifi:state: init -> auth (b0)
I (12912607) wifi:state: auth -> assoc (0)
I (12913607) wifi:state: assoc -> init (400)
I (12913607) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (12913607) WIFI_MGR: WiFi disconnected, attempting to reconnect (attempt 1)
I (12915987) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (12916067) wifi:state: init -> auth (b0)
I (12917067) wifi:state: auth -> init (200)
I (12917087) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
```

HarveyRong-Esp, 2020-11-17:
Hi @ryankurte, See Log, it should be a connection failure caused by timeout,

`run -> init (c800)` After connecting to WiFi, the beacon frame is not received within the specified time.
`assoc -> init (400)` During the connection to WiFi, no assoc frame was received within the specified time
`auth -> init (200)` During the connection to WiFi, no auth frame was received within the specified time

Can you try to provide wireless data packets? Or provide router model and simple reproduction code for our local reproduction.

HarveyRong-Esp, 2020-12-08:
Hi @ryankurte, is this issue must reproduce? Can it be restored when something goes wrong? If it can be recovered, it may be caused by sta not receiving beacon, auth, or assoc when the environment is complicated. This situation is normal.

closed 2020-12-23 by Alvin1Zhang