espressif/esp-idf #5977: Got ESP_ERR_NVS_NOT_ENOUGH_SPACE when write zero length data with nvs_set_blob

AxelLin, 2020-10-14:
## Environment

 - Module or chip used: ESP32-WROOM-32E
 - IDF version: v4.3-dev-1197-g8bc19ba893e5
 - Build System: idf.py
 - Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
 - Operating System: Linux
 - Power Supply: USB

## Problem Description

I try to randomly write blob data with 512B/1KB/2KB/3KB strings.
nvs_set_blob(handle, key, str, strlen(str));

And use below code to clear my blob data.
nvs_set_blob(handle, key, "", 0);

It works in most time.
But I found if I repeat writing the blob data more times, it's possible to hit ESP_ERR_NVS_NOT_ENOUGH_SPACE error
when I call nvs_set_blob(handle, key, "", 0).
What make me confused is when this happen, it's ok to call nvs_set_blob(handle, key, str, strlen(str)); with short string.

Note, when this issue happen, in writeMultiPageBlob()
```
    do {
        Page& page = getCurrentPage();
        size_t tailroom = page.getVarDataTailroom();
        size_t chunkSize =0;
printf("chunkCount=%d tailroom=%d dataSize=%d\r\n", chunkCount, tailroom, dataSize);
        if (!chunkCount && tailroom < dataSize && tailroom < Page::CHUNK_MAX_SIZE/10) {
```

Above printf shows "chunkCount=0 tailroom=0 dataSize=0".

I'm wondering if there is something wrong when calling nvs_set_blob with dataSize = 0.

### Expected Behavior
IMHO, clear blob data with writing dataSize=0 should not got ESP_ERR_NVS_NOT_ENOUGH_SPACE error.
Especially when it's ok to write short string with the same key.

### Actual Behavior
Sometimes nvs_set_blob(handle, key, "", 0) returns ESP_ERR_NVS_NOT_ENOUGH_SPACE.
But it's ok to write some short string (non-zero length) with the same key.

### Steps to reproduce

Just repeat read/write NVS with variable length data (to make it fragment).

Alvin1Zhang, 2020-10-14:
Thanks for reporting, we will look into.

0xjakob, 2020-10-19:
Hi @AxelLin, sorry for the late reply! From what you tell us, it looks like a bug somewhere in the NVS system. We're investigating this now. I'm not sure whether `nvs_set_blob` was ever meant to be used with 0 length... Does deleting the blob item instead of writing a blob with length 0 work for you?

AxelLin, 2020-10-19:
Hi @0xjakob 

Honestly, I don't know if repeating create key / delete key is better or worsen to the NVS system.
Our application API assumes the empty string in a blob key means clear the stetting.
So we prefer to keep the key with empty string rather than delete it.

0xjakob, 2020-10-20:
@AxelLin I think the empty string key isn't a problem. I kind of reproduced the issue locally now. The problem seems to be with the 0 length blob. It looks like [this check](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_storage.cpp#L186) fails in some situations when the length is zero and there isn't enough space in the current page anymore. The result is the failure `ESP_ERR_NVS_NOT_ENOUGH_SPACE` instead of requesting a new page.

I'll try to fix this tomorrow.

0xjakob, 2020-10-21:
@AxelLin Disregard the first sentence of my previous comment. I think I fixed the issue, at least all our tests work. Could you apply the attached patch to see whether it fixes your problem? There's certainly a situation where a blob write with 0 length can cause trouble, as mentioned above.

[0001-NVS-fixed-bug-when-blob-data-size-0.zip](https://github.com/espressif/esp-idf/files/5415236/0001-NVS-fixed-bug-when-blob-data-size-0.zip)

AxelLin, 2020-10-22:
Hi @0xjakob 
Thanks for your prompt response.

I have tested with your patch.
After hitting ESP_ERR_NVS_NOT_ENOUGH_SPACE, it's still possible not able to set empty value.

Below is my debug print when nvs_set_blob with a long string (3422 bytes) fails.
chunkCount=0 tailroom=32 dataSize=3422
chunkCount=0 tailroom=1440 dataSize=3422
chunkCount=1 tailroom=192 dataSize=3422
chunkCount=2 tailroom=32 dataSize=3422
chunkCount=3 tailroom=0 dataSize=3422

I got ESP_ERR_NVS_NOT_ENOUGH_SPACE.
Then sometimes set empty value also fails.

I'm not sure if the patch works or not because even without the patch setting empty value is also working in most of the time.
But I notice setting small string with nvs_set_str/nvs_set_blob also fails if setting empty value fails,
this is a difference without the patch.
(Although setting integer values still works)

Below is my nvs_stats numbers when this issue happen.
nvs_stats.free_entries = 127
nvs_stats.used_entries 1007
nvs_stats.total_entries = 1134

But I notice 2 unusual things:
1) When nvs_set_blob(handle, key, "", 0); fails
   If I reboot the device and try again, then nvs_set_blob(handle, key, "", 0); works.
   I'm not sure it will  work always or just sometimes, but it looks like reboot indeed makes some difference.

2) Sometimes, the nvs key disappear if nvs_set_blob fails with ESP_ERR_NVS_NOT_ENOUGH_SPACE.
   (Just like the key was deleted, this is unexpected behavior)
    I hit this twice, but I don't have a good steps to reproduce it. ( I just randomly upload various size data to different keys). 

BTW, I'm wondering what is the recommend action when NVS system is really too fragment.

0xjakob, 2020-10-22:
@AxelLin Hi.

Setting a blob item _with a key that hasn't been written yet_ involves at least two item writes (blob entry with key and blob index entry) into a page. If I'm not mistaken, they have to be two contiguous items. Not sure whether that's the case in your situation, but if, then your observation makes sense:

> But I notice setting small string with nvs_set_str/nvs_set_blob also fails if setting empty value fails,
> this is a difference without the patch.
> (Although setting integer values still works)

For your two listed observations 1. and 2., this looks not quite right. Does this happen with or without the patch?

The recommended action if NVS becomes too fragmented is to increase the partition size. NVS is intended for storage of small parameters, even though you can store blobs and strings. If you have a large blob, it might be better to store it in a separate partition either as raw data or using a file system. Keep in mind though that the IDF filesystems may not be completely power-off tolerant. NVS is designed to be power-off tolerant (I realize the irony since you sometimes get different results after rebooting the device). You can also store a chunk of raw data in another partition and use NVS to store some meta parameters about this chunk (like length, last written date, etc.), which might help for power-off tolerance.

AxelLin, 2020-10-22:
> @AxelLin Hi.
>
> Setting a blob item _with a key that hasn't been written yet_ involves at least two item writes (blob entry with key and blob index entry) into a page. If I'm not mistaken, they have [...]

In my test, I always test setting to existing key. (I never delete a key, I may set it to different value or empty value)

> For your two listed observations 1. and 2., this looks not quite right. Does this happen with or without the patch?

Currently, all my test is with your patch and a few debug prints I added (for all test in my previous report).

> The recommended action if NVS becomes too fragmented is to increase the partition size. NVS is intended for storage of small parameters, even though you can store blobs and strings. If you have a [...]

Now I confirmed indeed some difference between reboot when hit ESP_ERR_NVS_NOT_ENOUGH_SPACE.
1. nvs_set_blob with a long string failed and return ESP_ERR_NVS_NOT_ENOUGH_SPACE
2. call nvs_set_blob(handle, key, "", 0); also fails with ESP_ERR_NVS_NOT_ENOUGH_SPACE.
Then I check the nvs_stats numbers:
nvs_stats.free_entries = 127
nvs_stats.used_entries 1007
nvs_stats.total_entries = 1134

Then I reboot, check the nvs_stats numbers again:
nvs_stats.free_entries = 231
nvs_stats.used_entries 903
nvs_stats.total_entries = 1134

The free_entreis becomes larger, I think that's why nvs_set_blob(handle, key, "", 0); works after reboot.
Regarding the free_entreis count inconsistent after reboot, that probably worth tracing.

0xjakob, 2020-10-23:
Hi @AxelLin I will continue looking into this on Monday. I have confirmed that there is the issue I fixed in the patch but haven't yet looked into your other issues.

0xjakob, 2020-10-27:
@AxelLin Is there any chance that you can send us some reproducing code? When I try myself, I never get `ESP_ERR_NVS_NOT_ENOUGH_SPACE`, neither using the host tests, nor using the target tests. I tried both with and without the patch. The test code for the target looks like this:

```c
TEST_CASE("overwrite blob with new empty blob", "[nvs_storage]")
{
    const size_t BLOB_SIZE = 3422;

    nvs_handle_t handle;
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        nvs_flash_erase();
        err = nvs_flash_init();
    }
    TEST_ESP_OK(err);

    uint8_t *blob = calloc(BLOB_SIZE, 1);
    TEST_ASSERT(blob != NULL);
    TEST_ESP_OK(nvs_open("test_ns", NVS_READWRITE, &handle));

    char BLOB_NAME [] = "blob";
    uint8_t start = 0;
    for (size_t i = 0; i < 0xFF; i++) {
        blob[0] = (uint8_t) i;
        esp_err_t result = nvs_set_blob(handle, BLOB_NAME, blob, BLOB_SIZE);
        if (result != ESP_OK) {
            break;
        }
        start = (start + 1) % 0xFF;
    }

    // overwrite the blob with an empty blob
    TEST_ESP_OK(nvs_set_blob(handle, BLOB_NAME, "", 0));

    nvs_close(handle);

    free(blob);

    nvs_flash_deinit();
}
```

AxelLin, 2020-11-21:
I don't have code snipt to test it automatically.
I have various settings with different key types, I just randomly update the value.
I test it manually by repeating update nvs values.

I think maybe increase settings (with different key types) and reduce your NVS partition
can make it easier to reproduce.

AxelLin, 2021-01-14:
> @AxelLin Disregard the first sentence of my previous comment. I think I fixed the issue, at least all our tests work. Could you apply the attached patch to see whether it fixes your problem? There's [...]

Hi,
Just wondering if any issue with above patch?
It has been there for a long period but not merged to recent esp-idf update.

0xjakob, 2021-01-15:
Hi @AxelLin,

The fix has been merged to master, the patch contains some additional changes.

The important line is here:
https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/src/nvs_storage.cpp#L192

Hope this helps,
Jakob

0xjakob, 2021-01-22:
@AxelLin Do you still have the issue described above even if you rebase onto current master?

AxelLin, 2021-01-22:
Hi @0xjakob 
My test looks lood so far, but I think it needs more time to verify.
I will close the issue if it's still does not hit the issue next week.
Thanks a lot for your support.

closed 2021-01-27 by AxelLin