espressif/esp-idf #5979: Install of esp-idf-tools-setup-2.3.exe fails under Windows 10 exit code -2

s-marley, 2020-10-14:
## Environment

- Development Kit:      N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  N/A
- IDF version (run ``git describe --tags`` to find it): 4.1
- Build System:         N/A
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System:     Windows 10 v2004
- (Windows only) environment type: N/A
- Using an IDE?: No, can't install
- Power Supply: N/A

## Problem Description

Running esp-idf-tools-setup-2.3.exe fails to complete the installation with exit code -2. I have tried with 4.1, 4.0 and one of the 3.x branches, and get the same error. I have re-downloaded and attempted again with same result. I have rebooted numerous times.

### Expected Behavior

Installer succeeds

### Actual Behavior

Installer fails with exit code -2

### Steps to reproduce

1. Run esp-idf-tools-setup-2.3.exe
2. Accept defaults for all options
3. Installer installs Python and Git, Dowloads and extracts IDF Tools 4.3
4. Towards end of install process, installer fails with exit code -2

## Debug Logs
Install log below:
```
2020-10-14 20:20:52.726   Log opened. (Time zone: UTC+01:00)
2020-10-14 20:20:52.726   Setup version: Inno Setup version 6.0.2 (u)
2020-10-14 20:20:52.726   Original Setup EXE: E:\Desktop\esp-idf-tools-setup-2.3.exe
2020-10-14 20:20:52.727   Setup command line: /SL5="$306DC,350580993,721408,E:\Desktop\esp-idf-tools-setup-2.3.exe" 
2020-10-14 20:20:52.727   Windows version: 10.0.19041  (NT platform: Yes)
2020-10-14 20:20:52.727   64-bit Windows: Yes
2020-10-14 20:20:52.727   Processor architecture: x64
2020-10-14 20:20:52.727   User privileges: None
2020-10-14 20:20:52.727   Administrative install mode: No
2020-10-14 20:20:52.727   Install mode root key: HKEY_CURRENT_USER
2020-10-14 20:20:52.728   64-bit install mode: Yes
2020-10-14 20:20:52.731   Created temporary directory: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp
2020-10-14 20:20:52.736   -- DLL function import --
2020-10-14 20:20:52.736   Function name: idpAddFile
2020-10-14 20:20:52.736   DLL name: files:idp.dll
2020-10-14 20:20:52.736   Extracting temporary file: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\idp.dll
2020-10-14 20:20:52.747   Dest DLL name: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\idp.dll
2020-10-14 20:20:52.747   Importing the DLL function.
2020-10-14 20:20:52.797   Successfully imported the DLL function. Delay loaded? No
2020-10-14 20:20:52.797   -- DLL function import --
2020-10-14 20:20:52.797   Function name: idpAddFileComp
2020-10-14 20:20:52.797   DLL name: files:idp.dll
2020-10-14 20:20:52.797   Dest DLL name: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\idp.dll
2020-10-14 20:20:52.797   Importing the DLL function.
2020-10-14 20:20:52.797   Successfully imported the DLL function. Delay loaded? No
2020-10-14 20:20:52.797   -- DLL function import --
2020-10-14 20:20:52.797   Function name: idpAddMirror
2020-10-14 20:20:52.797   DLL name: files:idp.dll
2020-10-14 20:20:52.797   Dest DLL name: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\idp.dll
2020-10-14 20:20:52.797   Importing the DLL function.
2020-10-14 20:20:52.797   Successfully imported the DLL function. Delay loaded? No
2020-10-14 20:20:52.797   -- DLL function import --
2020-10-14 20:20:52.797   Function name: idpAddFtpDir
2020-10-14 20:20:52.797   DLL name: files:idp.dll
2020-10-14 20:20:52.797   Dest DLL name: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\idp.dll
[... 359 lines cut ...]
2020-10-14 20:23:38.200   Writing uninstall key values.
2020-10-14 20:23:38.201   Detected previous administrative 64-bit install? No
2020-10-14 20:23:38.201   Detected previous administrative 32-bit install? No
2020-10-14 20:23:38.208   Installation process succeeded.
2020-10-14 20:23:38.213   Setting PATH for this process: C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\;C:\Program Files\Git\cmd\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\AWRoot\bin\common;C:\AWRoot\bin\lib;C:\Android;C:\Windows\System32;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\;C:\Users\Scott Marley\AppData\Local\Microsoft\WindowsApps;
2020-10-14 20:23:38.213   PYTHONPATH=
2020-10-14 20:23:38.213   Extracting temporary file: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\7za.exe
2020-10-14 20:23:38.264   Extracting ESP-IDF reference repository: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\7za.exe x -oC:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"
2020-10-14 20:23:38.264   Reference repository path: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\esp-idf-v4.1
2020-10-14 20:23:38.283   ProcStart failed
2020-10-14 20:23:38.283   Done, exit code=-2
2020-10-14 20:23:38.283   --------
2020-10-14 20:23:38.283   Running command: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\7za.exe x -oC:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"
                          
2020-10-14 20:23:38.283   --------
2020-10-14 20:23:38.283   Message box (OK):
                          Installation has failed with exit code -2
2020-10-14 20:23:42.012   User chose OK.
2020-10-14 20:23:42.025   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-10-14 20:23:42.967   User chose Yes.
2020-10-14 20:23:42.993   CurStepChanged raised an exception.

```

krzychb, 2020-10-14:
The installer fails when extracting zipped ESP-IDF repository:

```
2020-10-14 20:23:38.213   Extracting temporary file: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\7za.exe
2020-10-14 20:23:38.264   Extracting ESP-IDF reference repository: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\7za.exe x -oC:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"
2020-10-14 20:23:38.264   Reference repository path: C:\Users\Scott Marley\AppData\Local\Temp\is-UA59D.tmp\esp-idf-v4.1
2020-10-14 20:23:38.283   ProcStart failed
2020-10-14 20:23:38.283   Done, exit code=-2
```

Exit code 2 is a fatal error in 7zip, but no more information is provided on the reason.

The repository size is almost 500 MB. Is it possible that you do not have enough space on your drive?

s-marley, 2020-10-14:
@krzychb I don't think that's the issue, I have 50 GB available on C: at the moment. Thanks for the suggestion though. I do have 7zip installed anyway, but I presume the installer uses its own version. I can extract the repository manually, but I wouldn't know where to go from there.

igrr, 2020-10-15:
@s-marley Thanks for reporting, will prepare a fix soon.

igrr, 2020-10-15:
@s-marley Could you please try again with this installer http://dl.espressif.com/dl/esp-idf-tools-setup-2.3-pathfix.exe?

s-marley, 2020-10-15:
@igrr I'm afraid it failed again with the same error. The (hopefully) useful bit of the log is posted below, but I think it's the same as last time. If you need the full log, it's on [pastebin](https://pastebin.com/pMeXAmju).
```
2020-10-15 18:48:48.468   Installation process succeeded.
2020-10-15 18:48:48.470   Setting PATH for this process: C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\;C:\Program Files\Git\cmd\;C:\Program Files\Mozilla Firefox;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\AWRoot\bin\common;C:\AWRoot\bin\lib;C:\Android;C:\Windows\System32;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\Scott Marley\AppData\Local\Programs\Python\Python37\;C:\Users\Scott Marley\AppData\Local\Microsoft\WindowsApps;
2020-10-15 18:48:48.470   PYTHONPATH=
2020-10-15 18:48:48.470   Extracting temporary file: C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp\7za.exe
2020-10-15 18:48:48.523   Extracting ESP-IDF reference repository: C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp\7za.exe x -o"C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp" -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"
2020-10-15 18:48:48.523   Reference repository path: "C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp\esp-idf-v4.1"
2020-10-15 18:48:48.549   ProcStart failed
2020-10-15 18:48:48.549   Done, exit code=-2
2020-10-15 18:48:48.549   --------
2020-10-15 18:48:48.549   Running command: C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp\7za.exe x -o"C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp" -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"
                          
2020-10-15 18:48:48.549   --------
2020-10-15 18:48:48.549   Message box (OK):
                          Installation has failed with exit code -2
```
Thank you for looking into this for me.

igrr, 2020-10-18:
@s-marley sorry that the fix didn't work! I thought that the error was happening because the path to the temporary directory contained spaces, and wasn't correctly quoted on the command line. However it looks like 7za.exe can not start...

Could you please try the following?

1. Run the installer, wait until it fails the same way. Do not exit the installer yet.
2. Open windows explorer, go to C:\Users\Scott Marley\AppData\Local\Temp and find the most recent directory named "is-XXXX.tmp".
3. In command prompt, try to run 7za.exe from that directory. Is any error shown on this step?

s-marley, 2020-10-22:
Hi @igrr and sorry for the delay. I've done as requested, and it seems to run. I asked it to do a benchmark, just to make sure it was funtioning. Image below:
[image]
That's probably not very useful for you though!

s-marley, 2020-10-22:
Jus to add to this, I tried to run the same command as the installer does, and got this message, looks like it can't find C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip
[image]
You can see I then go to C:\Users\Scott Marley\esp\releases and esp-idf-v4.1.zip is actually there.

igrr, 2020-10-23:
It seems like in the last case you were running the older command, where the destination path was not quoted. The version of the installer with quoting fix produced a slightly different command: `C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp\7za.exe x -o"C:\Users\Scott Marley\AppData\Local\Temp\is-0B23L.tmp" -r -aoa "C:\Users\Scott Marley\esp\releases\esp-idf-v4.1.zip"` (notice double quotes after `-o` flag).

s-marley, 2020-10-23:
Apologies. I've done it again with the path-fix version, and everything seems to work as expected. I'm still getting the error on the installer though.
[image]

georgik, 2021-04-16:
Confirmed. The problem is also present in Online Installer v 2.7.
Workaround: Use Offline Installer which does not require 7z to extract data.
You can download it from: https://dl.espressif.com/dl/esp-idf/

georgik, 2021-04-16:
One other workaround based on changing environment variable TMP which is used for temporary directory:
```
mkdir C:\Temp
set TMP=C:\Temp
esp-idf-tools-setup-online-2.7.exe
```
Please not that there might be an environment variable TEMP which does not have effect on installer.

georgik, 2021-04-20:
@s-marley Problem caused by white character in Windows user's home is solved. Path to 7za.exe is now closed in quotes.

Please download updated version of installer 2.7 from: https://dl.espressif.com/dl/esp-idf/

Solved by commit: https://github.com/espressif/idf-installer/commit/cad8a7ccf3a9a0beddf70f41bd4e5d7a9069f3b3

closed 2021-06-07 by Alvin1Zhang