espressif/esp-idf #5981: ULP '.fill' feature exhibits unexpected behaviour

istokm, 2020-10-15:
- Development Kit:      [FireBettle ESP32]
- Kit version
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-1197-g8bc19ba89
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (VisualStudio Code /w ESP-IDF extension)]
- Power Supply:         [USB]

## Problem Description

I don't know if there is a correct way of measuring the size of a ULP code section, so I just filled it up until it overflowed, and used that as a measuring tool:

`.set event_queue_size, 128`  
leads to:  
`esp32ulp-elf-ld: region 'ram' overflowed by 28 bytes`  

Compared to setting the queue size to 129:  
`.set event_queue_size, 129`  
leads to:  
`esp32ulp-elf-ld: region 'ram' overflowed by 32 bytes`

The constant is used like this:
```
.global event_queue
event_queue:
.fill event_queue_size
```

Leading me to believe that `.fill 1` in this case would mean 4 bytes - 32bits - 1 word in the ULP.
Though from my experience, initializing an array with `.fill 16`, and trying to write data past it's 4th index leads to that data being corrupted when read by the main CPU.

### Expected Behavior

`.fill 16` initializes an array with 64 bytes (16 words).

### Actual Behavior

`.fill 16` initializes an array with 16 bytes (4 words).

I can make a minimal example of this issue if needed, I first want to confirm that this is not intended behaviour and I'm just misunderstanding something - as `.fill` **is not actually documented anywhere** - so If I am, then just take this issue as a request to document this ULP feature better.

A way to fix this is also by using something I found in the `extensa_vectors.S`, `.fill XX, 4, 0`. Which leads to the same error messages, 1 being 4 bytes, but the corruption doesn't occur, so I do believe it actually allocates the correct number of words.

jcmvbkbc, 2020-10-15:
> .fill is not actually documented anywhere

It's the assembler feature and as such is documented in the assembler manual: [see](https://sourceware.org/binutils/docs/as/Fill.html#Fill).

istokm, 2020-10-15:
Thanks for the quick response! That makes sense, so I assume `.fill 1, 4, 0` would mean allocate 1x4bytes, but previously it was 1x1byte. It's still weird how the memory consumption increases by 4, even though it should only increase by 1 (for now I will assume padding), but atleast now I understand why the memory did not get allocated.

istokm, 2021-02-04:
If anyone has insight on why `.fill 1` uses up 4 bytes of memory in the program instead of 1, feel free to chime in. For now, as this is not an actual issue, I'm closing it.

closed 2021-02-04 by istokm