espressif/esp-idf #5982: ssl_mutual_auth not working with SPIFFS

MihirShahLNM, 2020-10-15:
- Development Kit:      [ESP32-DevKitC]
- Kit version (v1]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D]
- IDF version:  v4.3-dev-771-gc77c4ccf6
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes.  VS Code]
- Power Supply:         [USB]

## Problem Description

I am using ``esp-idf \ examples \ protocols \ mqtt \ ssl_mutual_auth`` to connect to AWS-IoT-Core.
When I use EMBEDDED certificates, it works very good.
But when I use SPIFFS files for the certificates it does not work.
I tried reading the certificates into a variable & then assigning it to the configuration structure.

###Actual code with configuration structure is as follows...

``
    size_t len_root_ca = 0, len_device_certificate = 0, len_device_private_key = 0;
    len_root_ca = getFileSize((const char *)ROOT_CA_PATH);
    len_device_certificate = getFileSize((const char *)DEVICE_CERTIFICATE_PATH);
    len_device_private_key = getFileSize((const char *)DEVICE_PRIVATE_KEY_PATH);
    char root_ca[len_root_ca];
    char device_certificate[len_device_certificate];
    char device_private_key[len_device_private_key];

    if (load_file_into_buffer("storage", (const char *)ROOT_CA_PATH, (char *)root_ca, len_root_ca) != ESP_OK)
    {
        TAG = (char *)oldTAG;
        return;
    }

    if (load_file_into_buffer("storage", (const char *)DEVICE_CERTIFICATE_PATH, (char *)device_certificate, len_device_certificate) != ESP_OK)
    {
        TAG = (char *)oldTAG;
        return;
    }

    if (load_file_into_buffer("storage", (const char *)DEVICE_PRIVATE_KEY_PATH, (char *)device_private_key, len_device_private_key) != ESP_OK)
    {
        TAG = (char *)oldTAG;
        return;
    }

    const esp_mqtt_client_config_t mqtt_cfg = {
        .host = HostAddress,      /*!< "a2vbvxlpw8zcql-ats.iot.us-west-2.amazonaws.com", */
        .port = AWS_MQTT_PORT,    /*!< 8883, */
        .client_id = "LNM_DSM4", 
        .disable_auto_reconnect = false,
        .keepalive = 120,
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)root_ca,
        .cert_len = len_root_ca,
        .client_cert_pem = (const char *)device_certificate,
        .client_cert_len = len_device_certificate,
        .client_key_pem = (const char *)device_private_key,
        .client_key_len = len_device_private_key,
        .transport = MQTT_TRANSPORT_OVER_SSL,
    };
``

I tried checking the content of variables, along with their length & it matches with EMBEDDED version. Each variable is NULL ( '\0' ) terminated.

###The error is as follows...
``
I (34715) MQTTS: Ack Other event id:7
E (35045) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180
E (35045) esp-tls-mbedtls: Failed to set client configurations
E (35045) esp-tls: create_ssl_handle failed
E (35055) esp-tls: Failed to open new connection
E (35055) TRANS_SSL: Failed to open a new connection
E (35065) MQTT_CLIENT: Error transport connect
I (35065) MQTTS: Ack MQTT_EVENT_ERROR
I (35075) MQTTS: Ack MQTT_EVENT_DISCONNECTED
``

Please help me resolve this.

Alvin1Zhang, 2020-10-16:
Thanks for reporting and letting us know, we will look into.

david-cermak, 2020-10-22:
Hi @MihirShahLNM 

> I tried reading the certificates into a variable & then assigning it to the configuration structure.

Are these variables still valid when performing the SSL handshake? Please check this note about the underlying API: https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/include/esp_transport_ssl.h#L35-L36 that mqtt library uses. It looks from your snippet that the variables are created on stack, might it be possible that the stack unwinds before connecting? I'd suggest allocating on heap and freeing once mqtt closes, such as:
`char root_ca[len_root_ca];`  -> `char* root_ca = malloc(len_root_ca);`, etc...

MihirShahLNM, 2020-11-01:
Thanks. It was something I did not try.

I was creating the variables after reading the size of the certificate-string. Now I have created global variables with a fixed size of 2048.

```
uint8_t ptr_root_ca[2048];
uint8_t ptr_device_certificate[2048];
uint8_t ptr_device_private_key[2048];
```

But the results are the same !   I tried it with 'char' declaration as well, but no success !!

MihirShahLNM, 2020-11-01:
I tried using the global variables.  I also removed specifying the length (.cert_len = len_root_ca, .client_cert_len = len_device_certificate, .client_key_len = len_device_private_key,).

And now it's working. Thanks @david-cermak . Thanks a lot.

```
    const esp_mqtt_client_config_t mqtt_cfg = {
        .host = HostAddress,      /*!< "a2vbvxlpw8zcql-ats.iot.us-west-2.amazonaws.com", */
        .port = AWS_MQTT_PORT,    /*!< 8883, */
        .client_id = "LNM_DSM4_", /*!< default client id is ``ESP32_%CHIPID%`` where %CHIPID% are last 3 bytes of MAC address in hex format */
        .disable_auto_reconnect = false,
        .keepalive = 120,
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)root_ca,
        .client_cert_pem = (const char *)device_certificate,
        .client_key_pem = (const char *)device_private_key,
        .transport = MQTT_TRANSPORT_OVER_SSL,
    };

```

AdityaHPatwardhan, 2020-12-15:
Hi @MihirShahLNM, if the problem is solved can you please close the issue ? Thanks.

closed 2020-12-15 by MihirShahLNM