espressif/esp-idf #5983: TG0WDT_SYS_RESET upon waking up from deepsleep if FreeRTOS unicore is enabled

istokm, 2020-10-15:
- Development Kit:      [FireBettle ESP32]
- Kit version
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-1197-g8bc19ba89
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (VisualStudio Code /w ESP-IDF extension)]
- Power Supply:         [USB]

## Problem Description

Enabling 'Run FreeRTOS only of first core` causes a `TG0WDT_SYS_RESET` upon waking up from deepsleep.
Tested on a timer wakeup, aswell as ULP wakeup.

### Steps to reproduce

1. Enable `Run FreeRTOS only of first core`

### Code to reproduce this issue

```cpp
#include <esp_sleep.h>

void app_main(void) {
	esp_deep_sleep(3000 * 1000);
}

```

I've also created a repo with everything needed to compile and flash this [here](https://github.com/istokm/esp-idf-wdt-on-wakeup).

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7040
ho 0 tail 12 room 4
load:0x40078000,len:14320
load:0x40080400,len:3484 
0x40080400: _init at ??:?

entry 0x40080634
I (30) boot: ESP-IDF v4.3-dev-1197-g8bc19ba89 2nd stage bootloader  
I (30) boot: compile time 21:19:28
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0    
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05928 ( 22824) map
I (123) esp_image: segment 1: paddr=0x00015950 vaddr=0x3ffb0000 size=0x01a48 (  6728) load
I (126) esp_image: segment 2: paddr=0x000173a0 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at D:/Projects/ESP32-IDF/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (133) esp_image: segment 3: paddr=0x000177ac vaddr=0x40080404 size=0x0886c ( 34924) load
I (156) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x12d94 ( 77204) map
0x400d0020: _stext at ??:?

I (186) esp_image: segment 5: paddr=0x00032dbc vaddr=0x40088c70 size=0x01090 (  4240) load
0x40088c70: rtc_sleep_init at D:/Projects/ESP32-IDF/esp-idf/components/soc/src/esp32/rtc_sleep.c:154

I (188) esp_image: segment 6: paddr=0x00033e54 vaddr=0x400c0000 size=0x00064 (   100) load
I (198) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (216) cpu_start: cpu freq: 160
I (216) cpu_start: Pro cpu up.
I (216) cpu_start: Single core mode
I (228) cpu_start: Pro cpu start user code
I (228) cpu_start: Application information:
I (228) cpu_start: Project name:     sandbox
I (233) cpu_start: App version:      b7d9811-dirty
I (238) cpu_start: Compile time:     Oct 15 2020 21:24:09
I (245) cpu_start: ELF file SHA256:  3b08505516b35916...
I (251) cpu_start: ESP-IDF:          v4.3-dev-1197-g8bc19ba89
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FF80064 len 00001F9C (7 KiB): RTCRAM
I (270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (277) heap_init: At 3FFB2190 len 0002DE70 (183 KiB): DRAM
I (283) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (289) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (295) heap_init: At 40089D00 len 00016300 (88 KiB): IRAM
I (302) spi_flash: detected chip: gd
I (306) spi_flash: flash io: dio
W (310) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (323) cpu_start: Starting scheduler on PRO CPU.
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7040
ho 0 tail 12 room 4
load:0x40078000,len:14320
load:0x40080400,len:3484
```

negativekelvin, 2020-10-16:
The problem is actually `flash read err, 1000` which implies the flash chip is slow to wake up, there is a setting for that.  Are you saying it never happens if you enter deepsleep from dual core mode?

Alvin1Zhang, 2020-10-16:
Thanks for reporting, we will look into.

istokm, 2020-10-16:
I guess you're right, the flash chip is the cause, the reset is just a symptom.
I cannot say that it **never** happens but I've never seen it happen, and it immediately dissapears when unicore mode is turned off. For more clarification, here is debug output from dual code & unicore without any other modifications (I've also turned all debug to error, so that the back-to-back resets are more easily visible):
https://gist.github.com/istokm/d50bf0bfe005be070c4cd1cafe5245ca

I've also tested this on multiple ESP32 boards I have next to me, all exhibit this behaviour.

igrr, 2020-10-16:
Hi @istokm, could you clarify what is the "ESP32 Devkit v1" board? Is it https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards-previous.html#esp32-core-board-v2-esp32-devkitc, or something else? A photo/link would help.

Also, could you please run `esptool.py --port YOUR_PORT flash_id` for all the boards where you observe failures? In particular, we need the "Manufacturer" and "Device" lines. Thanks.

istokm, 2020-10-16:
> could you clarify what is the "ESP32 Devkit v1" board?

It's the DOIT ESP32 DevKit v1 - https://docs.zerynth.com/latest/reference/boards/doit_esp32/docs/

> Also, could you please run `esptool.py --port YOUR_PORT flash_id` for all the boards where you observe failures?

**Firebeetle ESP32:**
```
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz      
MAC: c8:2b:96:9a:4f:0c
Uploading stub...     
Running stub...
Stub running...
Manufacturer: c8
Device: 4018
Detected flash size: 16MB
```

**DFRobot Beetle ESP32:**
```
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:98:5a:80
Uploading stub...     
Running stub...
Stub running...
Manufacturer: c8
Device: 4018
Detected flash size: 16MB
```

**DOIT ESP32 DevKit v1:**
```
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 0)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 30:ae:a4:02:f5:4c
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
```

**Sparkfun ESP32 Thing plus:**
```
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:66:eb:ac
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4018
Detected flash size: 16MB
```

And just for good measure, here is the only device I own where I do not observe this failure - **Adafruit ESP32 Feather**:
```
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:7d:6f:94
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
```

I guess the only thing it sets it apart is indeed the manufacturer number. Care to elaborate what that means exactly?
Thanks!

igrr, 2020-10-16:
@istokm This issue was fixed on the master branch in https://github.com/espressif/esp-idf/commit/ff8d05466e83049bca44e1584891bfafc8c46cd7. If you prefer to stay on your current commit, you can disable `CONFIG_ESP32_RTCDATA_IN_FAST_MEM` option in menuconfig as a workaround.

istokm, 2020-10-16:
Pulling master and recompiling indeed fixed it, thanks!

closed 2020-10-16 by istokm

negativekelvin, 2020-10-17:
So basically without the fix the default wake stub didn't run after deepsleep and `ESP32_DEEP_SLEEP_WAKEUP_DELAY` was skipped leading to `flash read err, 1000`