espressif/esp-idf #5984: Android DNS-SD does not find ESP32 service multicast (mDNS)

baldhead69, 2020-10-16:
Hi,

I am trying to implement a android APP using android sdk .

I am using mDNS lib on esp32 to send your service/ip in the local network. 

In android app i want to extract the Esp32 ip for the app to connect with esp32.

Esp32 are the server and android app are the client.

I am using this example to implement DNS-SD on android app: 
"https://developer.android.com/training/connect-devices-wirelessly/nsd#discover". 
Only the part for "Discover Services on the Network".

Using wireshark i can see that esp32 are sending the mDNS packets, but i dont know if i configured mDNS correctly.

In android app the Service discovery started but return nothing.

The play store app "Service Browser" dont encounter esp32.
The play store app ".local Finder" encounter esp32, if i select ipv4 and put the esp32 "input local name", returning esp32 ip.

Notes:
* ESP-IDF version: v4.2-dev-1235-g71dc5eb27
* Esp32 Station mode.

Images:

[image]
[image]

baldhead69, 2020-10-18:
Hi,

The dns-sd dont work on android sdk emulator( virtual machine phone ).

The dns-sd work on physical cell phone connected(by usb cable) on android sdk Debug.

gabsuren, 2020-11-12:
Hello,
Thank you for your email.
Regarding Esp32 part you have configured mDNS library correctly, the code below is correct
```
static void start_mdns_service() {
      char* hostname = "Quarto-Rangel";
      ESP_ERROR_CHECK(mdns_init());
      ESP_ERROR_CHECK(mdns_hostname_set(hostname));
      mdns_instance_name_set("EXAMPLE_MDNS_INSTANCE");
      mdns_txt_item_t serviceTxtData[3] = {
          {"board","{esp32}"},
          {"u","user"},
          {"p","password"}
      };
     ESP_ERROR_CHECK(mdns_service_add("ESP32-WebSocket-Server", "_wss", "_tcp", 443, serviceTxtData, 3));
}
```
You can test it via running
`dns-sd -Gv4v6 Quarto-Rangel.local`
 or if you want to discover services 
`dns-sd -L "EXAMPLE_MDNS_INSTANCE" _wss._tcp local`.
ESP32 mDNS library is Apple Certified library and the problem can be related with android configuration or library, which is not in our domain.

Best regards,
Suren Gabrielyan

closed 2021-08-23 by Alvin1Zhang