espressif/esp-idf #5985: esp 32s2 download helloworld.bin failed

simzhangbest, 2020-10-16:
[image]

when I finish build hello world then flash bin file with COM but failed. how does it mean?

chegewara, 2020-10-16:
`idf.py set-target esp32s2`

projectgus, 2020-10-16:
Hi @simzhangbest,

The ESP32-S2 doesn't seem to be resetting into download mode.

* If your board has a "BOOT" button try holding it down while flashing.
* If there is a "RST" button try pressing and releasing it while also holding down the BOOT button, then run the flash step.
* If that isn't possible, try finding a way to connect GPIO0 to Ground.

If none of this is working, please provide some more details about your hardware setup.

PS @chegewara I think the target is set correctly, it says "esp32s2" in the screenshot.

chegewara, 2020-10-16:
Ah, thats may be true. I got fooled by invalid head of packet error. Sorry.

projectgus, 2021-01-04:
@simzhangbest Did you manage to get your board flashing OK?

Alvin1Zhang, 2021-01-12:
@simzhangbest Thanks for reporting, would you please help share if any updates for this issue? Thanks.

projectgus, 2021-02-09:
Closing due to lack of updates.

closed 2021-02-09 by projectgus