espressif/esp-idf #5986: modbus master 是否可以支持传入的参数来设置停止位的功能

zjykymt, 2020-10-16:
components\freemodbus\port\portserial_m.c：xMBMasterPortSerialInit() 中串口的参数配置项停止位是默认为1的，是否可以支持传入参数来设置呢？
>uart_config_t xUartConfig = 
{
        .baud_rate = ulBaudRate,
        .data_bits = ucData,
        .parity = ucParity,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .rx_flow_ctrl_thresh = 2,
        .source_clk = UART_SCLK_APB,
    };

alisitsyn, 2020-10-20:
Hello zjykymt,

The extra stop bit can be a useful way to add a little extra receive processing time, especially at high baud rates and/or using software driven UART, where time is required to process the received byte. However the devices today have hardware driven UARTs and are fast enough, work at even higher baud rates without the need of an extra stop bit. The one stop bit configuration is usually enough for most of device UARTs and do not cause any issues. Do you have any issues communication between devices in your infrastructure due to fixed stop bit config?

If you have very strong opinion about two stop bit configuration it can be added to Modbus configuration and propagated to Modbus serial code initialization function.

zjykymt, 2020-10-21:
Hi alisitsyn, thanks for your reply.

Under the test there is no problem with the communication at 115200bps.

zjykymt, 2020-10-21:
In addition, I want to describe another modbus communication problem that I tested here.

ESP-IDF version: 4.1

I have configured the following two configurations: **Place UART ISR function into IRAM** or **Place timer interrupt handler into IRAM**
[image]

But when the program calls **esp_partition_write** or **esp_partition_erase_range** function, 
the following two read errors will still occur when the Master modbus command is sent at this time：

> MB_CONTROLLER_MASTER: mbc_master_send_request(131): SERIAL master send request failure error=(0x107) (ESP_ERR_TIMEOUT).<br>
> MB_CONTROLLER_MASTER: mbc_master_send_request(131): SERIAL master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).

alisitsyn, 2020-10-23:
@zjykymt,

What are the priority of the tasks which doing the partition erase and write?
I would recommend to change the priority of modbus tasks (CONFIG_FMB_PORT_TASK_PRIO) to be higher.
The modbus master uses polling approach to process input packets and reads FIFO byte by byte. 
This may cause the errors you showed. In this case possible solution is just retry the modbus write/read request.

This is known issue related to coexistence of Wi-Fi and flash functionality with modbus.
The fixes for the v4.1 as well for other branches are ready but not merged yet. You could test the patch below to apply the fixes. The modbus v4.1 with these patches works significantly better and allows avoiding most of the problems but reading errors are still possible in your application in some conditions (priority of the tasks which do intensive work related to Wi-Fi and flash access). In this case the only retry will help.
Please disable the CONFIG_MB_TIMER_PORT_ENABLED option after applying patch to allow longer delays during modbus packet processing due to periodical suspend of its tasks.

[freemodbus_backport_latest_fixes_v41.zip](https://github.com/espressif/esp-idf/files/5428126/freemodbus_backport_latest_fixes_v41.zip)

Please let me know the results. Thanks.

zjykymt, 2020-10-28:
Hi alisitsyn，

I use xTaskCreate to create application tasks, all tasks have a priority of 2, and the modbus serial task priority is 10.
[image]

After I apply the patch and set CONFIG_MB_TIMER_PORT_ENABLED to disable, the modbus read/write errors still occur but **the chances have been reduced**(ESP_ERR_TIMEOUT error no longer appears).

> After **esp_partition_erase_range**: 
> E (381754) MB_PORT_COMMON: vMBMasterRunResRelease(172): vMBMasterRunResRelease: resource release failure.

> After **esp_partition_write**: 
> E (637180) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).

The general logic of my code is: tell the modem to use HTTP GET to obtain a file of 1M size. After the modem is downloaded, use UART (115200bps) to transmit to the ESP32, and every 256 bytes received by the ESP32 are stored in the flash (of course, before downloading the flash has been erased). While downloading, read 6 consecutive modbus times as a complete implementation, and execute a complete implementation every 5 seconds

This patch seems to have some optimization effects, but I don’t quite understand that MODBUS_UART_ISR runs in RAM. 
The uart received FIFO is also in RAM, will the flash operation cause the MODBUS_UART_ISR insertion of the FIFO to fail?

Due to the development plan of my own project, I am sorry that I can only simply test and provide not very specific information.
When the development of my project reaches a certain level, I will try to understand the reasons for the above modbus uart tx/rx and flash r/w/erase errors in ESP-IDF.

Thanks.

alisitsyn, 2020-10-29:
Hi @zjykymt,

Thank you for thest results. I will investigate the issues related to using esp_partition_erase_range(). I guess it is due to constraints of current Modbus architecture. I am trying to reproduce and check solution for this. This will take some time.  As a workaround I can propose to do retries of modbus read/write requests if the error returned:

```
err == ESP_ERR_TIMEOUT;
for (int retry = 0; (retry < MB_MAX_RETRY) && (err != ESP_OK); retry++) {
    err = mbc_master_get_parameter(cid, (char*)param_descriptor->param_key, 
                                                    (uint8_t*)&value, &type);
}
```

JavierSVE, 2020-12-11:
I have the same problem with the freemodbus master component.
My application consists of a web server via a wifi access point with an embedded web page that allows updating the esp32 firmware via OTA, and a connection with a slave device via modbus.
For the OTA update firmware I use the native APIs provided by app_update component (esp_ota_write).
To avoid the error=(0x108) (ESP_ERR_INVALID_RESPONSE) during the firmware update I deactivate the communications via modbus and when the update is completed I perform a software reset (esp_restart()).
My problem is that after resetting the esp32 the modbus component is corrupted and all the requests fail with the "E (637180) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE)".
The only solution at the moment is to try a hardware reset with the EN pin.
I am developing the application with version 4.2 of the esp-idf.
Does alisitsyn have a freemodbus component patch for this version?
The esp_restart function does not reset the peripherals (except for WiFi, BT, UART0, SPI1, and legacy timers) but my modbus connection use the UART2.
Is there a software reset that does reset all the peripherals?

alisitsyn, 2020-12-11:
@JavierSVE,

The modbus actually was not designed to work with OTA update and there are things that might prevent to work properly. But in your case the modbus is restarted after update. It is interesting case and need more investigation to fix this issue. I do have the patch for v4.2 version to fix most of known issues. Please apply patch below to check if it fixes the issue.

[0001_backport_latest_fixes_v42.diff](https://github.com/espressif/esp-idf/files/5680191/0001_backport_latest_fixes_v42.diff.log)

I will try to reproduce your issue in my environment and will connect you then. Would appreciate if you can send the example to reproduce.
Thanks.

JavierSVE, 2020-12-14:
alisitsyn thank you very much for your quick response.
To reproduce this problem you can use the versamodule's example "ESP-OTA-Webserver" and add a task to it with a modbus master that interrogates a slave vía UART2:
[https://github.com/versamodule/ESP32-OTA-Webserver](url)
Now I'm going to test the patch for version 4.2 of esp-if.

JavierSVE, 2020-12-14:
I have applied the patch provided by alisitsyn to esp-idf v4.2 and the problem of the modbus master after the esp_restart continues.
This is the trace log.
First the ESP32 start from the ota_0 (at offset 0x10000). All run ok. So I run a firmware update from the embedded web and the esp_restart runs.
Now esp32 starts from partition ota_1 (at offset 0x380000) but when the vMasterTask task starts to interrogate the slave via modbus, error=(0x108) (ESP_ERR_INVALID_RESPONSE) appears forever.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7132
load:0x40078000,len:13696
load:0x40080400,len:4000
entry 0x40080688
I (28) boot: ESP-IDF v4.2-dev-1995-g636b964c8-dirty 2nd stage bootloader
I (28) boot: compile time 10:37:10
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 00370000
I (100) boot:  4 ota_1            OTA app          00 11 00380000 00370000
I (107) boot:  5 storage          Unknown data     01 82 006f0000 00100000
I (115) boot: End of partition table
I (119) boot_comm: chip revision: 1, min. application chip revision: 0
I (126) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3fe9c (261788) map
I (235) esp_image: segment 1: paddr=0x0004fec4 vaddr=0x3ffb0000 size=0x00154 (   340) load
I (236) esp_image: segment 2: paddr=0x00050020 vaddr=0x400d0020 size=0x9e66c (648812) map
I (489) esp_image: segment 3: paddr=0x000ee694 vaddr=0x3ffb0154 size=0x03d58 ( 15704) load
I (496) esp_image: segment 4: paddr=0x000f23f4 vaddr=0x40080000 size=0x00404 (  1028) load
I (496) esp_image: segment 5: paddr=0x000f2800 vaddr=0x40080404 size=0x17ea8 ( 97960) load
I (560) boot: Loaded app from partition at offset 0x10000
I (560) boot: Disabling RNG early entropy source...
I (560) cpu_start: Pro cpu up.
I (564) cpu_start: Application information:
I (569) cpu_start: Project name:     SAF840DU_WS
I (574) cpu_start: App version:      e4163eb-dirty
I (580) cpu_start: Compile time:     Dec 14 2020 10:59:25
I (586) cpu_start: ELF file SHA256:  93a8fe4335f060bf...
I (592) cpu_start: ESP-IDF:          v4.2-dev-1995-g636b964c8-dirty
I (599) cpu_start: Starting app cpu, entry point is 0x40081a94
I (0) cpu_start: App cpu up.
I (609) heap_init: Initializing. RAM available for dynamic allocation:
I (616) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (622) heap_init: At 3FFBAA20 len 000255E0 (149 KiB): DRAM
I (628) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (635) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (641) heap_init: At 400982AC len 00007D54 (31 KiB): IRAM
I (647) cpu_start: Pro cpu start user code
I (665) spi_flash: detected chip: gd
I (666) spi_flash: flash io: dio
I (666) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (729) MAIN: Press the switch to run...
I (731) uart: queue free spaces: 20

esp32 (I (1805) wifi_setup: starting tcpip adapter
/spiffs) > I (1807) wifi_setup: setting gateway IP
I (1808) wifi_setup: starting DHCPS adapter
I (1810) wifi_setup: starting event loop
I (1815) wifi_setup: initializing WiFi
[... 212 lines cut ...]
E (8866) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (8868) MASTER: ComandRequest (0x1) an invalid response from slave.
E (8949) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (8950) MASTER: ComandRequest (0x3) an invalid response from slave.
E (9214) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (9215) MASTER: ComandRequest (0x3) an invalid response from slave.
E (9349) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (9350) MASTER: ComandRequest (0x4) an invalid response from slave.
E (9619) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (9620) MASTER: ComandRequest (0x4) an invalid response from slave.
E (10816) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (10818) MASTER: ComandRequest (0x1) an invalid response from slave.
E (10890) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (10891) MASTER: ComandRequest (0x3) an invalid response from slave.
E (11154) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (11155) MASTER: ComandRequest (0x3) an invalid response from slave.
[... cut ...] PONSE).
E (11290) MASTER: ComandRequest (0x4) an invalid response from slave.
E (11556) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (11557) MASTER: ComandRequest (0x4) an invalid response from slave.
E (12853) MB_CONTROLLER_MASTER: mbc_master_send_request(131): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).
E (12854) MASTER: ComandRequest (0x1) an invalid response from slave.

# For ever ....
`

JavierSVE, 2020-12-14:
After applying the second proposed patch that corrects some fixes for esp_restart() to address the incorrect reset of UART1/2 tx/rx fifo, now everything works correctly.

Thank you very much.

alisitsyn, 2020-12-14:
@JavierSVE,

I was able to reproduce the issue with my simple modbus master example and OTA update through web with the result almost the same as yours (had timeout modbus errors instead of invalid response). Then it was possible to fix the issue applying patch below. The UART2 is able to work correctly after this and I do not have issues.

[0002_driver_fix_uart_module_reset_issue.diff](https://github.com/espressif/esp-idf/files/5688973/0002_driver_fix_uart_module_reset_issue.diff.log)

closed 2021-06-08 by zjykymt