espressif/esp-idf #5987: sd card read issue

chegewara, 2020-10-16:
Hi,
i am having some issues while reading bigger files from sd card on esp32-S2. Ive been trying to find similar issues, but i dont see any similar. Full description of my situation is:
- file server example updated a bit to use SD card instead of spiffs:
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving
- reading files tree and small files works good (around 600kB), also reading bigger files (over 1 MB) sometimes works,
- here is error i am usually getting:
```
E (58198924) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108
E (58198924) diskio_sdmmc: sdmmc_read_blocks failed (264)
```
Error: `ESP_ERR_INVALID_RESPONSE (0x108): Received response was invalid`
- commit i am using: 7d75213674b4572f90c68162ad6fe9b16dae65ad

Here is log just before error:
```
V (357577) bus_lock: dev 5 released.[0m
V (357580) sdspi_transaction: r1 = 0x00 hw_cmd.r[0]=0xff5f6d64[0m
V (357586) sdmmc_cmd: cmd response 00000000 00000000 00000000 00000000 err=0x0 state=0[0m
V (357594) sdmmc_cmd: sending cmd slot=2 op=17 arg=234e8 flags=1c50 data=0x3ffdf92c blklen=512 datalen=512 timeout=1000[0m
V (357605) sdspi_host: sdspi_host_start_command: slot=2, CMD17, arg=0x000234e8 flags=0x5, data=0x3ffdf92c, data_size=512 crc=0x4c[0m
V bus_lock: dev 5 acquired.
D (357619) spi_master: device5 locked the bus[0m
V (357624) spi_master: polling trans[0m
V (357627) spi_master: polling trans done[0m
V (357631) spi_master: polling trans[0m
V (357635) spi_master: polling trans done[0m
...
V (358628) spi_master: polling trans[0m
V (358632) spi_master: polling trans done[0m
D (358636) sdspi_host: poll_data_token: timeout[0m
V (358640) spi_master: polling trans[0m
V (358644) spi_master: polling trans done[0m
D (358648) spi_master: device5 release bus[0m
V (358652) bus_lock: dev 5 released.[0m
D (358656) sdspi_host: sdspi_host_start_command: cmd=17 error=0x107[0m
D (358662) sdmmc_cmd: cmd=17, sdmmc_req_run returned 0x107[0m
[0;31mE (358668) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107[0m
D (358675) sdmmc_cmd: sdmmc_read_sectors: error 0x107 writing block 144616+0[0m
[0;31mE (358682) diskio_sdmmc: sdmmc_read_blocks failed (263)[0m
D (358688) vfs_fat: vfs_fat_read: fresult=1[0m
```

[sdtest.txt](https://github.com/espressif/esp-idf/files/5388916/sdtest.txt)

closed 2021-04-26 by chegewara

adokitkat, 2021-10-07:
Thank you for the quick reply. 

I am going to reopen this issue then and try to reproduce it. If it was fixed by now then I will close it again.

ShenRen, 2022-06-05:
I have encountered the same problem, using ESP32S2 chip, there is no problem when using SDSPI alone, but after WIFI is turned on, there will be read and write error of SDMMC.
I doubt that there is a conflict between ESP32S2 SPI and WIFI. The same problem has occurred in IDFV4.3.2, IDFV4.4 and IDFV4.4.1.

Icarus113, 2022-08-24:
@adokitkat Hi, may I know if you reproduce it or not? Or maybe it would be good if you open a new issue. If it's a new issue, then you'll need to fill a form about issue backgrounds, reproduce steps, etc.

chegewara, 2023-12-07:
Hi,
probably i did, but it was so long time ago so i cant help with the issue.

lbxhacker, 2023-12-30:
I also encountered this problem, and the problem did occur when initializing WiFi-related resources