espressif/esp-idf #5988: Custom UUID for services

Zeni241, 2020-10-16:
In the “gatt_server_service_table”example  ( https://github.com/espressif/esp-‎idf/blob/release/v4.1/examples/bluetooth/bluedroid/ble/gatt_server_service_table/main/gatts_table_creat_demo.c ) **I want to create service with custom service 128bit UUID.**  <br/><br/>  I generated a 128bit‎ UUID ‎online and used it as:‎  
`static uint8_t service_uuid[16]={0x36, 0xa2, 0x14, 0xe0, 0x0f, 0x68, 0x11, 0xeb, 0xad, 0xc1, 0x02, 0x42, ‎‎0xac, 0x12, 0x00, 0x02,};`
    But the service UUID I see in client is “000042ac-0000-1000-8000-00805f9b34fb”; while I am expecting ‎‎“36a214e0-0f68-11eb-adc1-0242ac120002”. ‎

When creating the service in table this example use **“primary_service_uuid”**  which is defined as:‎
`static const uint16_t primary_service_uuid         = ESP_GATT_UUID_PRI_SERVICE;‎`

_‎[IDX_SVC]        =‎
‎    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&**primary_service_uuid**, ‎ESP_GATT_PERM_READ,‎
‎      sizeof(uint16_t), sizeof(GATTS_SERVICE_UUID_TEST), (uint8_t *)&GATTS_SERVICE_UUID_TEST}},‎_

**What is this ESP_GATT_UUID_PRI_SERVICE and why can’t I use “service_uuid” in its place?‎**

chegewara, 2020-10-16:
https://github.com/espressif/esp-idf/blob/cf056a7d0b90261923b8207f21dc270313b67456/examples/bluetooth/bluedroid/ble/gatt_client/tutorial/Gatt_Client_Example_Walkthrough.md#discovering-services

Zeni241, 2020-10-16:
thanks a lot for this quick response. I will comment after going through and digesting it thoroughly. BTW I am using  Web Bluetooth API for browser as a client.

Zeni241, 2020-10-16:
Dear chegewara
The question still remains. If I define service uuid like:‎
`static uint8_t service_uuid[16]={0x36, 0xa2, 0x14, 0xe0, 0x0f, 0x68, 0x11, 0xeb, 0xad, 0xc1, 0x02, 0x42, ‎‎0xac, 0x12, 0x00, 0x02,};` // (36a214e0-0f68-11eb-adc1-0242ac120002)‎
and `static const uint16_t GATTS_SERVICE_UUID_TEST = 0x42AC;‎`
Then my advertised Service UUID should be:  **020042ac-4202-c1ad-eb11-680fe014a236** but it is **‎‎000042ac-0000-1000-8000-00805f9b34fb** which is based on custom BLE service base UUID.‎
This is because of this line ‎(`uint8_t *)&primary_service_uuid‎` in code:‎
**‎[IDX_SVC] =‎
‎            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ‎ESP_GATT_PERM_READ, sizeof(uint16_t), sizeof(GATTS_SERVICE_UUID_TEST), (uint8_t ‎‎*)&GATTS_SERVICE_UUID_TEST}},‎**
Why can’t I use any randomly generated service UUID?‎

chegewara, 2020-10-16:
Ok, sorry for giving you wrong link, here is better one:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table/tutorial/Gatt_Server_Service_Table_Example_Walkthrough.md#creating-services-and-characteristics-with-the-attribute-table

Look at this example code:
```
    [HRS_IDX_SVC]                       =
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ,
      sizeof(uint16_t), sizeof(heart_rate_svc), (uint8_t *)&heart_rate_svc}},
```
In most cases this part wont change in your code:
- `ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ`, its because you will use primary services, (secondary/included services are rare), and service data is read only
- what you have to change in code is that part:
`sizeof(uint16_t), sizeof(heart_rate_svc), (uint8_t *)&heart_rate_svc}`
In example heart rate service UUID is 16 bit. Also you should be aware of this description from first link:
```
Note: In little-endian storage mode, you can define service UUID directly in the normal order if it's a 16-bit or a 32-bit UUID, but if service UUID is 128-bit, there is minor difference. 
For example, if the UUID of the service application that the user is interested in is 12345678-a1b2-c3d4-e5f6-9fafd205e457, 
REMOTE_SERVICE_UUID should be defined as {0x57,0xE4,0x05,0xD2,0xAF,0x9F,0xF6,0xE5,0xD4,0xC3,0xB2,0xA1,0x78,0x56,0x34,0x12}.
```

Zeni241, 2020-10-16:
Problem solved!!!!

I changed the code as follows:‎
My service_uuid :‎
`static uint8_t service_uuid[16]={0x36, 0xa2, 0x14, 0xe0, 0x0f, 0x68, 0x11, 0xeb, 0xad, 0xc1, 0x02, 0x42, ‎‎0xac, 0x12, 0x00, 0x02,};` // (**36a214e0-0f58-ffeb-adc1-0242ac120002**)‎
‎//Service Declaration
‎         ```
```
[IDX_SVC] =‎
‎            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ‎ESP_GATT_PERM_READ, sizeof(uint16_t), sizeof(**service_uuid**), (uint8_t *)&**service_uuid**}},‎
```
Now in client I see the Service UUID: **020012ac-4202-c1ad-eb11-680fe014a236‎**

Thanks a lot chegewara.
I hope I am doing it right.

closed 2020-11-10 by Zeni241