espressif/esp-idf #5989: BLE full scan feature seems ineffective for BLE mesh performance on esp-idf v4.1

AlpinProgrammer, 2020-10-16:
## Environment

- Development Kit:        none
- Module or chip used:  ESP32-WROVER-E
- IDF version:                v4.1
- Build System:              Make
- Compiler version:       (crosstool-NG esp-2020r2) 8.2.0
- Operating System:      Linux
- Using an IDE?:           No
- Power Supply:            USB

## Problem Description

Hello,
We develop a BLE mesh solution and we are currently sending BLE mesh generic onoff commands to a BLE mesh generic onoff server.

In order to assert that the command has been successful, we are expecting an ACK from the server. We already made sure that the ACK was sent by the server so we know the ESP32 doesn't receive all the ACKs.

This is why we used the **BLE full scan feature** (BTDM_CTRL_FULL_SCAN_SUPPORTED). We meet the direct dependencies (BTDM_CTRL_MODE_BLE_ONLY(=y) || BTDM_CTRL_MODE_BTDM(=n)).

The BLE full scan feature proved to be effective using esp-idf v4.0 release.
However, using esp-idf v4.1 release, we notice similar performance regardless of BLE full scan feature being activated.

### Expected Behavior

The BLE full scan feature improves BLE mesh performance in esp-idf v4.0 and v4.1.

### Actual Behavior

The BLE full scan feature shows to only be effective in esp-idf v4.0.

### Steps to reproduce

1. Use an ESP32 to send generic onoff commands to a BLE mesh generic onoff server (ESP32 or nRF)
2.  Compare the percentage of status messages received with and without BLE full scan feature

Thanks

AlpinProgrammer, 2020-10-21:
Has anyone noticed this issue ?

To illustrate what I observed, I launched 120 commands and observed the performance using esp-idf v4.0 and v4.1 releases:

- Using **v4.0**:
--> 71% of the status messages sent by the server were received by the ESP32 using full scan
--> 57% of the status messages sent by the server were received by the ESP32 **NOT** using full scan

- Using **v4.1**:
--> 57% of the status messages sent by the server were received by the ESP32 using full scan
--> 63% of the status messages sent by the server were received by the ESP32 **NOT** using full scan

To be accurate, I noticed that 100% of the commands were taken into account by the server. Also, the status messages are sent by the server. The problem is the ESP32 not receiving those messages.
I made sure to perform both tests in the same conditions (distance client <-> server, amount of wireless devices around...)

Campou, 2020-10-22:
Hi @AlpinProgrammer 

Thanks for reporting this.

If convenient, could you please help to provide the commit id used for esp-idf v4.0 and esp-idf v4.1?

Thanks.

AlpinProgrammer, 2020-10-22:
Hi @Campou 
Sure. For both I used the tagged release.
esp-idf v4.0: 463a9d8b7f9af8205222b80707f9bdbba7c530e1
esp-idf v4.1: 5ef1b390026270503634ac3ec9f1ec2e364e23b2

So yes, I rely on the tags you sent.

AlpinProgrammer, 2020-10-27:
Hi @Campou 
Did you make the same observation as I did ?
It's important to notice that with an ESP32 as server, the results are very good both with and without full scan activated.
If you can, I suggest you use an nRF server (we use an nRF52840) instead.

Thanks

AlpinProgrammer, 2020-11-30:
Hello @Campou !
I have some update as I tried to do the same test on the new esp-idf v4.0.2. 
It seems that it gives the same results as previously: the BLE full scan feature doesn't improve the mesh performance.

It is a major issue for us as what we are trying to develop does not give a satisfactory reliability with ESP32 for the reasons described above.
Don't hesitate if you need any additional information to help you reproduce the issue.

Thanks