espressif/esp-idf #5990: resetting via  esp_restart() does not flush UART hardware TX-FIFO

JB-DX, 2020-10-16:
ESP-IDF:   v4.2-dev-994-gc9f29e0b5-dirty

Problem:
When an app has been restarted via  esp_restart(), the uart tx hardware fifo appears to contain some random bytes that are not immediately sent out.

What we did:
At startup uart_driver_install() is called and then followed by this 9-bytes test-sequence.
`uart_write_bytes(2, "\x11\x22\x33\x44\x54\x66\x77\x88\x99", 9);
`

During a normal power-up boot that sequence is seen at the UART-TX output.
The second line is a message of our protocol
```
11 22 33 44 54 66 77 88 99 
55 AA 01 FC 16 4C 01 20 33 73 54 65 72 6D 2D 46 39 38 43 3A 39 30 30 35 00 00 00 55 47 

```
After a esp_restart() initiated boot the same number of bytes are sent but have different values. Now the UART sends the test-sequence starting at byte #3 in the second line, which is part of our protocol. The  protocol bytes start at byte #12. The message it too short, thus destroying our protocol.
Result: the byte protocol used in our application can't work anymore.
```
BF C1 55 74 70 93 DE 2D B2 
6E 65 11 22 33 44 54 66 77 88 99 55 AA 01 FC 16 4C 01 20 33 73 54 65 72 6D 2D 46 39 38 

```
Only re-applying power fixes this odd UART behaviour.
I ran into this problem after an OTA update that calls esp_restart() 

Is there a way to reset the UART hardware and its FIFO?

Or better: How can a hardware reset be triggered via software?
I can't use any IO-port for it, as the hardware boards are already in production.

Why on earth does the UART TX-FIFO keep bytes without sending them?
Or is there a different reason?

negativekelvin, 2020-10-17:
https://github.com/espressif/esp-idf/commit/46713a52752b4db1a5d594b7f9e6215405f6a67e

JB-DX, 2020-10-17:
Thanks for that.
I just downloaded the most recent esp-idf and now it works. Great that it already had been solved a few months ago.

AxelLin, 2020-10-19:
Seems this fix is only in master tree now, other branches also needs the fix.

Alvin1Zhang, 2021-08-23:
Thanks for reporting and sorry for slow turnaround. 

* Fix on release/4.2 is available https://github.com/espressif/esp-idf/commit/7dbd5e18559bc4125802b32166f7577f3f9d7564
* Fix on release/4.1 is available https://github.com/espressif/esp-idf/commit/ab1f6299ffb5ca52c0bcc14dbd16283b972d992c
* Fix on release/4.0 is available at https://github.com/espressif/esp-idf/commit/676905709ac7a5f04a231ffbe351f2faac4cf560
* Fix on release/3.3. is available https://github.com/espressif/esp-idf/commit/c0aee61ce0c1aae91067a6206f8a24ca56c1ebb5

Feel free to reopen.

closed 2021-08-23 by Alvin1Zhang