espressif/esp-idf #5992: IDF v4.1 truncates RMT data at 64 bits

atanisoft, 2020-10-16:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): release/v4.1 branch
- Build System:         cmake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply:    USB or external 5V, doesn't matter

## Problem Description

When configuring an RMT channel as TX using more than one memory block the data will be truncated at 64 items by [this](https://github.com/espressif/esp-idf/blob/release/v4.1/components/soc/esp32/include/hal/rmt_ll.h#L275) line. This line has been removed on IDF v4.2 and master which works as expected.

There are some protocols which are very sensitive to timing and using the RMT translate API can introduce small delays which are avoided when using 2 (or more) memory blocks and pre-encoding the full payload and transmitting in one shot.

### Expected Behavior

The data to be sent in full.

### Actual Behavior

Data is truncated at 64 items.

### Steps to reproduce

Configure an RMT channel to use 2 memory blocks and send at least 65 items to the RMT channel. Capture the output pin data to verify that data is truncated at 64 bits.

Alvin1Zhang, 2020-10-19:
Thanks for reporting, we will look into.

suda-morris, 2020-10-19:
sorry, sounds like we've missed some important backport to v4.1, will fix it! Thanks for spotting that.

closed 2021-06-04 by atanisoft