espressif/esp-idf #5993: sendto returns error EINVAL if ipv4 mapped v6 packet is used

fariouche, 2020-10-16:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version : v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-472-gcf056a7d0
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I'm having an issue with the ipv4 ipv6 dual stack with lwip.
I'm trying to send an UDP packet as a reply to a discover request. The function sendto is returning -EINVAL (-22)
After investigations, I've found the root cause to be a potential bug in the ipv4 mapped v6 packets not being handled properly.
The error comes from udp_sendto(), where a check IP_ADDR_PCB_VERSION_MATCH() is done.
here the ip address type is not the same and the function returns an error.
"pcb" returns an ipv6 type, while dst_ip returns ipv4.

In socket.c, function lwip_sendto(),  just below the comment /* Dual-stack: Unmap IPv4 mapped IPv6 addresses */, 
we enter the condition, and IPADDR_TYPE_V4 is set here.

then udp_sendto() is not happy because the addr type does not match anymore.

Either the check in udp_sendto() is not correct or lwi_sendto() should not to the unmap here...

What do you think?

Thank you!

freakyxue, 2020-11-10:
hi @fariouche 
If you want to map IPv6 address with IPv4, you must set PCB to V6
[image]

fariouche, 2020-11-10:
Hi @freakyxue ,
Thank you for your reply.
I believe that "soc" is an esp32 command? So in your test you tried to send an ipv4 mapped packet from the esp32?
I'm asking because my issue is the other way: receiving an ipv4 mapped packet is not parsed properly by the esp32. (the packet was sent by an android phone)

freakyxue, 2020-11-10:
hi @fariouche 
What you described above is that esp32 failed to send data under the IPv4 mapping IPv6 address?

If not, please tell us the detailed reproduction process. Thank you.

fariouche, 2020-11-10:
Yes, you are right. Sorry.
In my case the sending is to a ipv4 mapped ipv6 peer.
I will try to replicate the issue with a simpler test application

freakyxue, 2020-12-23:
hi @fariouche 
Do you have any other questions? If not, I will close this question. If there is a problem, please open it at any time

fariouche, 2020-12-23:
for now, I will close this issue. Thank you

closed 2020-12-23 by fariouche