espressif/esp-idf #5994: ESP32-S2-Kaluga-1 Kit v1.3 target = esp32s2beta : A fatal error occurred during startup

kg117, 2020-10-16:
- Development Kit:     ESP32-S2-Kaluga-1 Kit v1.3
- Operating System:     Microsoft Windows [Version 10.0.19041.572]
- (Windows only) environment type: Plain Command Prompt
## Problem Description
Is the target esp32s2beta? BETA? is not mentioned in your documentation?
Following your documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp32-s2-kaluga-1-kit.html#getting-started ENDS with a crash.
//Detailed problem description goes here.

[esp-idf-project16102020 esp32s2beta.txt](https://github.com/espressif/esp-idf/files/5393946/esp-idf-project16102020.esp32s2beta.txt)

esptool.py v2.9-dev
Serial port COM11
Connecting....
Chip is ESP32S2 Beta
Features: Engineering Sample
Crystal is 40MHz
MAC: 7c:df:a1:01:b1:7c
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x50)
esptool.py failed with exit code 2

atanisoft, 2020-10-16:
@kg117 have you tried with IDF master using the target "esp32s2" ?

kg117, 2020-10-16:
I followed [https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#get-started-get-esp-idf](url)
and used the windows ESP-IDF Tools Installer.
Does the installer provide the Master
I am using esp-idf-v4.1 with target=esp32s2beta which gave:
C:\Users\kevin\eclipse-workspace\Hello_World>idf.py set-target esp32s2
Usage: idf.py set-target [OPTIONS] [esp32|esp32s2beta]
Try 'idf.py set-target --help' for help.

Error: Invalid value for '[esp32|esp32s2beta]': invalid choice: esp32s2. (choose from esp32, esp32s2beta)

atanisoft, 2020-10-16:
You likely will need to switch to IDF master instead of v4.1.

felmue, 2020-10-16:
Hi guys

I do have the predecessor kit `ESP32-S2-Kaluga-1 Kit v1.2` and I am using the IDF `master` branch and `esp32s2` as target successfully.

Cheers
Felix

kg117, 2020-10-17:
Installed master.
Ran D:\GitHub\esp-idf>install.bat
Ran D:\GitHub\esp-idf>export.bat
"Go to the project directory and run:
 idf.py build"
Ran D:\GitHub\esp-idf>idf.py set-target esp32s2
ERROR
Traceback (most recent call last):
  File "D:\GitHub\esp-idf\tools\idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "D:\GitHub\esp-idf\tools\idf_py_actions\tools.py", line 1, in <module>
    import click
ImportError: No module named click
Hi @atanisoft next step?

atanisoft, 2020-10-17:
It appears that you are missing some of the required dependencies. I'm not on windows so I can't really help with that part.

kg117, 2020-10-17:
Thank you @atanisoft 
Any Windows help please?

kg117, 2020-10-17:
No module named CLICK   ?????

C:\Users\kevin\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200709 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Installing Python packages from C:\Users\kevin\esp\esp-idf\requirements.txt
Processing c:\users\kevin\esp\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 4)) (50.3.1)
Requirement already satisfied: click>=5.0 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 11)) (3.1.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 13)) (0.26)
Requirement already satisfied: gdbgui==0.13.2.0 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 17)) (0.9.0.2)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 19)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 20)) (3.1.7)
Requirement already satisfied: kconfiglib==13.7.1 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from -r C:\Users\kevin\esp\esp-idf\requirements.txt (line 21)) (13.7.1)
Requirement already satisfied: windows-curses in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from esp-windows-curses==0.1->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 27)) (2.2.0a1)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 11)) (1.14.3)
Requirement already satisfied: six>=1.4.1 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: Flask<1.0,>=0.12.2 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (0.12.5)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (2.7.1)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (1.7.0)
Requirement already satisfied: gevent<2.0,>=1.2.2 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (1.5.0)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 14)) (2.9.6)
Requirement already satisfied: pycparser in c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env\lib\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r C:\Users\kevin\esp\esp-idf\requirements.txt (line 11)) (2.20)
[... cut ...]
C:\Users\kevin\esp\esp-idf>cd C:\Users\kevin\eclipse-workspace\hello_world

C:\Users\kevin\eclipse-workspace\hello_world>idf.py build
Traceback (most recent call last):
  File "C:\Users\kevin\esp\esp-idf\tools\idf.py", line 45, in <module>
    from idf_py_actions.tools import (executable_exists, idf_version, merge_action_lists, realpath)  # noqa: E402
  File "C:\Users\kevin\esp\esp-idf\tools\idf_py_actions\tools.py", line 1, in <module>
    import click
ImportError: No module named click

C:\Users\kevin\eclipse-workspace\hello_world>

felmue, 2020-10-17:
HI @kg117 

unfortunately no, I am on Mac.

@all: anybody out there using Windows who could help @kg117 ?

Thanks
Felix

dobairoland, 2020-10-19:
Hi @kg117. Please check that you don't have a Windows-level association for running `*.py` files, i.e. I suspect that `idf.py` could be executed with some other Python interpreter different from the one in `c:\users\kevin.espressif\python_env\idf4.3_py3.9_env`.

kg117, 2020-10-20:
@dobairoland 
I removed all copies of python except 3.9
Under Settings -> Default Apps -> Choose default applications by file type: .py was associated with Visual Studio 2019. To change from VS to Python I needed to install Python from Microsoft Store at which time .py was associated with Python.
D:\eclipse-workspace\esp-idf-master.zip   (do not use zip file)
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html (Always check you are in the right section: esp32s2)
https://github.com/espressif/esp-idf
C:\Users\kevin\esp\
git clone --recursive https://github.com/espressif/esp-idf.git (must be recursive)
==============================================================
C:\Users\kevin\esp>git clone --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
remote: Enumerating objects: 2240, done.
remote: Counting objects: 100% (2240/2240), done.
remote: Compressing objects: 100% (1144/1144), done.
.................................................... takes quite a while
Submodule path 'components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a'
Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
Submodule path 'components/tinyusb/tinyusb': checked out 'a2ba3dccccf94022d31e939fa2ce4dca5f0a34f0'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'c3d3a69021cfec3236ca2c0b63be4048ec6643a4'

C:\Users\kevin\esp>
======================================================================================

C:\Users\kevin\eclipse-workspace\Hello_World
IDF_PATH C:\Users\kevin\esp\esp-idf

C:\Users\kevin\esp\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
C:\Users\kevin\esp\esp-idf\tools\idf_tools.py:308: DeprecationWarning: urllib.parse.splittype() is deprecated as of 3.8, use urllib.parse.urlparse() instead
  url_type, path = splittype(url)
Done
Extracting C:\Users\kevin\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0
Installing xtensa-esp32s2-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0
Installing xtensa-esp32s3-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\dist\xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\xtensa-esp32s3-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\kevin\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0
Installing esp32ulp-elf@2.28.51-esp-20191205
Downloading binutils-esp32ulp-win32-2.28.51-esp-20191205.zip to C:\Users\kevin\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp-20191205.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp-20191205.zip to C:\Users\kevin\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205
Installing esp32s2ulp-elf@2.28.51-esp-20191205
Downloading binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip to C:\Users\kevin\.espressif\dist\binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\binutils-esp32s2ulp-win32-2.28.51-esp-20191205.zip to C:\Users\kevin\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205
Installing cmake@3.16.4
Downloading cmake-3.16.4-win64-x64.zip to C:\Users\kevin\.espressif\dist\cmake-3.16.4-win64-x64.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\cmake-3.16.4-win64-x64.zip to C:\Users\kevin\.espressif\tools\cmake\3.16.4
Installing openocd-esp32@v0.10.0-esp32-20200709
Downloading openocd-esp32-win32-0.10.0-esp32-20200709.zip to C:\Users\kevin\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip to C:\Users\kevin\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709
Installing ninja@1.10.0
Downloading ninja-1.10.0-win64.zip to C:\Users\kevin\.espressif\dist\ninja-1.10.0-win64.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\ninja-1.10.0-win64.zip to C:\Users\kevin\.espressif\tools\ninja\1.10.0
Installing idf-exe@1.0.1
Downloading idf-exe-v1.0.1.zip to C:\Users\kevin\.espressif\dist\idf-exe-v1.0.1.zip.tmp
Done
Extracting C:\Users\kevin\.espressif\dist\idf-exe-v1.0.1.zip to C:\Users\kevin\.espressif\tools\idf-exe\1.0.1
Installing ccache@3.7
[... 72 lines cut ...]
All done! You can now run:
   export.bat
C:\Users\kevin\esp\esp-idf>
=======================================================================================
C:\Users\kevin\esp\esp-idf>export.bat
Setting IDF_PATH: C:\Users\kevin\esp\esp-idf

[... cut ...] sktop\bin

Checking if Python packages are up to date...
Python requirements from C:\Users\kevin\esp\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
========================================================================================

cd \Users\kevin\eclipse-workspace\hello_world
C:\Users\kevin\eclipse-workspace\hello_world>idf.py set-target esp32s2

Causes extra cmd window to flash by: is this a crash?

dobairoland, 2020-10-20:
@kg117 Looking at your PATH, you still have `C:\Program Files\Python38` and `C:\Python27` as well.

I'm assuming that `C:\Program Files\Python39` is the default one which was used to create `c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env` for IDF.

> Under Settings -> Default Apps -> Choose default applications by file type: .py was associated with Visual Studio 2019. To change from VS to Python I needed to install Python from Microsoft Store at which time .py was associated with Python.

This is what causing your issue. The packages are installed in `c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env`. However, when you run `idf.py` then `C:\Program Files\Python39` is used which doesn't have the packages installed.

You can solve this issue by *removing* all associations to `*.py` files. Note that association to `c:\users\kevin\.espressif\python_env\idf4.3_py3.9_env` is also not a good idea and has other side effects.

Please also note that Python from Windows Store has some known issues and I recommend to install Python from the official Python website.

dobairoland, 2020-10-20:
There is another solution. Moving `C:\Users\kevin\.espressif\tools\idf-exe\1.0.1\` before `C:\Users\kevin\esp\esp-idf\tools` in your PATH. I'm not sure right now what is causing this wrong order. Have you manually added ESP-IDF related paths to your environment (at the Windows level)?

kg117, 2020-10-20:
Thank you @dobairoland I will remove all references to python and restart the install process with no association to python .py and install python from python.org. This time I will clean python out of my PC including PATH
Good of you to help. 
Kevin.

kg117, 2020-10-21:
Thank you @dobairoland your comments have been most helpful. I have documented everything I have done. I have used 
idf.py -p COM11 monitor: perhaps some of the lines are errors?
[Issue 21102020 5994.txt](https://github.com/espressif/esp-idf/files/5413158/Issue.21102020.5994.txt)
I hope my file will be helpful to others.
Regards,
Kevin.

willz1200, 2020-11-27:
On windows I'd recommend sticking to the ESP-IDF Tools Installer, you can actually select the specific version of the ESP-IDF you want to checkout during the installation. Just go for the release branch for v4.2 as shown in the following image, then you should have support for the production esp32-s2.
[image]
Thanks,
Will

dobairoland, 2020-11-27:
Thank you @willz1200 for the additional help. I'm closing this now because the original issue were resolved.

closed 2020-11-27 by dobairoland