espressif/esp-idf #5995: get_sextet() does not return a value.

DerrickLau, 2020-10-17:
In your implementation of the <array> header, the function get_sextet() does not return a value, when it should return an uint32_t.

negativekelvin, 2020-10-17:
https://www.reddit.com/r/esp32/comments/jcoaql/why_is_get_sextet_not_returning_uint32_t/

Please close this and don't post extremely vague issues like this or issues with 3rd party code

istokm, 2020-10-17:
To clear this up, he seems to be referring to this [lambda function in a 3rd-party library](https://github.com/Thalhammer/jwt-cpp/blob/master/include/jwt-cpp/base.h#L155) that has nothing to do with ESP-IDF. 

Also, from that post, I assume by
> In your implementation of the header

you mean the `std::array` header, then I think you might be confusing it with something else, as that function is not a part of it. Source: https://en.cppreference.com/w/cpp/container/array

DerrickLau, 2020-10-18:
Sorry...I am not familiar with lamda function syntax in C++...closing the issue out.  And thanks for the CMake suggestion!

closed 2020-10-18 by DerrickLau