espressif/esp-idf #5997: Sound freeze A2DP src if SBC set to 48KHz

64-coreCPU, 2020-10-17:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version [v4.3-dev-1521-g0289d1cc8]
- Build System:         [idf.py]
- Compiler version [xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0]
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external 3.3V]

## Problem Description

(Sorry for my english)

I have also vs1053 audio codec connected over SPI and i2s and SDcard connected over SPI.
ESP32 reads audio file from SDcard and send it to vs1053 over SPI. vs1053 decode audio and return it over i2s.
Then ESP32 reads PCM from i2s and send it over A2DP. The problem is that the codec can only return audio in formats 48/96/192KHz, but SBC in ESP32 hardcodet to 44100.
I used this https://github.com/espressif/esp-idf/issues/1720#issuecomment-374894493 to setup SBC to 48KHz.

Now the very essence.
If I do not touch the files bta_av_co.c and bta_av_co.c and play the sound "as is", then it plays without problems (if you do not take into account distortion and cilks due to discrepancy in the sampling rate).
If I make changes to those files and tune to a frequency of 48KHz, then the bluetooth sound plays for only a few seconds, and then freezes.
For these few seconds, the sound over bluetooth plays good quality without distortion.
However, the vs1053 continues to play sound - I can hear it in wired headphones that are directly connected to the vs1053.
This also indicates that the ESP32 is not frozen, because if the vs1053 constantly needs to receive data over the bus to play.
There are also no errors in the console when the bluetooth sound hangs.

I am using the following callback function:
```cpp

...

esp_a2d_source_register_data_callback( a2dp_input_data_callback );

...

int a2dp_input_data_callback( unsigned char* data, int dataLength ){
	
	if (dataLength < 0 || data == NULL) return 0;
	
	size_t bytes_read = 0;
	
	printf("%i ", dataLength );
	
	i2s_read( I2S_NUM_0, data, dataLength, &bytes_read, portMAX_DELAY );
	
	//if( i2s_read( I2S_NUM_0, data, dataLength, &bytes_read, portMAX_DELAY ) )return 0;
	//if( i2s_read( I2S_NUM_0, data, dataLength, &bytes_read, 500 / portTICK_PERIOD_MS ) ) return 0;
	
	printf("%u\n", bytes_read );
	
	return bytes_read;
}
```

While the bluetooth sound is playing, the console outputs "512 512". After a few seconds, the bluetooth sound stops and the output to the console also stops.
The last line in the console looks like "512 512".
In my opinion, this says two things:
1) At some point, the callback function stops being called.
2) The most recent call was successful because as many bytes were read as requested.

After the end of the file playback (the sound plays only in wired headphones, the bluetooth is silent), a call to the esp_a2d_source_disconnect() function is made in my code and the code hangs at this point.
Bluetooth headphones remain connected (they usually signal when disconnected). The console is still quiet.

I also described the problem here https://esp32.com/viewtopic.php?f=12&t=17745

Any ideas?

sdkconfig https://0x0.st/iDKB.txt

Alvin1Zhang, 2020-10-19:
Thanks for reporting, we will look into.

Wth-Esp, 2020-10-19:
Hi

You say that when ESP32 as an A2DP source and also uses vs1053 to encode the PCM audio saved in the SD card with 48k. The audio stream stops after a while but it seems that ESP32 and vs1053 both are still working from the log and since you can still listen to the music by wired link. Am I right?

I think you can use the default sequence frequency to test whether the 48k codec is to be blamed first. And I suggest you don't use i2s_write in the data callback function, you can use a RingBuffer A in the data callback to save the data and post the data to another RingBuffer B to receive the data. And use i2s_write to read data from RingBuffer B in another task.

Thanks

64-coreCPU, 2020-10-19:
Wth-Esp

I am using vs1053 to decode audio files(such as mp3 ogg etc) from SDcard but not bare PCM data.
Yes, you understand correctly that the bluetooth audio stream freezes after a while, but the ESP32 and vs1053 continue  playing.

I didn't fully understand you. I don't use the i2s_write function at all. I transfer audiofile from ESP32 to the vs1053 over SPI bus and receive decoded audio over i2s from vs1053. vs1053 can output only 48 or 96 or 192 KHz audio over i2s. If I do not touch the SBC, then the bluetooth sound is reproduced with distortion but without freezes and problems. If i use this https://github.com/espressif/esp-idf/issues/1720#issuecomment-374894493 to setup SBC to 48KHz, then the bluetooth sound freezes in seconds after the start, but it plays well.

Wth-Esp, 2020-10-19:
@64-coreCPU 

OK, I understand now.

Oh, I mean that you directly call `i2s_read` in function `a2dp_input_data_callback` which you register to the BT stack.  that may block the tasks of the Bluetooth stack which eventually causes the audio stream blocked.

I will look into the audio stream of BT stack and I will contact you if any progress.

Thanks

64-coreCPU, 2020-10-19:
Wth-Esp

I'm not sure about the bluetooth stack locks in a2dp_input_data_callback(), because the last entry in the console is "512 512". The first value is the number of bytes requested, and the second is the number of bytes read. I gave the code in the first post.

I noticed something strange. Since I had never worked with ring buffers before and I don't know what it is, I decided to run a "quick test" to isolate the problem. I just replaced the code of the callback function with the code from the example that generates random noise

```
static int32_t a2dp_input_data_callback(uint8_t *data, int32_t len)
{
    if (len < 0 || data == NULL) {
        return 0;
    }

    // generate random sequence
    int val = rand() % (1 << 16);
    for (int i = 0; i < (len >> 1); i++) {
        data[(i << 1)] = val & 0xff;
        data[(i << 1) + 1] = (val >> 8) & 0xff;
    }

    return len;
}
```
I did not touch anything else - the  vs1053 is still playing in the background.  Using such a callback function does not cause the bluetooth sound to hang. 

I will try to examine the ring buffers and follow your advice you gave earlier.

64-coreCPU, 2020-10-20:
Wth-Esp

I have looked into  ring buffers.
Here are the changes I made to the code according to your recommendation.

I have created a separate task that reads the  i2s bus into the ring buffer.

```
...
TaskHandle_t g_i2s0_ringBufferTask_handler = NULL;
RingbufHandle_t g_i2s0_ringBuffer_handler = NULL;
...
g_i2s0_ringBuffer_handler = xRingbufferCreateNoSplit( 512, 8 );
xTaskCreatePinnedToCore( i2s0_ring_buffer_task, "i2s0_RBT", 1024, NULL, 2, &g_i2s0_ringBufferTask_handler, tskNO_AFFINITY );
...

static void i2s0_ring_buffer_task(){
	
	size_t bytes_read = 0;
	
	unsigned char* mem = malloc( 512 );
	
	if( !mem ){
		printf(" i2s0 task malloc err\n ");
		for(;;)vTaskDelay( 1000 / portTICK_PERIOD_MS );
	}
	
	printf("i2s0 task started\n");
	
	for(;;){
		if( i2s_read( I2S_NUM_0, mem, 512, &bytes_read, portMAX_DELAY ) )printf("i2s0 read err\n");
		if( xRingbufferSend( g_i2s0_ringBuffer_handler, mem, 512, portMAX_DELAY ) == pdFALSE )printf("i2s0 ringBufferSend err\n");
	}
	
}
```
In real code, return values from the xRingbufferCreateNoSplit() and xTaskCreatePinnedToCore() functions are checked - no errors.

I also changed the code of the callback function to
 ```
int a2dp_input_data_callback( unsigned char* data, int dataLength ){
	
	printf("%i ", dataLength );
	
	if (dataLength < 0 || data == NULL)return 0;
	
	size_t itemSize = 0;
	unsigned char* ringBufferItem = xRingbufferReceive( g_i2s0_ringBuffer_handler, &itemSize, portMAX_DELAY );
	
	if( ringBufferItem == NULL ){
		printf("ringBufferItem is NULL\n");
		return 0;
	}
	if( itemSize != dataLength ){
		printf( "a2dp_input_data_callback itemSize != dataLength (%u != %u)\n", itemSize, dataLength );
		
		vRingbufferReturnItem( g_i2s0_ringBuffer_handler, ringBufferItem );
		return 0;
	}
	
	memcpy( data, ringBufferItem, itemSize );
	
	vRingbufferReturnItem( g_i2s0_ringBuffer_handler, ringBufferItem );
	
	printf( "%u\n", itemSize );
	
	return itemSize;
}
```

The behavior hasn't changed at all. The bluetooth sound plays for several seconds and the console outputs "512 512". After a few seconds, the bluetooth sound stops and the output to the console also stops. The last entry in the console is still "512 512". The sound in wired headphones still plays to the end (while the bluetooth is quiet). The code also hangs at the very end when calling the esp_a2d_source_disconnect() function. There are still no errors in the console.

Hence I have one question - did I understand you correctly about the ring buffers or did I do something wrong?

boblane1, 2020-10-21:
Hi, @64-coreCPU 

Your understanding about ring buffers is correct, but I think it's better to replace `portMAX_DELAY` in `xRingbufferReceive( g_i2s0_ringBuffer_handler, &itemSize, portMAX_DELAY )` with `0`, because the Bluetooth stack on esp32 is non-blocking.

64-coreCPU, 2020-10-21:
boblane1

This gave some results.
First I replaced the line in the callback function 
```
printf("ringBufferItem is NULL\n");
```
 to
```
printf("NULL\n");
```
 to be short.

When I ran the code, I saw the following output in the console:
 ```
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 NULL
W (210466) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (210466) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 0
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 NULL
W (210496) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (210496) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
```

Now bluetooth does not freeze, but the sound is still interrupted and distortions and clicks appeared. In this case, it was obvious that the ESP32 was running out of time due to the output to the console, so I removed all printf () functions and tried again.

Only this is left in the console
 ```
W (196897) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 0
W (196927) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (196927) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
W (196957) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (196957) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 0
W (196987) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (196987) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
W (197017) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197017) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 0
W (197047) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197047) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
W (197077) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197077) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 0
W (197107) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197107) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
W (197137) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197137) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 0
W (197167) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197167) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 0
W (197197) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (197197) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 0
```
It looks like the callback is being called more often than the data has accumulated in the ring buffer. The bluetooth sound is played for several seconds without distortion and problems, but then it dies down. After about a second, it resumes again and plays for a few seconds. Then the situation continues. It's like you mute the sound and then unmute it. Throughout the entire track (about 3 minutes), the messages that I indicated above are constantly displayed in the console. Bluetooth now does not freeze and at the end of the test there is a normal disconnection using esp_a2d_source_disconnect().

I also tried without ring buffer:
 ```
int a2dp_input_data_callback( unsigned char* data, int dataLength ){
	
	if (dataLength < 0 || data == NULL) return 0;
	
	size_t bytes_read = 0;
	if( i2s_read( I2S_NUM_0, data, dataLength, &bytes_read, 0 ) )return 0;

	return bytes_read;
}
```
I similarly replaced portMAX_DELAY with 0. The behavior is exactly the same as with a ring buffer, except for two things. 
First, the console output is slightly different.
 ```
W (116016) BT_APPL: ### UNDERFLOW :: ONLY READ 256 BYTES OUT OF 512 ###
W (116016) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 256
W (116046) BT_APPL: ### UNDERFLOW :: ONLY READ 256 BYTES OUT OF 512 ###
W (116046) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 256
W (116076) BT_APPL: ### UNDERFLOW :: ONLY READ 256 BYTES OUT OF 512 ###
W (116076) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 256
W (116106) BT_APPL: ### UNDERFLOW :: ONLY READ 128 BYTES OUT OF 512 ###
W (116106) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 128
W (116136) BT_APPL: ### UNDERFLOW :: ONLY READ 256 BYTES OUT OF 512 ###
W (116136) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 256
W (116166) BT_APPL: ### UNDERFLOW :: ONLY READ 128 BYTES OUT OF 512 ###
W (116166) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 128
W (116196) BT_APPL: ### UNDERFLOW :: ONLY READ 128 BYTES OUT OF 512 ###
W (116196) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 10, 128
```
Secondly, clicks are constantly present in the bluetooth sound.

I also tried adding printf () to see how many bytes were requested and read each time. Here's an example of the console output:
 ```
128 128
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 128
W (196616) BT_APPL: ### UNDERFLOW :: ONLY READ 128 BYTES OUT OF 512 ###
W (196616) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 9, 128
384 384
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 512
512 384
W (196646) BT_APPL: ### UNDERFLOW :: ONLY READ 384 BYTES OUT OF 512 ###
W (196646) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 11, 384
```

In all other respects, everything is identical: the bluetooth does not freeze, the bluetooth sound stops every few seconds for about a second and resumes again, after playing the track, the bluetooth headset is successfully turned off. Throughout the playback, there is constant output in the console (I indicated the examples above). The  vs1053 plays in both cases without any problems - I can hear it in wired headphones that are directly connected to it.

This leads me to think that the i2s frequency is slightly less than 48000Hz (unfortunately, I do not have an oscilloscope to check this), so there are clicks, but why the bluetooth sound periodically turns off for a second is not clear.

boblane1, 2020-10-22:
Hi, @64-coreCPU 

Do you develop the project which base on the a2dp_source example? This example has a heart beat timer to disconnect the audio stream. By the way, I think the clicks may be caused by the print of `### UNDERFLOW...`.

64-coreCPU, 2020-10-22:
boblane1

No, I am developing my project almost from scratch and do not use the code from the example (except for the callback functions to generate random noise for diagnostics).

How can you turn off the display of these messages to check if they are the reason for the clicks?

boblane1, 2020-10-26:
Hi, @64-coreCPU 

If you want to turn off the display of messages like `### UNDERFLOW....`, you can just find and comment out them.

Moreover, would you please add `APPL_TRACE_ERROR("cnt:%d,res:%d\n", a2dp_source_local_param.btc_aa_src_cb.media_feeding_state.pcm.counter, result);` in `/esp-idf/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_a2dp_source.c` at line 1806 in function `btc_get_num_aa_frame` and upload your log.  This may be helpful to figure out the problem.

Thanks!

64-coreCPU, 2020-10-26:
boblane1

https://0x0.st/ik2S.txt

boblane1, 2020-10-27:
Hi， @64-coreCPU 

I think the problem is  due to the speed of i2s and SBC is not match, which you have been declared. So would you mind do a little experiment to prove it? Since you don't have an oscilloscope to check the i2s frequency, you can calculate the approximate speeds of i2s and a2dp data callback, to make sure whether have correct configuration of vs1053, i2s or SBC. For example, you can use the following code to calculate the approximate speed of SBC. 

```
static uint32_t byte_cnt = 0;
static uint32_t frame_cnt = 0;
static uint64_t start_time = 0;
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (frame_cnt == 0) {
        start_time = esp_timer_get_time();
    }
    ++frame_cnt;
    byte_cnt += len;
    if (frame_cnt == 100) {
        printf("speed:%.2f\n", (double)byte_cnt*1000*1000/(esp_timer_get_time() - start_time));
        frame_cnt = 0;
        byte_cnt = 0;
    }
    if (len < 0 || data == NULL) {
        return 0;
    }

    // generate random sequence
    int val = rand() % (1 << 16);
    for (int i = 0; i < (len >> 1); i++) {
        data[(i << 1)] = val & 0xff;
        data[(i << 1) + 1] = (val >> 8) & 0xff;
    }

    return len;
}
```
You can also use this method to calculate the approximate speed of i2s.

From your log, I can't see any signs of sound periodically turning off.

Thanks!

64-coreCPU, 2020-10-27:
boblane1

If i using random as sound, speed varies within about 191411.17 - 207274.02. 
If i reading i2s, speed varies within about 194439.49 - 214752.49.

boblane1, 2020-10-27:
Hi， @64-coreCPU

Do you calculate i2s speed in `i2s0_ring_buffer_task`? If not, I am sorry for misunderstanding you by saying ‘You can also use this method to calculate the approximate speed of i2s‘.  You can refer to the following code .
```
static uint32_t byte_cnt = 0;
static uint32_t frame_cnt = 0;
static uint64_t start_time = 0;

static void i2s0_ring_buffer_task(){
	
	size_t bytes_read = 0;
	
	unsigned char* mem = malloc( 512 );
	
	if( !mem ){
		printf(" i2s0 task malloc err\n ");
		for(;;)vTaskDelay( 1000 / portTICK_PERIOD_MS );
	}
	
	printf("i2s0 task started\n");
	
	for(;;){
        if(i2s_read( I2S_NUM_0, mem, 512, &bytes_read, portMAX_DELAY) == 0) {
            if (frame_cnt == 0) {
                start_time = esp_timer_get_time();
            }
            ++frame_cnt;
            byte_cnt += bytes_read;
            if (frame_cnt == 100) {
                printf("speed:%.2f\n", (double)byte_cnt*1000*1000/(esp_timer_get_time() - start_time));
                frame_cnt = 0;
                byte_cnt = 0;
            }
        } else {
            printf("i2s0 read err\n");
        }
    }
}
```

64-coreCPU, 2020-10-27:
boblane1

Yes, I misunderstood you.
In the ring task i2s speed about 189387.63.

boblane1, 2020-10-27:
Hi， @64-coreCPU

Thanks, It indeed seems that i2s frequency is slightly less than SBC. I can offer you some suggestion, but I don't know it whether helps.

1. Simplify the log output.
```
APPL_TRACE_WARNING("### UNDERFLOW :: ONLY READ %d BYTES OUT OF %d ###", nb_byte_read, read_size);
=> APPL_TRACE_WARNING("# %d/%d", nb_byte_read, read_size);

APPL_TRACE_WARNING("btc_media_aa_prep_sbc_2_send underflow %d, %d", nb_frame, a2dp_source_local_param.btc_aa_src_cb.media_feeding_state.pcm.aa_feed_residue);
=> APPL_TRACE_WARNING("[%d, %d]", nb_frame, a2dp_source_local_param.btc_aa_src_cb.media_feeding_state.pcm.aa_feed_residue);
```
2. Boost the priority of the `i2s0_ring_buffer_task`, for example `configMAX_PRIORITIES - 3`.
3. Enlarge `g_i2s0_ringBuffer_handler ` appropriately, for example `xRingbufferCreateNoSplit(512, 16);`
4. Let the task be distributed on the balance of the two cores.

64-coreCPU, 2020-10-27:
boblane1

This did not help - the bluetooth sound is still interrupted.

However, the original problem is solved - bluetooth does not freeze. Thank you all for your help!

64-coreCPU, 2020-11-04:
I figured out the problem completely The problem was that on the vs1053 debug board, quartz was crucified at 12,000, and it should have been 12.288. Because of this, the speed of the i2s bus was slightly slower and this led to problems.

I end up using the following callback function
```
int a2dp_input_data_callback( unsigned char* data, int dataLength ){
	
	if (dataLength < 0 || data == NULL) return 0;
	
	size_t bytes_read = 0;
	if( i2s_read( I2S_NUM_0, data, dataLength, &bytes_read, portMAX_DELAY ) )return 0;
	return bytes_read;
}
```

and it works great. I'll keep in mind that the bluetooth stack may freeze in case of insufficient data on the i2s bus.

Finally, I want to ask a question. Is the fact that the bluetooth stack freezes due to lack of data is a bug or not? While reading the documentation, I never came across a mention of not introducing delays in callback functions.

Anyway, the problem is solved. I want to thank everyone for their active help. I would also like to apologize, because in fact, the problem was not originally esp32 at all.

boblane1, 2020-11-05:
Hi, @64-coreCPU 

I'm very glad to hear that your problem has been resolved.

The Bluetooth stack on ESP32 freezes is not due to the lack of data but the blocking of callback. The Bluetooth stack on ESP32 is designed as an unblocking and asynchronous pattern, the upper layer uses a fixed length queue to cache the data from the lower layer. When the Bluetooth callback has delay, the consuming speed of the queue will slow down which makes the queue can be full easily, then the stack will freeze. I agree that it is a good suggestion to mention explicitly not introducing any delays in callback functions in API description.

Alvin1Zhang, 2020-11-05:
Thanks for reporting, feel free to reopen.

closed 2020-11-05 by Alvin1Zhang

64-coreCPU, 2020-11-05:
boblane1

Thanks, I'll keep it in mind.