espressif/esp-idf #5998: Panic on BT/WiFi deinit

fpgamaster, 2020-10-17:
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.3-dev-1472-g0b71a0a46
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
A panic produced by the following code after some times returning from deep sleep:
        // Deinit BLUFI
        esp_myg_blufi_profile_deinit();

        // Disable BT and WiFi
        esp_bluedroid_disable();
        esp_bluedroid_deinit();
        esp_bt_controller_disable();
        esp_bt_controller_deinit();
        esp_wifi_stop();
        esp_wifi_deinit();

//Detailed problem description goes here.
The above code is executed right before entering in deep sleep.

[0;32mI (7523) BLUFI: BLUFI Event callback 1

[0m

[0;32mI (7523) BLUFI: BLUFI deinit finish

[0m

[0;33mW (7543) BT_APPL: bta_dm_disable BTA_DISABLE_DELAY set to 200 ms[0m

[0;32mI (7543) GATTC: GATTC Event handler idx 0, event 1[0m

[0;32mI (7543) GATTC: DEFAULT_EVT[0m

[0;32mI (7553) GATTC: GATTC Event handler idx 1, event 1[0m

[0;32mI (7553) GATTC: DEFAULT_EVT[0m

[0;32mI (7563) GATTC: GATTC Event handler idx 2, event 1[0m

[0;32mI (7563) GATTC: DEFAULT_EVT[0m

[0;31mE (7753) BT_OSI: alarm_free null

[0m

CORRUPT HEAP: multi_heap.c:540 detected at 0x3ffbaa74

abort() was called at PC 0x40094b9b on core 0

[0;33m0x40094b9b: multi_heap_assert at /home/xxxx/esp/esp-idf/components/heap/multi_heap_platform.h:68
[0m

Backtrace:0x40090222:0x3fff5d00 0x40090885:0x3fff5d20 0x40096076:0x3fff5d40 0x40094b9b:0x3fff5db0 0x40094fa4:0x3fff5dd0 0x400856b1:0x3fff5df0 0x400960a1:0x3fff5e10 0x4014708a:0x3fff5e30 0x4014797a:0x3fff5e50 0x401479cb:0x3fff5e70 0x40147225:0x3fff5e90 0x40137a41:0x3fff5eb0 0x40163db6:0x3fff5ed0 0x40163e04:0x3fff5ef0 0x40137735:0x3fff5f60 0x401376ed:0x3fff5f80 0x40137843:0x3fff5fa0 0x40148047:0x3fff5fc0

[0;33m0x40090222: panic_abort at /home/xxxx/esp/esp-idf/components/esp_system/panic.c:361
[0m
[0;33m0x40090885: esp_system_abort at /home/xxxx/esp/esp-idf/components/esp_system/system_api.c:106
[0m
[0;33m0x40096076: abort at /home/xxxx/esp/esp-idf/components/newlib/abort.c:46
[0m
[0;33m0x40094b9b: multi_heap_assert at /home/xxxx/esp/esp-idf/components/heap/multi_heap_platform.h:68
[0m
[0;33m0x40094fa4: multi_heap_free_impl at /home/xxxx/esp/esp-idf/components/heap/multi_heap.c:540
[0m
[0;33m0x400856b1: heap_caps_free at /home/xxxx/esp/esp-idf/components/heap/heap_caps.c:305
[0m
[0;33m0x400960a1: free at /home/xxxx/esp/esp-idf/components/newlib/heap.c:47
[0m
[0;33m0x4014708a: bucket_free_ at /home/xxxx/esp/esp-idf/components/bt/common/osi/hash_map.c:244
[0m
[0;33m0x4014797a: list_free_node at /home/xxxx/esp/esp-idf/components/bt/common/osi/list.c:262
[0m
[0;33m0x401479cb: list_remove at /home/xxxx/esp/esp-idf/components/bt/common/osi/list.c:184
[0m
[0;33m0x40147225: hash_map_erase at /home/xxxx/esp/esp-idf/components/bt/common/osi/hash_map.c:175
[0m
[0;33m0x40137a41: bta_sys_free_timer at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:681
[0m
[0;33m0x40163db6: bta_dm_deinit_cb at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c:325
[0m
[0;33m0x40163e04: bta_dm_sys_hw_cback at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c:430
[0m
[0;33m0x40137735: bta_sys_hw_evt_disabled at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:443
[0m
[0;33m0x401376ed: bta_sys_sm_execute at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:228
[0m
[0;33m0x40137843: bta_sys_event at /home/xxxx/esp/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:499
[0m
[0;33m0x40148047: osi_thread_run at /home/xxxx/esp/esp-idf/components/bt/common/osi/thread.c:68
[0m

ELF file SHA256: 4725f1642b69c378

Rebooting...

xiewenxiang, 2020-10-29:
It is recommended to stop scanning, stop broadcasting, and disconnect before deinit stack.

cyberman54, 2020-12-18:
@fpgamaster try a `esp_ble_gap_stop_scanning()` before `esp_bluedroid_deinit()`