espressif/esp-idf #5999: TLS error while using both AWS-IoT shadow and topic services

rhr407, 2020-10-17:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-1197-g8bc19ba89
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I am using AWS-IoT component with shadow and topic service simultaneously. It works fine for some time but one error arrives ad given below:
```
I (964511) wifi:bcn_timout,ap_probe_send_start
I (966681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (967011) wifi:ap_probe_send over, resett wifi status to disassoc
I (967011) wifi:state: run -> init (c800)
I (967011) wifi:pm stop, total sleep time: 26345448 us / 41062945 us

I (967021) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (967021) BLE_PROV: Disconnected. Connecting to the AP again...
E (967031) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (967041) aws_iot:     Unable to verify the server's certificate. 
I (967151) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (967151) wifi:state: init -> auth (b0)
I (967161) wifi:state: auth -> assoc (0)
I (967171) wifi:state: assoc -> run (10)
I (967201) wifi:connected with TALKTALK4B89FC, aid = 1, channel 4, BW20, bssid = d0:c6:5b:4b:8a:0c
I (967201) wifi:security: WPA2-PSK, phy: bgn, rssi: -61
I (967211) wifi:pm start, type: 1

I (967211) wifi:AP's beacon interval = 102400 us, DTIM period = 1
```
### Expected Behavior
No such error
### Actual Behavior

### Steps to reproduce
[... 387 lines cut ...]

    ESP_LOGI(TAG, "Shadow Connect");

    rc = aws_iot_shadow_set_autoreconnect_status(&mqttClient, true);
    if (SUCCESS != rc) {
        ESP_LOGE(TAG, "Unable to set Auto Reconnect to true - %d, aborting...", rc);
        abort();
    }

}

```

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

chegewara, 2020-10-18:
You are not connected to wifi AP yet:
```
I (967021) BLE_PROV: Disconnected. Connecting to the AP again...
E (967031) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (967041) aws_iot:     Unable to verify the server's certificate. 
I (967151) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (967151) wifi:state: init -> auth (b0)
I (967161) wifi:state: auth -> assoc (0)
I (967171) wifi:state: assoc -> run (10)
I (967201) wifi:connected with TALKTALK4B89FC, aid = 1, channel 4, BW20, bssid = d0:c6:5b:4b:8a:0c
```

rhr407, 2020-10-18:
```cpp
Published
I (828881) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":40.000000}}, "clientToken":"net0001-274"}
I (828911) aws-shadow: Wrote 8 bytes

I (828981) aws-shadow: Update accepted
I (831681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (831831) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":25.000000}}, "clientToken":"net0001-275"}
I (831861) aws-shadow: Wrote 8 bytes

I (831921) aws-shadow: Update accepted
I (834681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (834781) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":40.000000}}, "clientToken":"net0001-276"}
I (834811) aws-shadow: Wrote 8 bytes

I (834871) aws-shadow: Update accepted
I (837681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (837951) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":25.000000}}, "clientToken":"net0001-277"}
I (837981) aws-shadow: Wrote 8 bytes

I (838041) aws-shadow: Update accepted
I (840681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (840901) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":40.000000}}, "clientToken":"net0001-278"}
I (840931) aws-shadow: Wrote 8 bytes

I (840991) aws-shadow: Update accepted
I (843681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (843851) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":25.000000}}, "clientToken":"net0001-279"}
I (843881) aws-shadow: Wrote 8 bytes

I (843941) aws-shadow: Update accepted
I (846681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (846811) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":40.000000}}, "clientToken":"net0001-280"}
I (846841) aws-shadow: Wrote 8 bytes

I (846901) aws-shadow: Update accepted
I (849681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (849771) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":25.000000}}, "clientToken":"net0001-281"}
I (849801) aws-shadow: Wrote 8 bytes

I (849861) aws-shadow: Update accepted
I (852681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (852941) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":40.000000}}, "clientToken":"net0001-282"}
I (852971) aws-shadow: Wrote 8 bytes

I (853061) aws-shadow: Update accepted
I (855681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

Published
I (855891) aws-shadow: Update Shadow: {"state":{"reported":{"setTemp":43.000000,"currentTemp":25.000000}}, "clientToken":"net0001-283"}
I (855921) aws-shadow: Wrote 8 bytes

I (858681) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

[... 217 lines cut ...]

I (1059691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1062691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1065691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1068691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1071691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1074691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1077691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1080691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1083691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1086691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1089691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":25,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

I (1092691) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}
```

chegewara, 2020-10-18:
I still dont see any issues here. Esp32 is disconnected from router which probably is caused by wifi sleep mode set up.
This is not an issue and place to get help is on forum.

rhr407, 2020-10-18:
So it means that it's not an internal issue. This could pertain to the wifi router?

chegewara, 2020-10-18:
Try to add this line to your code:
`esp_wifi_set_ps(WIFI_PS_NONE);`

Of course i may miss something, but to me it looks like normal behavior. esp32 is disconnected from AP, because you dont check in code if wifi is connected, you see mbedtls error, then after reconnecting to wifi AP you see shadow is sent again.

AxelLin, 2020-10-19:
> I still dont see any issues here. Esp32 is disconnected from router which probably is caused by wifi sleep mode set up.

Do you mean if setting wifi sleep mode, esp32 may be disconnected from router?
In what kind of condition the esp32 will be disconnected due to wifi sleep mode?
I don't find document mentioned such behavior.

AxelLin, 2020-10-19:
> ```c++
> run -> init (c800)
> ```
The disconnect reason is WIFI_REASON_BEACON_TIMEOUT, and the RSSI value seems between -61 ~ -62.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-beacon-timeout

rhr407, 2020-10-19:
Is this something a normal behavior by idf?

ncohen17, 2020-10-25:
Any updates regarding this issue ? Is there a solution to fix it ?

mahavirj, 2020-10-27:
@rhr407

-  Can you please confirm this issue with any of stable IDF release, e.g. IDF v4.0 or v4.1?
-  Also did you try with AP of different make/model, if possible would also request to log RSSI value using  `esp_wifi_sta_get_ap_info` API.

Might be relevant: https://github.com/espressif/esp-idf/issues/5915

rhr407, 2020-11-08:
I have tested this on v4.1 which is a stable version and got the following error although it's not that frequent. Now the tls error has gone but this one is still persistent.

```
I (767733) RX_UART_TASK: Read 99 bytes: '{"DT":2,"T":40,"S":0,"P":10.1,"F":999.9,"E":999.9,"U":11.1,"A":999,"LO":-122.200682,"LA":37.692625}

E (769663) aws-shadow: Update timed out
W (769873) wifi:alloc eb len=24 type=3 fail, heap:32

W (769873) wifi:m f null

I (770663) wifi:bcn_timout,ap_probe_send_start
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4008df20
0x4008df20: _esp_error_check_failed at /home/riz/esp/esp-idf/components/esp32/panic.c:739

file: "/home/riz/esp/esp-idf/components/esp_common/src/ets_timer_legacy.c" line 70
func: ets_timer_setfn
expression: esp_timer_create(&create_args, (esp_timer_handle_t*)&(ptimer->timer_arg))

ELF file SHA256: a98f5941951a2c56

Backtrace: 0x4008d971:0x3ffd04f0 0x4008df23:0x3ffd0510 0x4010b085:0x3ffd0530 0x400d3de1:0x3ffd0560 0x40163980:0x3ffd0580 0x40160d58:0x3ffd05a0 0x40160ebe:0x3ffd05c0 0x40097692:0x3ffd05e0 0x4008f1d1:0x3ffd0610
0x4008d971: invoke_abort at /home/riz/esp/esp-idf/components/esp32/panic.c:157

0x4008df23: _esp_error_check_failed at /home/riz/esp/esp-idf/components/esp32/panic.c:740

0x4010b085: ets_timer_setfn at /home/riz/esp/esp-idf/components/esp_common/src/ets_time
```

mahavirj, 2020-11-09:
@rhr407 

Good that earlier issue is no more seen with stable release!

> ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4008df20

This points to problem where no memory is available for allocation or its too fragmented to satisfy current block size requirement. Can you please log heap statistics (free memory, largest free block, minimum free heap) parameters at this point? Post on memory analysis [here](https://medium.com/the-esp-journal/esp32-memory-analysis-case-study-eacc75fe5431) can be useful as reference.

Mahavir

rhr407, 2020-11-09:
I am surprised that I could not find the data type 'heap_task_stat_t' anywhere in files of esp-idf folder and compiler giving me the following error.

```
esp-idf/aws/CMakeFiles/__idf_aws.dir/aws.c.obj -MF esp-idf/aws/CMakeFiles/__idf_aws.dir/aws.c.obj.d -o esp-idf/aws/CMakeFiles/__idf_aws.dir/aws.c.obj   -c ../components/aws/aws.c
../components/aws/aws.c: In function 'esp_dump_per_task_heap_info':
../components/aws/aws.c:33:5: error: unknown type name 'heap_task_stat_t'; did you mean 'esp_eth_state_t'?
     heap_task_stat_t tstat = {};
     ^~~~~~~~~~~~~~~~
     esp_eth_state_t
../components/aws/aws.c:33:30: error: empty scalar initializer
     heap_task_stat_t tstat = {};
                              ^
../components/aws/aws.c:33:30: note: (near initialization for 'tstat')
../components/aws/aws.c:38:22: error: implicit declaration of function 'heap_caps_get_next_task_stat'; did you mean 'heap_caps_get_total_size'? [-Werror=implicit-function-declaration]
         size_t ret = heap_caps_get_next_task_stat(&tstat, begin);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                      heap_caps_get_total_size
../components/aws/aws.c:43:38: error: request for member 'task' in something not a structure or union
         const char *task_name = tstat.task ? pcTaskGetTaskName(tstat.task) : "Pre-Scheduler allocs";
                                      ^
../components/aws/aws.c:43:69: error: request for member 'task' in something not a structure or union
         const char *task_name = tstat.task ? pcTaskGetTaskName(tstat.task) : "Pre-Scheduler allocs";
                                                                     ^
../components/aws/aws.c:46:36: error: request for member 'peak' in something not a structure or union
                    task_name, tstat.peak[0], tstat.peak[1]);
                                    ^
../components/aws/aws.c:46:51: error: request for member 'peak' in something not a structure or union
                    task_name, tstat.peak[0], tstat.peak[1]);
                                                   ^
../components/aws/aws.c: In function 'aws_shadow':
../components/aws/aws.c:153:31: warning: variable 'connectParams' set but not used [-Wunused-but-set-variable]
     IoT_Client_Connect_Params connectParams = iotClientConnectParamsDefault;
                               ^~~~~~~~~~~~~
../components/aws/aws.c:152:28: warning: variable 'mqttInitParams' set but not used [-Wunused-but-set-variable]
     IoT_Client_Init_Params mqttInitParams = iotClientInitParamsDefault;
                            ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

mahavirj, 2020-11-10:
@rhr407 

There is note in article about IDF repository pointer. Alternatively you may use example https://github.com/espressif/esp-idf/tree/master/examples/system/heap_task_tracking. Also please use APIs `esp_get_free_heap_size`, `esp_get_minimum_free_heap_size` to log heap statistics.

mahavirj, 2020-11-30:
@rhr407 Any update here? Can we close the issue?

rhr407, 2020-12-09:
I am really sorry for not responding in a while. I will put the code on test and let you know in a couple of days.

rhr407, 2020-12-15:
I would request to please close this issue.

closed 2020-12-15 by rhr407