espressif/esp-idf #6000: ESP32 unable to send data to SIM7600CE modem

chaisme, 2020-10-18:
I have been trying to send and receive GPS data using SIM7600 Modem (by waveshare) and have been unable to either send/receive anything over UART. Initial baud rate set is 115200. I have tried changing it to 9600 as well, but it doesn't help. I can see the commands being printed on the console but they aren't changing anything within the SIM7600.
For example, ATE0 should have stopped the modem from echoing but that didn't happen. 
Code snippets-
   uart_write_bytes(UART_NUM_0, "ATE0\r\n", 4);
    vTaskDelay(2000/portTICK_PERIOD_MS);

Is this the correct way to send data? 

I am able to get easily from a NEO-6M GPS module but with the modem have been unable to do so.

igrr, 2020-10-18:
"ATE0\r\n" is 6 bytes long, not 4?

chaisme, 2020-10-18:
My bad. This was a modified byte length as a part of another hopeless
attempt. I have been trying all this while with 6 bytes all this time.
shivi mishra

suda-morris, 2021-04-14:
Hi @chaisme 
esp-idf has SIM7600 support https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client/components/modem/src and in there, we disable the echo functionality by sending `ATE0\r`, without `\n`. https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/esp_modem_dce_service.c#L209
If you want to use SIM7600 only for the GPS, you can also take a look at [this example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/nmea0183_parser). 

For now, we will close this issue since it's not related to any esp-idf driver bug. Feel free to open another one with reproducible steps if you still have troubles in using PPPoS functionality or ask in [forum](https://esp32.com/viewforum.php?f=23&sid=bb070746471fd95ab4de54133124b155) where more people have experience with GPS module.

closed 2021-04-14 by suda-morris