espressif/esp-idf #6001: Wi-Fi Provisioning example "manager"

Zeni241, 2020-10-18:
I am working on provisioning example “manager” (https://github.com/espressif/esp-‎idf/t ... ng/manager ).‎
On device side I have flashed the above code with security 0. ‎
On client side I am using “Web Bluetooth API” to interact with device from browser. ‎
I can connect to the device and get Descriptor value. The response I get in client is as below:‎
`Requesting Bluetooth Device...‎
‎> Name: PROV_211114‎
‎> Id: EWxPB84jlYAHqvjJlt1drA==‎
‎> Connected: false‎
Connecting to GATT Server...‎
Getting Custom Service...‎
Getting Custom Level Characteristic...‎
Getting Custom Level Desriptor...‎
Descritor value=prov-config ‎`

But **I don’t know how to send my SSID and Password to the device OR Read characteristic values**.  Should I write to characteristic or ‎descriptor? ‎
If I write to descriptor:‎
`let encoder = new TextEncoder('utf-8');‎
‎     let userDescription = encoder.encode('ssid:myssid ; password:mypassword');‎
‎     descriptor.writeValue(userDescription);‎`

I get the error: ble.html:1 Uncaught (in promise) DOMException: GATT operation not permitted
If I write to characteristic:‎
`Let encoder = new TextEncoder('utf-8');‎
‎     let userDescription = encoder.encode('ssid:myssid ; password:mypassword');‎
‎     characteristic.writeValue(userDescription);‎`
I get the error on device side: ‎
E (803534) WiFiProvConfig: Unable to unpack config data
E (803534) protocomm_nimble: Invalid content received, killing connection‎
Any help please.‎

prasad-alatkar, 2020-10-20:
Hi @Zeni241 I believe you are referring to [wifi_prov_mgr](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) example.  For client side implementation/steps, you may refer to [console-mode](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr#unsupported-platform).  The `wifi_prov_mgr` example can also help you get wifi scan results to choose AP from the list of scanned APs, ref: [wifi-scan-results](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr#wi-fi-scanning). We also have `esp-prov` tool (found under $IDF_PATH/tools/esp_prov) to demonstrate client side implementation. So you may try to use `esp-prov` in console mode and follow the very same steps for your "Web Bluetooth API". Please let me know if it helps you resolve your issue.

shahpiyushv, 2020-10-20:
@Zeni241 , just to add to what @prasad-alatkar has already mentioned, note that the client to device communication uses protobuf and not plain text. You can check some additional documentation and pointers [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#user-side-implementation).

Zeni241, 2020-10-20:
Thanks @prasad-alatkar and @shahpiyushv . This gives me a clue to proceed. I will let you know after trying this solution.

closed 2020-12-18 by Alvin1Zhang