espressif/esp-idf #6003: modbus_slave example compile error in release/v4.0 branch if CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT=y

AxelLin, 2020-10-19:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.0.1-441-g6093407d7876
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Got below compile error if CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT=y

/home/axel/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c: In function 'mbc_serial_slave_start':
/home/axel/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: error: 'MB_SLAVE_ID_SHORT' undeclared (first use in this function); did you mean 'MB_SLAVE_ASSERT'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                            ^~~~~~~~~~~~~~~~~
                            MB_SLAVE_ASSERT
/home/axel/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:28: note: each undeclared identifier is reported only once for each function it appears in
/home/axel/esp/esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c:99:61: error: 'mb_slave_id' undeclared (first use in this function); did you mean 'mbc_slave_init'?
     status = eMBSetSlaveID(MB_SLAVE_ID_SHORT, TRUE, (UCHAR*)mb_slave_id, sizeof(mb_slave_id));
                                                             ^~~~~~~~~~~
                                                             mbc_slave_init
[627/680] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/serial_master/modbus_controller/mbc_serial_master.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

### Steps to reproduce

cd examples/protocols/modbus_slave
idf.py menuconfig (Set CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT=y)
idf.py build

Then got compile error.

CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT is default Y in newer ESP-IDF version.
So I notice this issue when I build my application in newer SDK first (compile ok),
then switch to older v4.0 release it compile failed.

alisitsyn, 2020-10-19:
@AxelLin,

This is known issue in release/v4.0. The release/v4.0 is already contain all known fixes but still not merged yet. It is planned to be merged in release v4.0.3. I can provide you accomulated patch for the v4.0 to fix the issue.

AxelLin, 2020-10-19:
@alisitsyn 

That's fine as long as it will be fixed, no hurry.
But I'm wondering if you mean v4.0.2 or v4.0.3? (next v4.0.x release should be v4.0.2)

alisitsyn, 2020-10-19:
@AxelLin 

This fix as well as other fixes will be merged in v4.0.3!

alisitsyn, 2020-10-19:
@AxelLin,

To avoid compilation issues you can disable the CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT. This option is disabled by default in release/v4.0. Let me know if you need modbus patch for the v4.0.

AxelLin, 2020-10-22:
Hi @alisitsyn 
I can fix the compilation issue by myself.
But I think you can upload the patch in case if someone else still hit the same issue.

alisitsyn, 2020-10-22:
Hi @AxelLin,

I agree, the link to the patch for v4.0 is below. Thanks.

https://esp32.com/viewtopic.php?f=13&t=17412&p=66731#p66731

AxelLin, 2021-01-20:
@alisitsyn 
Just to remind v4.0.2-206-g5630b17e1144 still has the build error.
AFAICT, a compilation fix should not take such long time to merge.

alisitsyn, 2021-01-21:
Hi @AxelLin,

I do have the MR which fixes this together with some other fix. However there is some work related to new chip support and merge of my MRs is posponed again. The v4.0 branch is frozen now. Please be patient or let me know if you would like to propose other solution.
Thanks.

AxelLin, 2021-01-21:
My point is simple, a compile error should be fix ASAP.
You probably need to send a seperate MR for build fix only.
I don't think anyone will pospone a compilation fix.

alisitsyn, 2021-01-21:
Done. Still need to wait when maintainer will merge it.

alisitsyn, 2021-01-25:
The fix is merged.

AxelLin, 2021-02-24:
I'm not sure what's wrong, but the fix is not yet available after you replied it's merged 1 month ago.

alisitsyn, 2021-02-24:
AxelLin, 
The fix will be replicated automatically to Github once the CI tests pass on master. The  commit 8e8e54737d6f5a591ee4517c701f08e418a33214  was merged on 25.01.2021. If it is urgent for you I can ask integrator to merge this manually.

closed 2021-02-24 by AxelLin