espressif/esp-idf #6004: Support compilation from 64bit ARM

galah92, 2020-10-19:
I'm using Raspberry Pi 4 as a development platform. Just switching to Ubuntu 64bit (from Raspberry Pi OS which is 32bit) and found out there's no toolchain supported for it. I'm not familiar with toolchains portability, but is it possible to port the current one to work on 64bit ARM?

Thanks

igrr, 2020-10-21:
Hi @galah92, it should be possible. We will consider adding such builds in the next toolchain release, but I can't give you exact timeline when this will happen. For now you can try the steps for building the toolchain locally: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup-scratch.html#setup-linux-toolchain-from-scratch.

galah92, 2020-10-22:
> Hi @galah92, it should be possible. We will consider adding such builds in the next toolchain release, but I can't give you exact timeline when this will happen. For now you can try the steps for [...]

Thank you. Would be great if you could ping me (here?) when there are updates.

Alvin1Zhang, 2022-11-21:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/00077545bf23134379133cfd979aaad35d5f3552, feel free to reopen.

closed 2022-11-21 by Alvin1Zhang