espressif/esp-idf #6005: PPP re-initialization fails

kannannair084, 2020-10-20:
I am handling a reconnect of PPP which esp_modem_setup_ppp(dte) and esp_modem_exit_ppp(dte)
i am getting stuck at 
esp_dte->ppp = pppapi_pppos_create(&(esp_dte->pppif), pppos_low_level_output, on_ppp_status_changed, dte);

could someone please recommend a proper init and de-init sequence for PPP.

Alvin1Zhang, 2020-10-20:
@kannannair084 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

kannannair084, 2020-10-20:
esp_err_t esp_modem_setup_ppp(modem_dte_t *dte)
{
    modem_dce_t *dce = dte->dce;
    MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
    /* Set PDP Context */
    MODEM_CHECK(dce->define_pdp_context(dce, 1, "IP", CONFIG_EXAMPLE_MODEM_APN) == ESP_OK, "set MODEM APN failed", err);
    /* Enter PPP mode */
    MODEM_CHECK(dte->change_mode(dte, MODEM_PPP_MODE) == ESP_OK, "enter ppp mode failed", err);
    /* Create PPPoS interface */
    printf("PPPOS create \n");
    esp_dte->ppp = pppapi_pppos_create(&(esp_dte->pppif), pppos_low_level_output, on_ppp_status_changed, dte);
    MODEM_CHECK(esp_dte->ppp, "create pppos interface failed", err);
#if PPP_NOTIFY_PHASE
    ppp_set_notify_phase_callback(esp_dte->ppp, on_ppp_notify_phase);
#endif
    /* Initiate PPP client connection */
    /* Set default route */
    printf("PPPOS set default \n");
    MODEM_CHECK(pppapi_set_default(esp_dte->ppp) == ERR_OK, "set default route failed", err);
    /* Ask the peer for up to 2 DNS server addresses */
    ppp_set_usepeerdns(esp_dte->ppp, 1);
    /* Auth configuration */
#if PAP_SUPPORT
    pppapi_set_auth(esp_dte->ppp, PPPAUTHTYPE_PAP, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);
#elif CHAP_SUPPORT
    pppapi_set_auth(esp_dte->ppp, PPPAUTHTYPE_CHAP, CONFIG_ESP_MODEM_PPP_AUTH_USERNAME, CONFIG_ESP_MODEM_PPP_AUTH_PASSWORD);
#else
#error "Unsupported AUTH Negotiation"
#endif
    /* Initiate PPP negotiation, without waiting */
    printf("PPPOS connect \n");
    MODEM_CHECK(pppapi_connect(esp_dte->ppp, 0) == ERR_OK, "initiate ppp negotiation failed", err);
    esp_event_post_to(esp_dte->event_loop_hdl, ESP_MODEM_EVENT, MODEM_EVENT_PPP_START, NULL, 0, 0);
    return ESP_OK;
err:
    return ESP_FAIL;
}

esp_err_t esp_modem_exit_ppp(modem_dte_t *dte)
{
    modem_dce_t *dce = dte->dce;
    MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
    /* Shutdown of PPP protocols */
    MODEM_CHECK(pppapi_close(esp_dte->ppp, 0) == ERR_OK, "close ppp connection failed", err);
    /* Enter command mode */
    MODEM_CHECK(dte->change_mode(dte, MODEM_COMMAND_MODE) == ESP_OK, "enter command mode failed", err);
    /* Hang up */
    MODEM_CHECK(dce->hang_up(dce) == ESP_OK, "hang up failed", err);
    return ESP_OK;
err:
    return ESP_FAIL;
}

kannannair084, 2020-10-20:
[image]

kannannair084, 2020-10-20:
i think this issue is very similar to another thread 
https://github.com/espressif/esp-idf/issues/4268

just that, not clear what needs to be done to resolve the issue. I am currently using IDF v4.0.1, do i need to update IDF?

kannannair084, 2020-10-21:
Also, tried to change IDF to version 4.2 and i am  not able to get the IP address

entry 0x40080688
I (28) boot: ESP-IDF v4.2-beta1-227-gf0e87c933 2nd stage bootloader
I (28) boot: compile time 09:51:10
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18e74 (102004) map
I (152) esp_image: segment 1: paddr=0x00028e9c vaddr=0x3ffb0000 size=0x02194 (  8596) load
I (156) esp_image: segment 2: paddr=0x0002b038 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/xtensa_vectors.S:1730

I (159) esp_image: segment 3: paddr=0x0002b444 vaddr=0x40080404 size=0x04bd4 ( 19412) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x631a0 (405920) map
0x400d0020: _stext at ??:?

I (331) esp_image: segment 5: paddr=0x000931c8 vaddr=0x40084fd8 size=0x0676c ( 26476) load
0x40084fd8: panic_print_dec at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/panic.c:101

I (349) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (350) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (358) cpu_start: Project name:     pppos_client
I (363) cpu_start: App version:      1
I (368) cpu_start: Compile time:     Oct 21 2020 09:50:33
I (374) cpu_start: ELF file SHA256:  21e1d8438c072f35...
I (380) cpu_start: ESP-IDF:          v4.2-beta1-227-gf0e87c933
I (386) cpu_start: Starting app cpu, entry point is 0x400818e4
0x400818e4: call_start_cpu1 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB3A20 len 0002C5E0 (177 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 4008B744 len 000148BC (82 KiB): IRAM
I (435) cpu_start: Pro cpu start user code
I (453) spi_flash: detected chip: gd
I (454) spi_flash: flash io: dio
I (454) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (464) uart: queue free spaces: 30
I (1624) pppos_example: Module: BG96
I (1624) pppos_example: Operator: "Telstra Mobile SORACOM"
I (1624) pppos_example: IMEI: 862785041881510
I (1624) pppos_example: IMSI: 295050911339686
I (1824) pppos_example: rssi: 15, ber: 99
I (2024) pppos_example: Battery voltage: 3857 mV
I (2224) pppos_example: Modem PPP Started
I (74374) esp-netif_lwip-ppp: Connection lost
I (74374) pppos_example: Modem Disconnect from PPP Server

This is based on the example ppos_client

kannannair084, 2020-10-21:
enabled the lwip debug and below are the logs

ppp phase changed[1]: phase=0
E (1346) esp-modem: esp_dte_handle_line(91): handle line failed
W (1346) pppos_example: Unknow line received: +COPS: 0

I (1646) pppos_example: Module: BG96
I (1646) pppos_example: Operator:
I (1646) pppos_example: IMEI: 862785041881510
I (1646) pppos_example: IMSI: 295050911339686
I (1846) pppos_example: rssi: 12, ber: 99
I (2046) pppos_example: Battery voltage: 3855 mV
I (2246) pppos_example: Modem PPP Started
ppp_connect[1]: holdoff=0
ppp phase changed[1]: phase=3
pppos_connect: unit 1: connecting
ppp_start[1]
ppp phase changed[1]: phase=6
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=FF FF FF FF
ppp_recv_config[1]
ppp: auth protocols: PAP=1 CHAP=0 CHAP_MD5=0 CHAP_MS=0 CHAP_MS2=0
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xbe6aeb27> <pcomp> <accomp>]
pppos_write[1]: len=24
ppp_start[1]: finished
pppos_input[1]: got 99 bytes
rcvd [LCP ConfReq id=0xc <asyncmap 0x0> <auth chap MD5> <magic 0x88229fe2> <pcomp> <accomp>]
sent [LCP ConfNak id=0xc <auth pap>]
pppos_write[1]: len=12
rcvd [LCP ConfAck id=0x1 <asyncmap 0x0> <magic 0xbe6aeb27> <pcomp> <accomp>]
pppos_input[1]: got 51 bytes
rcvd [LCP ConfReq id=0xd <asyncmap 0x0> <auth pap> <magic 0x88229fe2> <pcomp> <accomp>]
sent [LCP ConfAck id=0xd <asyncmap 0x0> <auth pap> <magic 0x88229fe2> <pcomp> <accomp>]
pppos_write[1]: len=28
netif_set_mtu[1]: mtu=1500
pppos_send_config[1]: out_accm=0 0 0 0
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
ppp phase changed[1]: phase=7
sent [PAP AuthReq id=0x1 user="sora" password="sora"]
pppos_write[1]: len=18
pppos_input[1]: got 32 bytes
rcvd [LCP DiscReq id=0xe magic=0x88229fe2]
rcvd [PAP AuthAck id=0x1 ""]
PAP authentication succeeded
ppp phase changed[1]: phase=9
ccp_set[1]: is_open=1, is_up=0, receive_method=0, transmit_method=0
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
pppos_input[1]: got 14 bytes
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[1]: len=32
IPCP: timeout sending Config-Requests
ppp phase changed[1]: phase=11
ccp_set[1]: is_open=0, is_up=0, receive_method=0, transmit_method=0
ppp phase changed[1]: phase=6
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
sent [LCP TermReq id=0x2 "No network protocols running"]
pppos_write[1]: len=36
sent [LCP TermReq id=0x3 "No network protocols running"]
pppos_write[1]: len=36
ppp phase changed[1]: phase=12
Connection terminated.
ppp_link_terminated[1]
ppp_link_end[1]
ppp phase changed[1]: phase=0
I (74466) esp-netif_lwip-ppp: Connection lost
I (74466) pppos_example: Modem Disconnect from PPP Server
ppp_link_terminated[1]: finished.

kannannair084, 2020-10-21:
I (29) boot: ESP-IDF v4.2-beta1-227-gf0e87c933 2nd stage bootloader
I (29) boot: compile time 11:09:36
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1a0c0 (106688) map
I (154) esp_image: segment 1: paddr=0x0002a0e8 vaddr=0x3ffb0000 size=0x035bc ( 13756) load
I (160) esp_image: segment 2: paddr=0x0002d6ac vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/xtensa_vectors.S:1730

I (161) esp_image: segment 3: paddr=0x0002dab8 vaddr=0x40080404 size=0x02560 (  9568) load
I (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x694cc (431308) map
0x400d0020: _stext at ??:?

I (343) esp_image: segment 5: paddr=0x000994f4 vaddr=0x40082964 size=0x09068 ( 36968) load
0x40082964: _xt_medint3 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/xtensa_vectors.S:1274

I (366) boot: Loaded app from partition at offset 0x10000
I (366) boot: Disabling RNG early entropy source...
I (367) cpu_start: Pro cpu up.
I (370) cpu_start: Application information:
I (375) cpu_start: Project name:     pppos_client
I (380) cpu_start: App version:      1
I (385) cpu_start: Compile time:     Oct 21 2020 11:08:51
I (391) cpu_start: ELF file SHA256:  8ddacdba63411b99...
I (397) cpu_start: ESP-IDF:          v4.2-beta1-227-gf0e87c933
I (403) cpu_start: Starting app cpu, entry point is 0x4008192c
0x4008192c: call_start_cpu1 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:287

I (396) cpu_start: App cpu up.
I (414) heap_init: Initializing. RAM available for dynamic allocation:
I (421) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (427) heap_init: At 3FFB5610 len 0002A9F0 (170 KiB): DRAM
I (433) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (440) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (446) heap_init: At 4008B9CC len 00014634 (81 KiB): IRAM
I (452) cpu_start: Pro cpu start user code
I (470) spi_flash: detected chip: gd
I (471) spi_flash: flash io: dio
I (471) esp_core_dump_uart: Init core dump to UART
I (473) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (488) uart: queue free spaces: 30
E (998) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (998) dce_service: esp_modem_dce_sync(48): send command failed
E (998) bg96: bg96_init(461): sync failed
assertion "dce != NULL" failed: file "../main/pppos_client_main.c", line 260, function: app_main

abort() was called at PC 0x40120e3b on core 0
0x40120e3b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x40085013:0x3ffb7340 0x40085701:0x3ffb7360 0x4008959a:0x3ffb7380 0x40120e3b:0x3ffb73f0 0x400d6c05:0x3ffb7420 0x400d536a:0x3ffb7550 0x40085709:0x3ffb7580
0x40085013: panic_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/panic.c:330

0x40085701: esp_system_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/system_api.c:106

0x4008959a: abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/newlib/abort.c:46

0x40120e3b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

[... 167 lines cut ...]
.iram0.data 0x4008b9cc 0x0 RW
.iram0.bss 0x4008b9cc 0x0 RW
.dram0.heap_start 0x3ffb5610 0x0 RW
.coredump.tasks.data 0x3ffb7608 0x168 RW
.coredump.tasks.data 0x3ffb7280 0x380 RW
.coredump.tasks.data 0x3ffb84e8 0x168 RW
.coredump.tasks.data 0x3ffb8340 0x1a0 RW
.coredump.tasks.data 0x3ffb7d78 0x168 RW
.coredump.tasks.data 0x3ffb7bd0 0x1a0 RW
.coredump.tasks.data 0x3ffba744 0x168 RW
.coredump.tasks.data 0x3ffba510 0x22c RW
.coredump.tasks.data 0x3ffbeac8 0x168 RW
.coredump.tasks.data 0x3ffbe8f0 0x1d0 RW
.coredump.tasks.data 0x3ffb8f54 0x168 RW
.coredump.tasks.data 0x3ffb8d80 0x1cc RW
.coredump.tasks.data 0x3ffafa8c 0x168 RW
.coredump.tasks.data 0x3ffaf8c0 0x1c4 RW
.coredump.tasks.data 0x3ffb5f48 0x168 RW
.coredump.tasks.data 0x3ffb5da0 0x1a0 RW
.coredump.tasks.data 0x3ffbb610 0x168 RW
.coredump.tasks.data 0x3ffbb400 0x208 RW
.coredump.tasks.data 0x3ffafe48 0x168 RW
.coredump.tasks.data 0x3ffb5860 0x1c8 RW

===================== ESP32 CORE DUMP END =====================

kannannair084, 2020-10-21:
I (29) boot: ESP-IDF v4.2-beta1-227-gf0e87c933 2nd stage bootloader
I (29) boot: compile time 11:09:36
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18e10 (101904) map
I (152) esp_image: segment 1: paddr=0x00028e38 vaddr=0x3ffb0000 size=0x035dc ( 13788) load
I (158) esp_image: segment 2: paddr=0x0002c41c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/xtensa_vectors.S:1730

I (159) esp_image: segment 3: paddr=0x0002c828 vaddr=0x40080404 size=0x037f0 ( 14320) load
I (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6456c (410988) map
0x400d0020: _stext at ??:?

I (333) esp_image: segment 5: paddr=0x00094594 vaddr=0x40083bf4 size=0x07dd8 ( 32216) load
0x40083bf4: spi_flash_mmap at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/spi_flash/flash_mmap.c:143

I (354) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (355) cpu_start: Pro cpu up.
I (359) cpu_start: Application information:
I (363) cpu_start: Project name:     pppos_client
I (369) cpu_start: App version:      1
I (373) cpu_start: Compile time:     Oct 21 2020 11:38:38
I (379) cpu_start: ELF file SHA256:  8b358d93eb67733f...
I (385) cpu_start: ESP-IDF:          v4.2-beta1-227-gf0e87c933
I (392) cpu_start: Starting app cpu, entry point is 0x4008192c
0x4008192c: call_start_cpu1 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:287

I (384) cpu_start: App cpu up.
I (402) heap_init: Initializing. RAM available for dynamic allocation:
I (409) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (415) heap_init: At 3FFB54B8 len 0002AB48 (170 KiB): DRAM
I (422) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (428) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (434) heap_init: At 4008B9CC len 00014634 (81 KiB): IRAM
I (441) cpu_start: Pro cpu start user code
I (459) spi_flash: detected chip: gd
I (459) spi_flash: flash io: dio
I (460) esp_core_dump_uart: Init core dump to UART
I (462) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (476) uart: queue free spaces: 30
I (1636) pppos_example: Module: BG96
I (1636) pppos_example: Operator: "Telstra Mobile SORACOM"
I (1636) pppos_example: IMEI: 862785041881510
I (1636) pppos_example: IMSI: 295050911339686
I (1836) pppos_example: rssi: 14, ber: 99
I (2036) pppos_example: Battery voltage: 3856 mV
I (2236) pppos_example: Modem PPP Started
I (2536) esp-netif_lwip-ppp: Connected
I (2536) esp-netif_lwip-ppp: Name Server1: 100.127.0.53
I (2536) esp-netif_lwip-ppp: Name Server2: 100.127.1.53
I (2536) pppos_example: Modem Connect to PPP Server
I (2546) pppos_example: ~~~~~~~~~~~~~~
I (2546) pppos_example: IP          : 10.170.33.19
I (2556) pppos_example: Netmask     : 255.255.255.255
I (2556) pppos_example: Gateway     : 10.64.64.64
I (2566) pppos_example: Name Server1: 100.127.0.53
I (2566) pppos_example: Name Server2: 100.127.1.53
[... 412 lines cut ...]
wCUIQPAbCEAwCgYA/IYIgNBX+z8AAAAAAAAAAIxB+z/wOv4/AwAAACMBBgDwGwiA
sFf7P9wA8D8BAAAAtBD7PyMABgABAAAAIwEGAAAAAAD//wAAAAAAAAAAAAAAAAAA
AAAAAMQnCEAjAAYA0FkIQFDK+T8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAMAlCEAEVwhAMAAFAAAAAACvMQiA8Ff7P+z9+j8AAAAA
AAAAAHwxCEAAAAAAAAAAAAxXCIAwWPs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
/////wAAAAAAAAAAAAAAADz++j8AAAAAAQAAAAIAAAAAAAAAUFj7P3wxCEAAAAAA
Iw8GAIBB+z9I/vo/AQAAAAAAAABwWPs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
6RgIgLA7/j/8Pvs/CAxAPwAAAAAAAAAAfFj7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAA==
FAAAAEgAAABKIAAA
RVNQX0NPUkVfRFVNUF9JTkZPAAA=
AQEAADhiMzU4ZDkzZWI2NzczM2YAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
DAAAAJQAAAClAgAA
RVhUUkFfSU5GTwAA
sHT7P+gAAAAdAAAA7gAAAAAAAADCAAAAAAAAAMMAAAAAAAAAxAAAAAAAAADFAAAA
AAAAAMYAAAAAAAAAsQAAAMMPE0CyAAAAAAAAALMAAAAAAAAAtAAAAAAAAAC1AAAA
AAAAALYAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
iIzjA7lD+C/JnNK0VlZO1AR0VkxFrlpyjAmFnXkoVow=
================= CORE DUMP END =================
Coredump SHA256='888ce303b943f82fc99cd2b456564ed40474564c45ae5a728c09859d7928568c'
I (11174) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...

kannannair084, 2020-10-21:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7260
load:0x40078000,len:13732
load:0x40080400,len:3988
0x40080400: _init at ??:?

entry 0x40080688
I (29) boot: ESP-IDF v4.2-beta1-227-gf0e87c933 2nd stage bootloader
I (29) boot: compile time 11:09:36
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18e10 (101904) map
I (152) esp_image: segment 1: paddr=0x00028e38 vaddr=0x3ffb0000 size=0x035dc ( 13788) load
I (158) esp_image: segment 2: paddr=0x0002c41c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/xtensa_vectors.S:1730

I (159) esp_image: segment 3: paddr=0x0002c828 vaddr=0x40080404 size=0x037f0 ( 14320) load
I (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6456c (410988) map
0x400d0020: _stext at ??:?

I (333) esp_image: segment 5: paddr=0x00094594 vaddr=0x40083bf4 size=0x07dd8 ( 32216) load
0x40083bf4: spi_flash_mmap at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/spi_flash/flash_mmap.c:143

I (354) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (355) cpu_start: Pro cpu up.
I (359) cpu_start: Application information:
I (363) cpu_start: Project name:     pppos_client
I (369) cpu_start: App version:      1
I (373) cpu_start: Compile time:     Oct 21 2020 11:38:38
I (379) cpu_start: ELF file SHA256:  8b358d93eb67733f...
I (385) cpu_start: ESP-IDF:          v4.2-beta1-227-gf0e87c933
I (392) cpu_start: Starting app cpu, entry point is 0x4008192c
0x4008192c: call_start_cpu1 at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:287

I (384) cpu_start: App cpu up.
I (402) heap_init: Initializing. RAM available for dynamic allocation:
I (409) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (415) heap_init: At 3FFB54B8 len 0002AB48 (170 KiB): DRAM
I (422) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (428) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (434) heap_init: At 4008B9CC len 00014634 (81 KiB): IRAM
I (441) cpu_start: Pro cpu start user code
I (459) spi_flash: detected chip: gd
I (459) spi_flash: flash io: dio
I (460) esp_core_dump_uart: Init core dump to UART
I (462) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (476) uart: queue free spaces: 30
I (1636) pppos_example: Module: BG96
I (1636) pppos_example: Operator: "Telstra Mobile SORACOM"
I (1636) pppos_example: IMEI: 862785041881510
[... 226 lines cut ...]
.coredump.tasks.data 0x3ffb8390 0x168 RW
.coredump.tasks.data 0x3ffb81e0 0x1a8 RW
.coredump.tasks.data 0x3ffb7c20 0x168 RW
.coredump.tasks.data 0x3ffb7a70 0x1a8 RW
.coredump.tasks.data 0x3ffbb3ec 0x168 RW
.coredump.tasks.data 0x3ffbb1c0 0x224 RW
.coredump.tasks.data 0x3ffbf5d8 0x168 RW
.coredump.tasks.data 0x3ffbf3c0 0x210 RW
.coredump.tasks.data 0x3ffb8dfc 0x168 RW
.coredump.tasks.data 0x3ffb8c30 0x1c4 RW
.coredump.tasks.data 0x3ffafa8c 0x168 RW
.coredump.tasks.data 0x3ffaf8c0 0x1c4 RW
.coredump.tasks.data 0x3ffb5df0 0x168 RW
.coredump.tasks.data 0x3ffb5c40 0x1a8 RW
.coredump.tasks.data 0x3ffbc3e8 0x168 RW
.coredump.tasks.data 0x3ffbc1e0 0x200 RW
.coredump.tasks.data 0x3ffafe48 0x168 RW
.coredump.tasks.data 0x3ffb5710 0x1c0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump SHA256='29db22facabe59548fff08f240fd3a7107b59f203db90ea8b3b9f49458b6a017'
I (11474) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...

kannannair084, 2020-10-21:
Narrowed down the issue to be that BG96 doesn't enter Command mode in the esp_modem_stop_ppp.

is there a way to fix this?

kannannair084, 2020-10-22:
@Alvin1Zhang could you help, our project has entered a critical juncture and this issue is impeding the progress.

david-cermak, 2020-10-22:
Hi @kannannair084 

Thanks for the report and your investigation, is this a random issue or does the happen consistently, all the time?
Possible reasons, why the modem couldn't not enter the command mode, are:
1) `+++` command is mistaken by data (there should be 1s delay before and after the command)
2) the modem is already in command mode (cause by a disconnection from PPP server)
3) the modem doesn't respond for some reason (reset or power line should be pulled down)

The current PPPoS example doesn't correctly address points 1) and 2), unfortunately. Could you please add this debugging line to `esp_modem.c` into the `esp_handle_uart_data()` function:
```
    if (length) {
        ESP_LOG_BUFFER_HEXDUMP(MODEM_TAG, esp_dte->buffer, length, ESP_LOG_INFO);
        esp_dte->receive_cb(esp_dte->buffer, length, esp_dte->receive_cb_ctx);
    }
```
So that we understand the issue better.
The output could be for example:
```
I (2875) pppos_example: PPP state changed event 256
D (2885) event: running post NETIF_PPP_STATUS:5 with handler 0x400d86bc and context 0x3ffbbb38 on loop 0x3ffbabb4
0x400d86bc: on_ppp_changed at /home/david/dev3/esp-idf/examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

I (2895) pppos_example: PPP state changed event 5
I (2905) pppos_example: User interrupted event from netif:0x3ffb6574
I (3955) esp-modem: 0x3ffbbbc8   0d 0a 4e 4f 20 43 41 52  52 49 45 52 0d 0a 0d 0a  |..NO CARRIER....|
I (3955) esp-modem: 0x3ffbbbd8   2b 50 50 50 44 3a 20 44  49 53 43 4f 4e 4e 45 43  |+PPPD: DISCONNEC|
I (3965) esp-modem: 0x3ffbbbe8   54 45 44 0d 0a                                    |TED..|
D (13755) bg96: Sending +++!
E (23755) esp-modem: esp_modem_dte_send_cmd(247): process command timeout
```
The above shows the issue where the device already exited data mode, sending commands, but the `esp_modem` still assumes data mode.

david-cermak, 2020-10-22:
Could you please test with the below patch? 

[0001-esp_modem-Fixed-stop_ppp-to-fist-exit-the-data-mode-.patch.txt](https://github.com/espressif/esp-idf/files/5423465/0001-esp_modem-Fixed-stop_ppp-to-fist-exit-the-data-mode-.patch.txt)

It works around the problem to first switch back to the command mode and then exit the networking mode of the PPP client. I've also added some debug prints and retries to better understand the issue.

kannannair084, 2020-10-23:
It failed but not showing the set mode change error any more:

I (38953) pppos_example: GOT ip event!!!
I (38963) system_api: Base MAC address is not set
I (38963) system_api: read default base MAC address from EFUSE
I (38973) pppos_example: MQTT other event id: 7
pppos_netif_output[0]: proto=0x21, len = 62
pppos_netif_output[0]: proto=0x21, len = 62
I (39763) esp-modem: 0x3ffbd924   7e 21 45 00 00                                    |~!E..|
pppos_input[0]: got 5 bytes
I (39813) esp-modem: 0x3ffbd924   4e 63 63 00 00 40 11 86  cb 64 7f 00 35 0a aa 21  |Ncc..@...d..5..!|
I (39813) esp-modem: 0x3ffbd934   13 00 35 3b b5 00 3a e5  8c 78 3b 81 80 00 01 00  |..5;..:..x;.....|
I (39813) esp-modem: 0x3ffbd944   01 00 00 00 00 04 6d 71  74 74 07 65 63 6c 69 70  |......mqtt.eclip|
I (39823) esp-modem: 0x3ffbd954   73 65 03 6f 72 67 00 00  01 00 01 c0 0c 00 01 00  |se.org..........|
I (39833) esp-modem: 0x3ffbd964   01 00 00 4e bf 00 04 89  87 53 d9 ec 88 7e        |...N.....S...~|
pppos_input[0]: got 78 bytes
ppp_input[0]: ip in pbuf len=78
pppos_netif_output[0]: proto=0x21, len = 44
I (39953) esp-modem: 0x3ffbd924   7e 21 45 00 00 4e 63 87  00 00 40 11 86 a7 64 7f  |~!E..Nc...@...d.|
I (39953) esp-modem: 0x3ffbd934   00 35 0a aa 21 13 00 35  3b b5 00 3a e5 8d 78 3b  |.5..!..5;..:..x;|
I (39953) esp-modem: 0x3ffbd944   81 80 00 01 00 01 00 00  00 00 04 6d 71 74 74 07  |...........mqtt.|
I (39963) esp-modem: 0x3ffbd954   65 63 6c 69 70 73 65 03  6f 72 67 00 00 01 00 01  |eclipse.org.....|
I (39973) esp-modem: 0x3ffbd964   c0 0c 00 01 00 01 00 00  4e be 00 04 89 87 53 d9  |........N.....S.|
I (39983) esp-modem: 0x3ffbd974   ce 95 7e                                          |..~|
pppos_input[0]: got 83 bytes
ppp_input[0]: ip in pbuf len=78
pppos_netif_output[0]: proto=0x21, len = 56
I (40353) esp-modem: 0x3ffbd924   7e 21 45 00 00 2c 00 00  40 00 16 06 5b af 89 87  |~!E..,..@...[...|
I (40353) esp-modem: 0x3ffbd934   53 d9 0a aa 21 13 07 5b  dc 0c c5 fc 0d 0c 00 00  |S...!..[........|
I (40353) esp-modem: 0x3ffbd944   19 6e 60 12 fa f0 c4 a3  00 00 02 04 05 3a d2 0e  |.n`..........:..|
I (40363) esp-modem: 0x3ffbd954   7e                                                |~|
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
pppos_netif_output[0]: proto=0x21, len = 40
pppos_netif_output[0]: proto=0x21, len = 66
I (41083) esp-modem: 0x3ffbd924   7e 21 45 00 00 28 e0 0c  40 00 16 06 7b a6 89 87  |~!E..(..@...{...|
I (41083) esp-modem: 0x3ffbd934   53 d9 0a aa 21 13 07 5b  dc 0c c5 fc 0d 0d 00 00  |S...!..[........|
I (41093) esp-modem: 0x3ffbd944   19 88 50 10 fa d6 db e6  00 00 50 b1 7e 7e 21 45  |..P.......P.~~!E|
I (41103) esp-modem: 0x3ffbd954   00 00 2c e0 0d 40 00 16  06 7b a1 89 87 53 d9 0a  |..,..@...{...S..|
I (41113) esp-modem: 0x3ffbd964   aa 21 13 07 5b dc 0c c5  fc 0d 0d 00 00 19 88 50  |.!..[..........P|
I (41113) esp-modem: 0x3ffbd974   18 fa d6 bb d8 00 00 20  02 00 00 d5 14 7e        |....... .....~|
pppos_input[0]: got 94 bytes
ppp_input[0]: ip in pbuf len=40
ppp_input[0]: ip in pbuf len=44
I (41143) pppos_example: MQTT_EVENT_CONNECTED
pppos_netif_output[0]: proto=0x21, len = 63
I (41143) pppos_example: sent subscribe successful, msg_id=49215
I (41733) esp-modem: 0x3ffbd924   7e 21 45 00 00 2d e0 0e  40 00 16 06 7b 9f 89 87  |~!E..-..@...{...|
I (41733) esp-modem: 0x3ffbd934   53 d9 0a aa 21 13 07 5b  dc 0c c5 fc 0d 11 00 00  |S...!..[........|
I (41733) esp-modem: 0x3ffbd944   19 9f 50 18 fa bf 8b 92  00 00 90 03 c0 3f 00 8b  |..P..........?..|
I (41743) esp-modem: 0x3ffbd954   37 7e                                             |7~|
pppos_input[0]: got 50 bytes
ppp_input[0]: ip in pbuf len=45
I (41763) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=49215
pppos_netif_output[0]: proto=0x21, len = 71
I (41773) pppos_example: sent publish successful, msg_id=0
I (42363) esp-modem: 0x3ffbd924   7e 21 45 00 00 47 e0 0f  40 00 16 06 7b 84 89 87  |~!E..G..@...{...|
I (42363) esp-modem: 0x3ffbd934   53 d9 0a aa 21 13 07 5b  dc 0c c5 fc 0d 16 00 00  |S...!..[........|
I (42363) esp-modem: 0x3ffbd944   19 be 50 18 fa a0 97 63  00 00 30 1d 00 10 2f 74  |..P....c..0.../t|
I (42373) esp-modem: 0x3ffbd954   6f 70 69 63 2f 65 73 70  2d 70 70 70 6f 73 65 73  |opic/esp-ppposes|
I (42383) esp-modem: 0x3ffbd964   70 33 32 2d 70 70 70 6f  73 c2 bb 7e              |p32-pppos..~|
pppos_input[0]: got 76 bytes
ppp_input[0]: ip in pbuf len=71
I (42403) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos
DATA=esp32-pppos
pppos_netif_output[0]: proto=0x21, len = 42
I (43003) esp-modem: 0x3ffbd924   7e 21 45 00 00 28 e0 10  40 00 16 06 7b a2 89 87  |~!E..(..@...{...|
I (43003) esp-modem: 0x3ffbd934   53 d9 0a aa 21 13 07 5b  dc 0c c5 fc 0d 35 00 00  |S...!..[.....5..|
I (43003) esp-modem: 0x3ffbd944   19 c0 50 11 fa 9e db bd  00 00 38 36 7e           |..P.......86~|
[... 31 lines cut ...]

Backtrace:0x40132816:0x3ffbb1d0 0x40132bfb:0x3ffbb1f0 0x400de225:0x3ffbb210 0x400f0570:0x3ffbb230 0x400f0759:0x3ffbb250 0x400efc1a:0x3ffbb280 0x400fadd7:0x3ffbb2a0 0x400fae2f:0x3ffbb2d0 0x400e71a5:0x3ffbb2f0 0x400deeda:0x3ffbb310 0x400def8c:0x3ffbb330 0x40085709:0x3ffbb360
0x40132816: esp_modem_dte_transmit at c:\olf\development\firmware\ppos\pppos_client\build/../components/modem/src/esp_modem_netif.c:41

0x40132bfb: esp_netif_transmit at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_netif/lwip/esp_netif_lwip.c:756

[... cut ...]
0x400fae2f: fsm_timeout at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/fsm.c:282

0x400e71a5: sys_check_timeouts at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/core/timeouts.c:400

0x400deeda: tcpip_timeouts_mbox_fetch at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/api/tcpip.c:115

0x400def8c: tcpip_thread at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/api/tcpip.c:148

0x40085709: vPortTaskWrapper at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:143

kannannair084, 2020-10-23:
Not sure why, i couldnt post it correctly, so attaching as text file:
[Log_and_dump.txt](https://github.com/espressif/esp-idf/files/5426955/Log_and_dump.txt)

kannannair084, 2020-10-23:
These are the modified functions as per patch , please double check if i have missed any or mis-placed it
```cpp
/* esp modem netif */
static esp_err_t esp_modem_dte_transmit(void *h, void *buffer, size_t len)
 {
     modem_dte_t *dte = h;
    if (dte->dce->mode == MODEM_COMMAND_MODE) 
	{
       return ESP_OK;
    }
    if (dte->send_data(dte, (const char *)buffer, len) > 0) 
	{
       return ESP_OK;
    }
	return ESP_FAIL;
 }

//esp_modem.c
esp_err_t esp_modem_stop_ppp(modem_dte_t *dte)
{
    modem_dce_t *dce = dte->dce;
    MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
   /* Fist enter the command mode */
     MODEM_CHECK(dte->change_mode(dte, MODEM_COMMAND_MODE) == ESP_OK, "enter command mode failed", err);
     /* Hang up */
     MODEM_CHECK(dce->hang_up(dce) == ESP_OK, "hang up failed", err);
   /* Next exit PPP network interface */
    /* post PPP mode stopped event */
    esp_event_post_to(esp_dte->event_loop_hdl, ESP_MODEM_EVENT, ESP_MODEM_EVENT_PPP_STOP, NULL, 0, 0);
    return ESP_OK;
err:
    return ESP_FAIL;
}

static esp_err_t esp_modem_dte_change_mode(modem_dte_t *dte, modem_mode_t new_mode)
{
    modem_dce_t *dce = dte->dce;
    MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
    esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
    MODEM_CHECK(dce->mode != new_mode, "already in mode: %d", err, new_mode);
    switch (new_mode) {
    case MODEM_PPP_MODE:
        MODEM_CHECK(dce->set_working_mode(dce, new_mode) == ESP_OK, "set new working mode:%d failed", err, new_mode);
        uart_disable_pattern_det_intr(esp_dte->uart_port);
        uart_enable_rx_intr(esp_dte->uart_port);
        break;
    case MODEM_COMMAND_MODE:
        uart_flush(esp_dte->uart_port);
		uart_enable_pattern_det_baud_intr(esp_dte->uart_port, '\n', 1, MIN_PATTERN_INTERVAL, MIN_POST_IDLE, MIN_PRE_IDLE);
		uart_disable_rx_intr(esp_dte->uart_port);        
        uart_pattern_queue_reset(esp_dte->uart_port, esp_dte->pattern_queue_size);
        dce->mode = MODEM_COMMAND_MODE;
        vTaskDelay(pdMS_TO_TICKS(1000));
        MODEM_CHECK(dce->set_working_mode(dce, new_mode) == ESP_OK, "set new working mode:%d failed", err, new_mode);
        vTaskDelay(pdMS_TO_TICKS(1000));
        break;
    default:
        break;
    }
    return ESP_OK;
err:
    return ESP_FAIL;
}

static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)
{
    if (esp_dte->parent.dce->mode != MODEM_PPP_MODE) {
       // ESP_LOGE(MODEM_TAG, "Error: Got data event in PPP mode");
        /* pattern detection mode -> ignore date event on uart
[... 80 lines cut ...]
			ESP_LOGD(DCE_TAG, "Sending +++ for %d time", retry);
			dce->handle_line = bg96_handle_exit_data_mode;
			DCE_CHECK(dte->send_cmd(dte, "+++", 20000) == ESP_OK, "send command failed", again);
			DCE_CHECK(dce->state == MODEM_STATE_SUCCESS, "enter command mode failed", err);
		}
        ESP_LOGD(DCE_TAG, "enter command mode ok");
        dce->mode = MODEM_COMMAND_MODE;
        break;
    case MODEM_PPP_MODE:
        dce->handle_line = bg96_handle_atd_ppp;
        DCE_CHECK(dte->send_cmd(dte, "ATD*99***1#\r", MODEM_COMMAND_TIMEOUT_MODE_CHANGE) == ESP_OK, "send command failed", err);
        DCE_CHECK(dce->state == MODEM_STATE_SUCCESS, "enter ppp mode failed", err);
        ESP_LOGD(DCE_TAG, "enter ppp mode ok");
        dce->mode = MODEM_PPP_MODE;
        break;
    default:
        ESP_LOGW(DCE_TAG, "unsupported working mode: %d", mode);
        goto err;
        break;
		
    }
    return ESP_OK;
err:
    return ESP_FAIL;
}

kannannair084, 2020-10-23:
ELF file SHA256: da60e7829b14a192

Initiating core dump!
I (25290) esp_core_dump_uart: Press Enter to print core dump to UART...
I (25299) esp_core_dump_uart: Print core dump to uart...
I (25303) esp_core_dump_elf: Found tasks: 10
Core dump started (further output muted)
Received  21 kB...
Core dump finished!
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffbb3ec, name: 'tiT', GDB name: 'process 1073460204'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1c (LoadProhibitedCause)
excvaddr       0x64
epc1           0x0
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
epc7           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
eps7           0x4013278f
pc             0x40132819       0x40132819 <esp_modem_dte_transmit+5>
lbeg           0x400014fd       1073747197
lend           0x4000150d       1073747213
lcount         0xfffffff5       4294967285
sar            0x10     16
ps             0x60420  394272
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80132bfe       -2146227202
a1             0x3ffbb1d0       1073459664
a2             0x3ffb6278       1073439352
a3             0x3ffbfae0       1073478368
a4             0x1d     29
a5             0x3ffbad90       1073458576
a6             0x20     32
a7             0x10     16
a8             0x0      0
a9             0x3ffbb1a0       1073459616
a10            0x3ffb4fa0       1073434528
a11            0x3ffae8f0       1073408240
a12            0x0      0
a13            0x3ffbf500       1073476864
a14            0x3ffbf4f0       1073476848
a15            0x1      1

==================== CURRENT THREAD STACK =====================
[... 114 lines cut ...]
.coredump.tasks.data 0x3ffbf178 0x168 RW
.coredump.tasks.data 0x3ffbef60 0x210 RW
.coredump.tasks.data 0x3ffb8390 0x168 RW
.coredump.tasks.data 0x3ffb81e0 0x1a8 RW
.coredump.tasks.data 0x3ffb7c20 0x168 RW
.coredump.tasks.data 0x3ffb7a70 0x1a8 RW
.coredump.tasks.data 0x3ffb74b0 0x168 RW
.coredump.tasks.data 0x3ffb71c0 0x2e8 RW
.coredump.tasks.data 0x3ffb8dfc 0x168 RW
.coredump.tasks.data 0x3ffb8c30 0x1c4 RW
.coredump.tasks.data 0x3ffafa8c 0x168 RW
.coredump.tasks.data 0x3ffaf8c0 0x1c4 RW
.coredump.tasks.data 0x3ffb5df0 0x168 RW
.coredump.tasks.data 0x3ffb5c40 0x1a8 RW
.coredump.tasks.data 0x3ffbc3e8 0x168 RW
.coredump.tasks.data 0x3ffbc1e0 0x200 RW
.coredump.tasks.data 0x3ffafe48 0x168 RW
.coredump.tasks.data 0x3ffb5710 0x1c0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump SHA256='216b7a7a1a0af98a174f0491d5d606254fd54acc3c02d2228bc0a1586c5763b7'
I (392) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...

kannannair084, 2020-10-23:
After a while running like this, it gets stuck 

[image]

ppp_recv_config[0]
sent [LCP TermReq id=0x2 "User request"]
pppos_write[0]: len=20
I (46863) pppos_example: Power down
I (46863) pppos_example: Restart after 60 seconds
sent [LCP TermReq id=0x3 "User request"]
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40132819  PS      : 0x00060430  A0      : 0x80132bfe  A1      : 0x3ffbb1d0
0x40132819: esp_modem_dte_transmit at c:\olf\development\firmware\ppos\pppos_client\build/../components/modem/src/esp_modem_netif.c:43

A2      : 0x3ffb6278  A3      : 0x3ffbfae0  A4      : 0x0000001d  A5      : 0x3ffbad90
A6      : 0x00000020  A7      : 0x00000010  A8      : 0x00000000  A9      : 0x3ffbb1a0
A10     : 0x3ffb4fa0  A11     : 0x3ffae8f0  A12     : 0x00000000  A13     : 0x3ffbf500
A14     : 0x3ffbf4f0  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000064  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5

Backtrace:0x40132816:0x3ffbb1d0 0x40132bfb:0x3ffbb1f0 0x400de225:0x3ffbb210 0x400f0570:0x3ffbb230 0x400f0759:0x3ffbb250 0x400efc1a:0x3ffbb280 0x400fadd7:0x3ffbb2a0 0x400fae2f:0x3ffbb2d0 0x400e71a5:0x3ffbb2f0 0x400deeda:0x3ffbb310 0x400def8c:0x3ffbb330 0x40085709:0x3ffbb360
0x40132816: esp_modem_dte_transmit at c:\olf\development\firmware\ppos\pppos_client\build/../components/modem/src/esp_modem_netif.c:41

0x40132bfb: esp_netif_transmit at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_netif/lwip/esp_netif_lwip.c:756

0x400de225: pppos_low_level_output at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_netif/lwip/esp_netif_lwip_ppp.c:233

0x400f0570: pppos_output_last at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/pppos.c:878

0x400f0759: pppos_write at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/pppos.c:244

0x400efc1a: ppp_write at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/ppp.c:993

0x400fadd7: fsm_sdata at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/fsm.c:796

0x400fae2f: fsm_timeout at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/netif/ppp/fsm.c:282

0x400e71a5: sys_check_timeouts at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/core/timeouts.c:400

0x400deeda: tcpip_timeouts_mbox_fetch at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/api/tcpip.c:115

0x400def8c: tcpip_thread at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/lwip/lwip/src/api/tcpip.c:148

0x40085709: vPortTaskWrapper at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:143

ELF file SHA256: da60e7829b14a192

Initiating core dump!
I (25590) esp_core_dump_uart: Press Enter to print core dump to UART...
I (25599) esp_core_dump_uart: Print core dump to uart...
I (25603) esp_core_dump_elf: Found tasks: 10
Core dump started (further output muted)
Received  21 kB...
Core dump finished!
Failed to run espcoredump script: Command '['C:\\OLF\\Development\\Firmware\\ESP-IDF\\tools_4_2\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe', 'C:\\OLF\\Development\\Firmware\\ESP-IDF\\esp-idf_4_2\\tools\\..\\components\\espcoredump\\espcoredump.py', 'info_corefile', '--core', 'C:\\Users\\SUMESH~1\\AppData\\Local\\Temp\\tmp9koor4iw', '--core-format', 'b64', 'c:\\olf\\development\\firmware\\ppos\\pppos_client\\build\\pppos_client.elf']' returned non-zero exit status 1.

================= CORE DUMP START =================
2D0AAAEBAAAKAAAAaAEAAAAAAAA=
f0VMRgEBAQAAAAAAAAAAAAQAXgABAAAAAAAAADQAAAAAAAAAAAAAADQAIAAWACgA
AAAAAA==
BAAAAPQCAAAAAAAAAAAAAMAXAADAFwAABgAAAAAAAAA=
AQAAALQaAADss/s/7LP7P2gBAABoAQAABgAAAAAAAAA=
AQAAABwcAAAQsfs/ELH7P9QCAADUAgAABgAAAAAAAAA=
AQAAAPAeAACQg/s/kIP7P2gBAABoAQAABgAAAAAAAAA=
AQAAAFggAADggfs/4IH7P6gBAACoAQAABgAAAAAAAAA=
AQAAAAAiAAAgfPs/IHz7P2gBAABoAQAABgAAAAAAAAA=
AQAAAGgjAABwevs/cHr7P6gBAACoAQAABgAAAAAAAAA=
AQAAABAlAACwdPs/sHT7P2gBAABoAQAABgAAAAAAAAA=
[... 523 lines cut ...]
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x1 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
IPCP: timeout sending Config-Requests
ppp phase changed[0]: phase=11
ppp phase changed[0]: phase=6
pppos_send_config[0]: out_accm=FF FF FF FF
ppp_send_config[0]
pppos_recv_config[0]: in_accm=0 0 0 0
ppp_recv_config[0]
sent [LCP TermReq id=0x2 "No network protocols running"]
pppos_write[0]: len=36
sent [LCP TermReq id=0x3 "No network protocols running"]
pppos_write[0]: len=36
ppp phase changed[0]: phase=12
Connection terminated.
ppp_link_terminated[0]
ppp_link_end[0]
ppp phase changed[0]: phase=0
I (110723) esp-netif_lwip-ppp: Connection lost
I (110723) pppos_example: Modem Disconnect from PPP Server
ppp_link_terminated[0]: finished.

kannannair084, 2020-10-23:
[image]

[image]

I am seeing that there is a power down message gets sent, and it is crashing after that?

[image]

Also, could this crash be triggered by POWER DOWN output from BG96?

kannannair084, 2020-10-23:
[Stack_overflow_core_dump.txt](https://github.com/espressif/esp-idf/files/5427660/Stack_overflow_core_dump.txt)

kannannair084, 2020-10-23:
[Exception_was-unhandle.txt](https://github.com/espressif/esp-idf/files/5427681/Exception_was-unhandle.txt)

kannannair084, 2020-10-23:
[pppos_client_main.txt](https://github.com/espressif/esp-idf/files/5427692/pppos_client_main.txt)

Please check if making any out of sequence calls or missing something

kannannair084, 2020-10-23:
I made the following change and it started working 

```cpp
  /* Exit PPP mode */
        ESP_ERROR_CHECK(esp_modem_stop_ppp(dte));
		xEventGroupWaitBits(event_group, STOP_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
		ESP_LOGI(TAG, "10 sec delay start");		
		 vTaskDelay(pdMS_TO_TICKS(10000));
		 ESP_LOGI(TAG, "10 sec delay end");	
   /* Power down module */
    ESP_ERROR_CHECK(dce->power_down(dce));
    ESP_LOGI(TAG, "Power down");
    ESP_ERROR_CHECK(dce->deinit(dce));	
    ESP_LOGI(TAG, "Restart after 60 seconds");
    vTaskDelay(pdMS_TO_TICKS(60000));

david-cermak, 2020-10-23:
> Also, could this crash be triggered by POWER DOWN output from BG96?

Not really, from the coredump it looks more like the ppp mode in lwip is still active and trying to send data to dce, which as been already destroyed in line
``` c
ESP_ERROR_CHECK(dce->deinit(dce))
```

it looks like the `+++` command workaround did work, but need to wait until netif really stops.

> I made the following change and it started working , basically added a 10 sec delay before powering down and after stop.

Yes, it makes sense, needed to wait for the lwip PPP netif to stop

---
Okay, thanks very much for testing this, now I understand the problem so that it could be properly fixed.

kannannair084, 2020-10-23:
however, after a couple of runs, i ran into this.
I (153653) pppos_example: Module: BG96
I (153653) pppos_example: Operator:
I (153653) pppos_example: IMEI: 862785041881510
I (153653) pppos_example: IMSI: 295050911339686
I (154103) pppos_example: rssi: 99, ber: 99
I (154453) pppos_example: Battery voltage: 3855 mV
I (154803) pppos_example: Modem PPP Started
ppp_connect[0]: holdoff=0
ppp phase changed[0]: phase=3
pppos_connect: unit 0: connecting
ppp_start[0]
ppp phase changed[0]: phase=6
pppos_send_config[0]: out_accm=FF FF FF FF
ppp_send_config[0]
pppos_recv_config[0]: in_accm=FF FF FF FF
ppp_recv_config[0]
ppp: auth protocols: PAP=1
sent [LCP ConfReq id=0x4 <asyncmap 0x0> <magic 0xd0131f3a> <pcomp> <accomp>]
pppos_write[0]: len=24
ppp_start[0]: finished
I (154883) esp-modem: 0x3ffbd924   7e ff 7d 23 c0 21 7d 21  7d 20 7d 20 7d 39 7d 22  |~.}#.!}!} } }9}"|
I (154883) esp-modem: 0x3ffbd934   7d 26 7d 20 7d 20 7d 20  7d 20 7d 23 7d 25 c2 23  |}&} } } } }#}%.#|
I (154893) esp-modem: 0x3ffbd944   7d 25 7d 25 7d 26 d0 84  69 7d 2a 7d 27 7d 22 7d  |}%}%}&..i}*}'}"}|
I (154903) esp-modem: 0x3ffbd954   28 7d 22 a0 fc 7e 7e ff  7d 23 c0 21 7d 22 7d 24  |(}"..~~.}#.!}"}$|
I (154913) esp-modem: 0x3ffbd964   7d 20 7d 34 7d 22 7d 26  7d 20 7d 20 7d 20 7d 20  |} }4}"}&} } } } |
I (154923) esp-modem: 0x3ffbd974   7d 25 7d 26 d0 7d 33 7d  3f 3a 7d 27 7d 22 7d 28  |}%}&.}3}?:}'}"}(|
I (154933) esp-modem: 0x3ffbd984   7d 22 7d 27 b9 7e                                 |}"}'.~|
pppos_input[0]: got 102 bytes
rcvd [LCP ConfReq id=0x0 <asyncmap 0x0> <auth 0xc223 05> <magic 0xd084690a> <pcomp> <accomp>]
sent [LCP ConfNak id=0x0 <auth pap>]
pppos_write[0]: len=12
rcvd [LCP ConfAck id=0x4 <asyncmap 0x0> <magic 0xd0131f3a> <pcomp> <accomp>]
I (154993) esp-modem: 0x3ffbd924   7e ff 7d 23 c0 21 7d 21  7d 21 7d 20 7d 38 7d 22  |~.}#.!}!}!} }8}"|
I (154993) esp-modem: 0x3ffbd934   7d 26 7d 20 7d 20 7d 20  7d 20 7d 23 7d 24 c0 23  |}&} } } } }#}$.#|
I (155003) esp-modem: 0x3ffbd944   7d 25 7d 26 d0 84 69 7d  2a 7d 27 7d 22 7d 28 7d  |}%}&..i}*}'}"}(}|
I (155003) esp-modem: 0x3ffbd954   22 b2 c9 7e                                       |"..~|
pppos_input[0]: got 52 bytes
rcvd [LCP ConfReq id=0x1 <asyncmap 0x0> <auth pap> <magic 0xd084690a> <pcomp> <accomp>]
sent [LCP ConfAck id=0x1 <asyncmap 0x0> <auth pap> <magic 0xd084690a> <pcomp> <accomp>]
pppos_write[0]: len=28
netif_set_mtu[0]: mtu=1500
pppos_send_config[0]: out_accm=0 0 0 0
ppp_send_config[0]
pppos_recv_config[0]: in_accm=0 0 0 0
ppp_recv_config[0]
ppp phase changed[0]: phase=7
sent [PAP AuthReq id=0x2 user="sora" password="sora"]
pppos_write[0]: len=18
I (155073) esp-modem: 0x3ffbd924   7e ff 7d 23 c0 21 7d 2b  7d 22 7d 20 7d 28 d0 84  |~.}#.!}+}"} }(..|
I (155073) esp-modem: 0x3ffbd934   69 7d 2a 8d d8 7e 7e c0  23 02 02 00 05 00 30 15  |i}*..~~.#.....0.|
I (155083) esp-modem: 0x3ffbd944   7e                                                |~|
pppos_input[0]: got 33 bytes
rcvd [LCP DiscReq id=0x2 magic=0xd084690a]
rcvd [PAP AuthAck id=0x2 ""]
PAP authentication succeeded
ppp phase changed[0]: phase=9
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
I (155143) esp-modem: 0x3ffbd924   0d 0a 4e 4f 20 43 41 52  52 49 45 52 0d 0a        |..NO CARRIER..|
pppos_input[0]: got 14 bytes
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
sent [IPCP ConfReq id=0x4 <compress VJ 0f 01> <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
pppos_write[0]: len=32
IPCP: timeout sending Config-Requests
ppp phase changed[0]: phase=11
ppp phase changed[0]: phase=6
pppos_send_config[0]: out_accm=FF FF FF FF
ppp_send_config[0]
pppos_recv_config[0]: in_accm=0 0 0 0
ppp_recv_config[0]
sent [LCP TermReq id=0x5 "No network protocols running"]
pppos_write[0]: len=36
sent [LCP TermReq id=0x6 "No network protocols running"]
pppos_write[0]: len=36
ppp phase changed[0]: phase=12
Connection terminated.
ppp_link_terminated[0]
ppp_link_end[0]
ppp phase changed[0]: phase=0
I (227123) esp-netif_lwip-ppp: User interrupt
ppp_link_terminated[0]: finished.

kannannair084, 2020-10-23:
it is waiting for the started bit  , i think here

xEventGroupWaitBits(event_group, GOT_DATA_BIT, pdTRUE, pdTRUE, portMAX_DELAY);

how to avoid this from happening, as my code gets stuck there.

david-cermak, 2020-10-23:
```
I (155143) esp-modem: 0x3ffbd924 0d 0a 4e 4f 20 43 41 52 52 49 45 52 0d 0a |..NO CARRIER..|
```
Seem like a connection problem with your modem here -- no connection, so way to get an IP address.

kannannair084, 2020-10-23:
Is this a correct approach, if not able to get the connect, shall i do stop_ppp or go straight to power down?
```cpp

    xEventGroupWaitBits(event_group, CONNECT_BIT, pdTRUE, pdTRUE, 70000 / portTICK_PERIOD_MS);
	if(xEventGroupGetBits(event_group, CONNECT_BIT) == 1 )
	{}
	else{
	 goto  exitppp;		
	}

-
-
-

 exitppp:  
        /* Exit PPP mode */
        ESP_ERROR_CHECK(esp_modem_stop_ppp(dte));
		xEventGroupWaitBits(event_group, STOP_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
		ESP_LOGI(TAG, "10 sec delay start");		
		 vTaskDelay(pdMS_TO_TICKS(10000));
		 ESP_LOGI(TAG, "10 sec delay end");	
   /* Power down module */
    ESP_ERROR_CHECK(dce->power_down(dce));

kannannair084, 2020-10-23:
I (41773) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=48260
pppos_netif_output[0]: proto=0x21, len = 71
I (41783) pppos_example: sent publish successful, msg_id=0
I (42373) esp-modem: 0x3ffbd92c   7e 21 45 00 00 47 3a 38  40 00 23 06 14 5c 89 87  |~!E..G:8@.#..\..|
I (42373) esp-modem: 0x3ffbd93c   53 d9 0a aa 21 13 07 5b  d0 29 82 06 f3 07 00 00  |S...!..[.)......|
I (42373) esp-modem: 0x3ffbd94c   19 be 50 18 fa a0 01 4b  00 00 30 1d 00 10 2f 74  |..P....K..0.../t|
I (42383) esp-modem: 0x3ffbd95c   6f 70 69 63 2f 65 73 70  2d 70 70 70 6f 73 65 73  |opic/esp-ppposes|
I (42393) esp-modem: 0x3ffbd96c   70 33 32 2d 70 70 70 6f  73 bb d5 7e              |p32-pppos..~|
pppos_input[0]: got 76 bytes
ppp_input[0]: ip in pbuf len=71
I (42413) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos
DATA=esp32-pppos
pppos_netif_output[0]: proto=0x21, len = 42
I (43013) esp-modem: 0x3ffbd92c   7e 21 45 00 00 28 3a 39  40 00 23 06 14 7a 89 87  |~!E..(:9@.#..z..|
I (43013) esp-modem: 0x3ffbd93c   53 d9 0a aa 21 13 07 5b  d0 29 82 06 f3 26 00 00  |S...!..[.)...&..|
I (43013) esp-modem: 0x3ffbd94c   19 c0 50 11 fa 9e 45 a5  00 00 d6 4e 7e           |..P...E....N~|

***ERROR*** A stack overflow in task uart_event has been detected.

Backtrace:0x40085013:0x3ffbedb0 0x40085701:0x3ffbedd0 0x400859a9:0x3ffbedf0 0x4008671d:0x3ffbee70 0x40085a9c:0x3ffbee90 0x40085a52:0x00000001 |<-CORRUPTED
0x40085013: panic_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/panic.c:330

0x40085701: esp_system_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/system_api.c:106

0x400859a9: vApplicationStackOverflowHook at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:436

0x4008671d: vTaskSwitchContext at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/tasks.c:2771

0x40085a9c: _frxt_dispatch at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/portasm.S:431

0x40085a52: _frxt_int_exit at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/portasm.S:231

ELF file SHA256: f3b33259a1a38b48

Initiating core dump!
I (16205) esp_core_dump_uart: Press Enter to print core dump to UART...
I (16215) esp_core_dump_uart: Print core dump to uart...
I (16219) esp_core_dump_elf: Found tasks: 11
Core dump started (further output muted)
Received  24 kB...
Core dump finished!
Failed to run espcoredump script: Command '['C:\\OLF\\Development\\Firmware\\ESP-IDF\\tools_4_2\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe', 'C:\\OLF\\Development\\Firmware\\ESP-IDF\\esp-idf_4_2\\tools\\..\\components\\espcoredump\\espcoredump.py', 'info_corefile', '--core', 'C:\\Users\\SUMESH~1\\AppData\\Local\\Temp\\tmprycxpjdd', '--core-format', 'b64', 'c:\\olf\\development\\firmware\\ppos\\pppos_client\\build\\pppos_client.elf']' returned non-zero exit status 1.

kannannair084, 2020-10-23:
@david-cermak could you take a look at the logs, i am just running it in a loop to see how reliable this is and i am still getting crashes. seems like miles away from destination and have a tight schedule to meet.

[crashing.txt](https://github.com/espressif/esp-idf/files/5428255/crashing.txt)

kannannair084, 2020-10-23:
It is crashing every 5th or 6th iteration... 

I (739323) pppos_example: GOT ip event!!!
I (739333) pppos_example: MQTT other event id: 7
pppos_netif_output[0]: proto=0x21, len = 44
I (740483) esp-modem: 0x3ffbd934   7e 21 45 00 00 2c 00 00  40 00 14 06 5d af 89 87  |~!E..,..@...]...|
I (740483) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  fa 68 2f e8 ff b5 00 00  |S...!..[.h/.....|
I (740493) esp-modem: 0x3ffbd954   1a 93 60 12 fa f0 48 8d  00 00 02 04 05 3a 89 21  |..`...H......:.!|
I (740503) esp-modem: 0x3ffbd964   7e                                                |~|
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
pppos_netif_output[0]: proto=0x21, len = 40
pppos_netif_output[0]: proto=0x21, len = 66
I (741113) esp-modem: 0x3ffbd934   7e 21 45 00 00 28 58 9c  40 00 14 06 05 17 89 87  |~!E..(X.@.......|
I (741113) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  fa 68 2f e8 ff b6 00 00  |S...!..[.h/.....|
I (741123) esp-modem: 0x3ffbd954   1a ad 50 10 fa d6 5f d0  00 00 21 7b 7e 7e 21 45  |..P..._...!{~~!E|
I (741133) esp-modem: 0x3ffbd964   00 00 2c 58 9d 40 00 14  06 05 12 89 87 53 d9 0a  |..,X.@.......S..|
I (741143) esp-modem: 0x3ffbd974   aa 21 13 07 5b fa 68 2f  e8 ff b6 00 00 1a ad 50  |.!..[.h/.......P|
I (741153) esp-modem: 0x3ffbd984   18 fa d6 3f c2 00 00 20  02 00 00 14 79 7e        |...?... ....y~|
pppos_input[0]: got 94 bytes
ppp_input[0]: ip in pbuf len=40
ppp_input[0]: ip in pbuf len=44
I (741173) pppos_example: MQTT_EVENT_CONNECTED
pppos_netif_output[0]: proto=0x21, len = 63
I (741173) pppos_example: sent subscribe successful, msg_id=29658
I (741763) esp-modem: 0x3ffbd934   7e 21 45 00 00 2d 58 9e  40 00 14 06 05 10 89 87  |~!E..-X.@.......|
I (741763) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  fa 68 2f e8 ff ba 00 00  |S...!..[.h/.....|
I (741763) esp-modem: 0x3ffbd954   1a c4 50 18 fa bf 5b e1  00 00 90 03 73 da 00 d2  |..P...[.....s...|
I (741773) esp-modem: 0x3ffbd964   e3 7e                                             |.~|
pppos_input[0]: got 50 bytes
ppp_input[0]: ip in pbuf len=45
I (741793) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=29658
pppos_netif_output[0]: proto=0x21, len = 71
I (741803) pppos_example: sent publish successful, msg_id=0
I (742393) esp-modem: 0x3ffbd934   7e 21 45 00 00 47 58 9f  40 00 14 06 04 f5 89 87  |~!E..GX.@.......|
I (742393) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  fa 68 2f e8 ff bf 00 00  |S...!..[.h/.....|
I (742403) esp-modem: 0x3ffbd954   1a e3 50 18 fa a0 1b 4d  00 00 30 1d 00 10 2f 74  |..P....M..0.../t|
I (742413) esp-modem: 0x3ffbd964   6f 70 69 63 2f 65 73 70  2d 70 70 70 6f 73 65 73  |opic/esp-ppposes|
I (742423) esp-modem: 0x3ffbd974   70 33 32 2d 70 70 70 6f  73 90 13 7e              |p32-pppos..~|

***ERROR*** A stack overflow in task uart_event has been detected.

Backtrace:0x40085013:0x3ffbedc0 0x40085701:0x3ffbede0 0x400859a9:0x3ffbee00 0x4008671d:0x3ffbee80 0x40085a9c:0x3ffbeea0 0x40085a52:0x00000001 |<-CORRUPTED
0x40085013: panic_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/panic.c:330

0x40085701: esp_system_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/system_api.c:106

0x400859a9: vApplicationStackOverflowHook at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:436

0x4008671d: vTaskSwitchContext at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/tasks.c:2771

0x40085a9c: _frxt_dispatch at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/portasm.S:431

0x40085a52: _frxt_int_exit at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/portasm.S:231

ELF file SHA256: dd787fca6e349398

kannannair084, 2020-10-23:
[6th_round_crashing.txt](https://github.com/espressif/esp-idf/files/5428383/6th_round_crashing.txt)

kannannair084, 2020-10-23:
[Crashing_at_PPP_connection_stack_overflow.txt](https://github.com/espressif/esp-idf/files/5428849/Crashing_at_PPP_connection_stack_overflow.txt)

kannannair084, 2020-10-24:
I (130893156) pppos_example: GOT ip event!!!
I (130894156) pppos_example: MQTT other event id: 7
pppos_netif_output[0]: proto=0x21, len = 44
pppos_netif_output[0]: proto=0x21, len = 44
I (130898406) esp-modem: 0x3ffbd934   7e 21 45 00 00 2c 00 00  40 00 11 06 60 af 89 87  |~!E..,..@...`...|
I (130898406) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  e3 52 93 b5 4a 99 00 3a  |S...!..[.R..J..:|
I (130898416) esp-modem: 0x3ffbd954   ef 1a 60 12 fa f0 dc 30  00 00 02 04 05 3a 50 e0  |..`....0.....:P.|
I (130898426) esp-modem: 0x3ffbd964   7e                                                |~|
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
pppos_netif_output[0]: proto=0x21, len = 40
pppos_netif_output[0]: proto=0x21, len = 66
I (130898486) esp-modem: 0x3ffbd934   7e 21 45 00 00 2c 00 00  40 00 11 06 60 af 89 87  |~!E..,..@...`...|
I (130898486) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  e3 52 93 b5 4a 99 00 3a  |S...!..[.R..J..:|
I (130898496) esp-modem: 0x3ffbd954   ef 1a 60 12 fa f0 dc 30  00 00 02 04 05 3a 50 e0  |..`....0.....:P.|
I (130898506) esp-modem: 0x3ffbd964   7e                                                |~|
pppos_input[0]: got 49 bytes
ppp_input[0]: ip in pbuf len=44
pppos_netif_output[0]: proto=0x21, len = 40
pppos_netif_output[0]: proto=0x21, len = 66
I (130902746) esp-modem: 0x3ffbd934   7e 21 45 00 00 28 29 d5                           |~!E..().|
pppos_input[0]: got 8 bytes
I (130902796) esp-modem: 0x3ffbd934   40 00 11 06 36 de 89 87  53 d9 0a aa 21 13 07 5b  |@...6...S...!..[|
I (130902796) esp-modem: 0x3ffbd944   e3 52 93 b5 4a 9a 00 3a  ef 34 50 10 fa d6 f3 73  |.R..J..:.4P....s|
I (130902806) esp-modem: 0x3ffbd954   00 00 e1 63 7e                                    |...c~|
pppos_input[0]: got 37 bytes
ppp_input[0]: ip in pbuf len=40
I (130902966) esp-modem: 0x3ffbd934   7e 21 45 00 00 2c 29 d6  40 00 11 06 36 d9 89 87  |~!E..,).@...6...|
I (130902966) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  e3 52 93 b5 4a 9a 00 3a  |S...!..[.R..J..:|
I (130902976) esp-modem: 0x3ffbd954   ef 34 50 18 fa d6 d3 65  00 00 20 02 00 00 07 1e  |.4P....e.. .....|
I (130902986) esp-modem: 0x3ffbd964   7e 7e 21 45 00 00 28 29  d8 40 00 11 06 36 db 89  |~~!E..().@...6..|
I (130902996) esp-modem: 0x3ffbd974   87 53 d9 0a aa 21 13 07  5b e3 52 93 b5 4a 9e 00  |.S...!..[.R..J..|
I (130903006) esp-modem: 0x3ffbd984   3a ef 34 50 10 fa d6 f3  6f 00 00 b0 00 7e 7e 21  |:.4P....o....~~!|
I (130903016) esp-modem: 0x3ffbd994   45 00 00 2c 29 d7 40 00  11 06 36 d8 89 87 53 d9  |E..,).@...6...S.|
I (130903026) esp-modem: 0x3ffbd9a4   0a aa 21 13 07 5b e3 52  93 b5 4a 9a 00 3a ef 34  |..!..[.R..J..:.4|
I (130903036) esp-modem: 0x3ffbd9b4   50 18 fa d6 d3 65 00 00  20 02 00 00 8a 14 7e     |P....e.. .....~|
pppos_input[0]: got 143 bytes
ppp_input[0]: ip in pbuf len=44
ppp_input[0]: ip in pbuf len=40
ppp_input[0]: ip in pbuf len=44
pppos_netif_output[0]: proto=0x21, len = 40
I (130903056) pppos_example: MQTT_EVENT_CONNECTED
pppos_netif_output[0]: proto=0x21, len = 63
I (130903066) pppos_example: sent subscribe successful, msg_id=21850
I (130903846) esp-modem: 0x3ffbd934   7e 21 45 00 00 28 29 d9  40 00 11 06 36 da 89 87  |~!E..().@...6...|
I (130903846) esp-modem: 0x3ffbd944   53 d9 0a aa 21 13 07 5b  e3 52 93 b5 4a 9e 00 3a  |S...!..[.R..J..:|
I (130903856) esp-modem: 0x3ffbd954   ef 4b 50 10 fa bf f3 6f  00 00 89 7c 7e 7e 21 45  |.KP....o...|~~!E|
I (130903866) esp-modem: 0x3ffbd964   00 00 2d 29 da 40 00 11  06 36 d4 89 87 53 d9 0a  |..-).@...6...S..|
I (130903876) esp-modem: 0x3ffbd974   aa 21 13 07 5b e3 52 93  b5 4a 9e 00 3a ef 4b 50  |.!..[.R..J..:.KP|
I (130903886) esp-modem: 0x3ffbd984   18 fa bf 0e 05 00 00 90  03 55 5a 00 a6 d6 7e     |.........UZ...~|
pppos_input[0]: got 95 bytes
ppp_input[0]: ip in pbuf len=40
ppp_input[0]: ip in pbuf len=45
pppos_netif_output[0]: proto=0x21, len = 40
I (130903906) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=21850
pppos_netif_output[0]: proto=0x21, len = 71
I (130903916) pppos_example: sent publish successful, msg_id=0
pppos_netif_output[0]: proto=0x21, len = 71
I (130909786) esp-modem: 0x3ffbd934   7e 21 45 00 00                                    |~!E..|
pppos_input[0]: got 5 bytes
I (130909836) esp-modem: 0x3ffbd934   47 29 db 40 00 11 06 36  b9 89 87 53 d9 0a aa 21  |G).@...6...S...!|
I (130909836) esp-modem: 0x3ffbd944   13 07 5b e3 52 93 b5 4a  a3 00 3a ef 6a 50 18 fa  |..[.R..J..:.jP..|
I (130909846) esp-modem: 0x3ffbd954   a0 ae f0 00 00 30 1d 00  10 2f 74 6f 70 69 63 2f  |.....0.../topic/|
I (130909856) esp-modem: 0x3ffbd964   65 73 70 2d 70 70 70 6f  73 65 73 70 33 32 2d 70  |esp-ppposesp32-p|
I (130909866) esp-modem: 0x3ffbd974   70 70 6f 73 97 ec 7e 7e  21 45 00 00 28 29 dd 40  |ppos..~~!E..().@|
I (130909876) esp-modem: 0x3ffbd984   00 11 06 36 d6 89 87 53  d9 0a aa 21 13 07 5b e3  |...6...S...!..[.|
I (130909886) esp-modem: 0x3ffbd994   52 93 b5 4a c2 00 3a ef  6a 50 10 fa a0 f3 4b 00  |R..J..:.jP....K.|
I (130909896) esp-modem: 0x3ffbd9a4   00 1a 43 7e 7e 21 45 00  00 47 29 dc 40 00 11 06  |..C~~!E..G).@...|
I (130909906) esp-modem: 0x3ffbd9b4   36 b8 89 87 53 d9 0a aa  21 13 07 5b e3 52 93 b5  |6...S...!..[.R..|
I (130909916) esp-modem: 0x3ffbd9c4   4a a3 00 3a ef 6a 50 18  fa a0 ae f0 00 00 30 1d  |J..:.jP.......0.|
[... 54 lines cut ...]
0x40083ec0: _esp_error_check_failed at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_common/src/esp_err.c:45

file: "../main/pppos_client_main.c" line 343
func: app_main
[... cut ...]
0x40083ec3: _esp_error_check_failed at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_common/src/esp_err.c:46

0x400d6fc4: app_main at c:\olf\development\firmware\ppos\pppos_client\build/../main/pppos_client_main.c:343 (discriminator 1)

0x400d539e: main_task at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:600 (discriminator 2)

0x40085709: vPortTaskWrapper at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:143

ELF file SHA256: 55179065fefdb290

david-cermak, 2020-10-26:
Thanks for testing this, it looks like this works for switching modes, but has multiple other issues in `power_down()` etc. No need to post any other crashes, just wanted to understand your issue. Now I think it is clear, and we have to:
* synchronize closing the PPP netif with switching modes
* include 6d70b7c352d8cf0340ac14e127372a7a83f4779a (which wasn't present on your branch AFAIK)

Added the two fixes into the following patch
[0001-esp_moded-Fix-PPP-reinit-on-v4.1.patch.txt](https://github.com/espressif/esp-idf/files/5438389/0001-esp_moded-Fix-PPP-reinit-on-v4.1.patch.txt)

Could you please test it and let us know if it works in your setup? This is the patch against f0e87c933ac2ee47640b452dacb7d10343b51eb4 version (i believe that's the version of IDF you've indicated above)

kannannair084, 2020-10-27:
hi David,

my idf was 4.0.1 and since that didnt use the netif and didn't had modem_stop.. so i upgraded the IDF to v4.2-beta1-227-gf0e87c933-dirty ( 4.2 stable release) and  fix you suggested started working on 4.2 

however, there are still issues as indicated in my crash report above, based on this, could you recommend what i need to change?

i have added the 6d70b7c by changing timeout to 5000 and adding two lines
res = uart_set_rx_timeout(esp_dte->uart_port, 1);
MODEM_CHECK(res == ESP_OK, "set rx timeout failed", err_uart_config);

but i don't know how to do this:
- synchronize closing the PPP netif with switching modes. 

considering my idf version is  IDF to v4.2-beta1-227-gf0e87c933-dirty , the 0001-esp_moded-Fix-PPP-reinit-on-v4.1.patch.txt is it still applicable?

david-cermak, 2020-10-27:
Hi @kannannair084 

Thanks for explaining on the versions. Okay, so the `v4.2-beta1` it is, sorry I somehow missed it and posted the wrong `v4.1` patch.

Here's the correct one, based on this commit f0e87c933:
[0001-esp_moded-Fix-PPP-reinit-on-v4.2-beta1.txt](https://github.com/espressif/esp-idf/files/5443538/0001-esp_moded-Fix-PPP-reinit-on-v4.2-beta1.txt)

(Note: this already includes the https://github.com/espressif/esp-idf/commit/6d70b7c352d8cf0340ac14e127372a7a83f4779a, here's how you can apply it:
```
$ git checkout v4.2-beta1-227-gf0e87c933
$ git checkout -b test/ppp_reinit_fix
$ git am < 0001-esp_moded-Fix-PPP-reinit-on-v4.2-beta1.txt
```

kannannair084, 2020-10-28:
Thank you David for explaining, i didnt use git to get the IDF, i think i downloaded the zip and extracted it.. however, i will follow the change list as per 0001-esp_moded-Fix-PPP-reinit-on-v4.2-beta1.txt  and will run it tonight and get back to you.

kannannair084, 2020-10-28:
i am facing another issue, that is, when i enable PPP in  port_num = UART_NUM_1, my UART_NUM_2 is getting affected..

to show an example,
when PPP is not initialized and connected.
[image]
above is intended byte[] to send and actual sent is matching
[image]

however, once the PPP is intialized, Although i am sending the same Message
it only sends it partially, and the remaining is sent in the next sent cycle.
[image]

[image]

kannannair084, 2020-10-28:
this is one such sending cycle 
[image]

kannannair084, 2020-10-28:
[image]

It looks like the data is getting repositioned...

[image]

void UARTSendCMD_PAMgr(const uint8_t* data, uint32_t len)
{
	gpio_set_level(PA_RX_TX_SWITCH_PIN, 1);
	memset(ui8pPADeviceReadBuffer, '\0', PA_RXBUF_SIZE);
	uart_write_bytes(UART_NUM_2, (char*)data, len);
	//Delay(14);
	ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_2, (100 / portTICK_PERIOD_MS))); // original was 100 ticks
	gpio_set_level(PA_RX_TX_SWITCH_PIN, 0);
}

This is my data sending function, which is working fine, untill i init dte

dte = esp_modem_dte_init(&config);

where 
#define ESP_MODEM_DTE_DEFAULT_CONFIG()          \
    {                                           \
        .port_num = UART_NUM_1,                 \
        .data_bits = UART_DATA_8_BITS,          \
        .stop_bits = UART_STOP_BITS_1,          \
        .parity = UART_PARITY_DISABLE,          \
        .baud_rate = 115200,                    \
        .flow_control = MODEM_FLOW_CONTROL_NONE,\
        .tx_io_num = 33,                        \
        .rx_io_num = 25,                        \
        .rts_io_num = 14,                       \
        .cts_io_num = 26,                       \
        .rx_buffer_size = 2048,                 \
        .tx_buffer_size = 1024,                  \
        .pattern_queue_size = 30,               \
        .event_queue_size = 35,                 \
        .event_task_stack_size = 4096,          \
        .event_task_priority = 5,               \
        .line_buffer_size = 1024                 \
    }

kannannair084, 2020-10-28:
Also, i noticed that, issue starts to appear after this step in esp_modem.c  in the function esp_modem_dte_init(), i tried commenting out section by section to identify where it starts. 
```cpp
/*    MODEM_CHECK(uart_param_config(esp_dte->uart_port, &uart_config) == ESP_OK, "config uart parameter failed", err_uart_config);
    if (config->flow_control == MODEM_FLOW_CONTROL_HW) {
        res = uart_set_pin(esp_dte->uart_port, config->tx_io_num, config->rx_io_num,
                           config->rts_io_num, config->cts_io_num);
    } else {
        res = uart_set_pin(esp_dte->uart_port, config->tx_io_num, config->rx_io_num,
                           UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    }
    MODEM_CHECK(res == ESP_OK, "config uart gpio failed", err_uart_config); */
    /* Set flow control threshold */

kannannair084, 2020-10-28:
@david-cermak  i need to fix the issues and deliver a working solution by end of this week.. i could stay back late as i understand our time differences.. i am blocked by this issue.

It would be great if you could take a look and suggest a way to fix this.. i could stick around.. please keep an eye on this thread..

kannannair084, 2020-10-28:
@david-cermak could you respond when you come online and see my msg. i am here so that this issue could be resolved at the earliest. sorry for pushing this so hard, like i said, i am blocked by this issue to meet my end of the week delivery timeline.

kannannair084, 2020-10-29:
i got this issue, UART 2 getting corrupted, resolved by commenting this line in uart.c
```cpp
  // uart_hal_txfifo_rst(&(uart_context[uart_num].hal));

kannannair084, 2020-10-29:
i am having an issue now, with this ESP_IDF change, when i connect to ble, it is crashing and after the reboot it says issue in modem.c

I (6083) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (6083) NewEsp: REGISTER_APP_EVT, status 0, app_id 1

I (6093) NewEsp: REGISTER_APP_EVT, status 0, app_id 2

I (6093) NewEsp: REGISTER_APP_EVT, status 0, app_id 3

I (6103) NewEsp: BluetoothLE has been started!
I (6123) NewEsp: SERVICE_START_EVT, status 0, service_handle 40

I (6123) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 46

I (6123) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 62

I (6133) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 78

I (6133) NewEsp: REGISTER_APP_EVT, status 0, app_id 4

I (6143) NewEsp: REGISTER_APP_EVT, status 0, app_id 5

I (6153) NewEsp: SERVICE_START_EVT, status 0, service_handle 46

I (6153) NewEsp: SERVICE_START_EVT, status 0, service_handle 62

I (6163) NewEsp: SERVICE_START_EVT, status 0, service_handle 78

I (6163) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 98

I (6173) NewEsp: CREATE_SERVICE_EVT, status 0,  service_handle 102

I (6183) NewEsp: SERVICE_START_EVT, status 0, service_handle 98

I (6193) NewEsp: SERVICE_START_EVT, status 0, service_handle 102

I (6733) NewEsp: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
I (6733) NewEsp: CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
E (6733) esp-modem: esp_dte_handle_line(103): DTE has not yet bind with DCE
I (6733) NewEsp: CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
I (6743) NewEsp: CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
I (6753) NewEsp: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
I (6763) NewEsp: CONNECT_EVT, conn_id 0, remote 78:59:f7:4f:cc:ab:
W (6773) BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.
E (6793) esp-modem: esp_dte_handle_line(103): DTE has not yet bind with DCE
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6036
load:0x40078000,len:13372
load:0x40080400,len:3820
0x40080400: _init at ??:?

entry 0x40080658
I (58) boot: ESP-IDF v4.2-beta1-227-gf0e87c933-dirty 2nd stage bootloader
I (58) boot: compile time 11:25:57
I (60) boot: chip revision: 1
I (63) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (71) boot.esp32: SPI Speed      : 40MHz
I (75) boot.esp32: SPI Mode       : DIO
I (80) boot.esp32: SPI Flash Size : 16MB
W (84) boot.esp32: PRO CPU has been reset by WDT.
W (90) boot.esp32: WDT reset info: PRO CPU PC=0x40092e9f
0x40092e9f: xthal_save_extra_nw at /Users/igrokhotkov/e/esp32/hal/hal/state_asm.S:72

[... 287 lines cut ...]
.coredump.tasks.data 0x3ffb9fa0 0x15c RW
.coredump.tasks.data 0x3ffb9df0 0x1a8 RW
.coredump.tasks.data 0x3ffb5ad8 0x15c RW
.coredump.tasks.data 0x3ffd2750 0x20c RW
.coredump.tasks.data 0x3ffb76ac 0x15c RW
.coredump.tasks.data 0x3ffb74e0 0x1c4 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
I (3248) esp_core_dump_uart: Core dump has been written to uart.
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6036
load:0x40078000,len:13372
load:0x40080400,len:3820
0x40080400: _init at ??:?

entry 0x40080658

kannannair084, 2020-10-29:
issue is happening after calling this line..

```cpp
ESP_ERROR_CHECK(esp_netif_init());

david-cermak, 2020-11-02:
Hi @kannannair084 

If I understand correctly, now the modem/ppp work okay until you get a BLE crash, consequent reboot and yet another crash from the modem then? If so, it seems like the modem is in the data mode sending packets to the ESP32 which hasn't yet initialized the data mode, thus failing.

david-cermak, 2020-11-03:
@kannannair084 The crash you're experiencing looks similar to one in another issue. Could you please try if applying the patch from [this comment](https://github.com/espressif/esp-idf/issues/6024#issuecomment-721365171) helps?

kannannair084, 2020-11-03:
@david-cermak thank you for getting back, sure, will try this and let you know how it goes.

kannannair084, 2020-11-09:
hi David, i have made the modification as suggested and no longer see issues happening... you may close this thread.

closed 2020-11-09 by kannannair084