espressif/esp-idf #6006: Inconsistent TLS negotiation failure

jgcjoec, 2020-10-20:
#5638 3 Environment
- Development Kit: HiLetgo ESP32 from Amazon
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.0.1 and v4.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I have an ESP-IDF application that has the following major components:
1. http_server for provisioning and to view sensor readings
2. Wi-Fi connection for provisioning and to transmit sensor readings
3. A task that reads various sensors attached to the ESP32

This application issues the following HTTP POST requests to an AWS-hosted application. The application is hosted using AWS ECS, with an application load balancer in front of ECS. SSL is terminated at the load balancer
1. Provisioning
2. Heartbeat
3. Alerting

For simplicity, let's just look at provisioning (all HTTP requests seem to be equally affected by this problem)

Sample provisioning request (NOTE: JSON has been purposefully obfuscated)
POST /api/Tenant/Register HTTP/1.1
User-Agent: kIoT
Host: www.jdiflowbot.com
Content-Type: application/json
Content-Length: 230

{"FirstName":"somename","LastName":"somename","PhoneNumber":"somenumber","EmailAddress":"someone@somewhere.com","NickName":"test","DeviceAddress":"1122334455","DeviceName":"Test","ModelName":"Test"}

Sometimes, these HTTP requests will succeed, but more often than not, they appear to fail during the TLS negotiation phase.

Things I know:
1. These HTTP requests ALWAYS succeed when issued from a PC say, using Postman
2. ECDHE-RSA-AES128-GCM-SHA256 is negotiated when HTTP requests that are successful are issued from ESP-IDF
3. ECDHE-RSA-AES128-GCM-SHA256 is also negotiated when HTTP requests are issued from my PC
4. The error is not always the same. Sometimes there is no clear error message. Error messages that I have seen include "mbedtls_ssl_handshake returned -0x50", "esp-tls: Failed to open new connection in specified timeout" and probably some others I am forgetting

Things I've tried:
1. Unicore more... no effect
2. Adjust clock speed... no effect
3. Monitoring dynamic memory... always seems to be > 140KB free
4. Disabling hardware acceleration in mbedTLS... no effect
5. Disabling unused ciphers in mbedTLS... this seemed to reduce the frequency of HTTP request failures
6. Upgrading from ESP-IDF 4.0.1 to 4.1... no effect. I did not try anything newer than 4.1

### Expected Behavior
The HTTP requests should always succeed from ESP-IDF if they succeed from a PC on the same network

### Actual Behavior
The HTTP requests from ESP-IDF do not always succeed

### Steps to reproduce
I am hopeful that if someone sees my code snippet and the logs, someone more familiar with mbedTLS than I am can point me in the right direction

### Code to reproduce this issue
Portions not relevent to HTTP requests have been omitted for simplicity. I also did not include the code for the http handler, but if necessary, I can provide that. It is fairly close to the sample from Espressif, though. This code is obviously designed to retry any failed HTTP request (up to 20 times currently)
```cpp
http_response_t send_http_post_request(const char* url, char* json)
{
	while (!network_is_ready)
	{
		ESP_LOGD(TAG, "Waiting for network connection...");
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
	ESP_LOGD(TAG, "Network is ready.");
[... 1957 lines cut ...]

V (82900) mbedtls: ssl_tls.c:3426 00f0:  3f 71 2d b3 a6 9b b0 f6 ac ff 9c 87 29 cb c7 54  ?q-.........)..T

V (82910) mbedtls: ssl_tls.c:3426 0100:  61 1c 48                                         a.H

I (82920) mbedtls: ssl_tls.c:2755 => flush output

I (82920) mbedtls: ssl_tls.c:2774 message length: 259, out_left: 259

I (82930) mbedtls: ssl_tls.c:2779 ssl->f_send() returned 259 (-0xfffffefd)

I (82940) mbedtls: ssl_tls.c:2807 <= flush output

I (82940) mbedtls: ssl_tls.c:3476 <= write record

I (82950) mbedtls: ssl_tls.c:8710 <= write

I (87950) FlowBot: Current free heap: 175816, Minimum heap: 143264
E (87950) FlowBot: HTTP POST to https://www.jdiflowbot.com/api/Tenant/Register FAILED with error ESP_ERR_HTTP_FETCH_HEADER. 230 bytes
JSON: (REDACTED... see sample)
I (87980) FlowBot: Current free heap: 175816, Minimum heap: 143264
E (87980) FlowBot: Retrying HTTP POST to https://www.jdiflowbot.com/api/Tenant/Register... 2 of 20 times
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5412247/sdkconfig.txt)

Alvin1Zhang, 2020-10-21:
Thanks for the very detailed report and letting us know, we will look into.

jgcjoec, 2020-10-21:
Thank you @Alvin1Zhang!

jgcjoec, 2020-10-21:
In case it is helpful, attached is a network trace (in PCAP format... change .txt to .pcap). It contains a mixture of many HTTP request failures, co-mingled with I believe 2 successes (I wrote a for() to re-do the same HTTP request 100 times and took a 120 second capture of that traffic) I see many TCP RST and retransmissions within, but not sure why. Minimum heap space throughout was 108660. Free heap was typically between 150k and 180k throughout

[_tmp_network-tests_9f297284-e711-4607-8121-bf8e8c62fdd3.txt](https://github.com/espressif/esp-idf/files/5416233/_tmp_network-tests_9f297284-e711-4607-8121-bf8e8c62fdd3.txt)

jgcjoec, 2020-10-21:
I did some more experimentation today, trying to narrow down the cause and noticed something interesting. After the device is initially programmed, it spins up a Wi-Fi AP for provisioning and uses a captive portal to do so (using a highly modified version of https://github.com/tonyp7/esp32-wifi-manager). It works as follows:
1. Device spins up internal AP and captive portal hosted internally
2. User connects to AP and supplies registration information, including connection details for an in-range Wi-Fi network
3. Device connects to in-range Wi-Fi network and issues the registration HTTP request that is the subject of this ticket... that's when the inconsistencies start.

...HOWEVER...

I noticed that if I reboot the device, and issue the same HTTP requests after rebooting (after obviously connecting to the in-range Wi-Fi network), the HTTP requests NEVER fail - I tried hundreds of requests and they all succeeded. The only difference I can think of here is that after rebooting, the device never spins up the internal AP - it goes right into STA mode (since it already knows which Wi-Fi network it needs to connect to).

Could there be some edge case that only presents itself when the device goes from AP -> STA but not when it jumps right into STA mode?

negativekelvin, 2020-10-22:
Possibly the additional overhead of operating as STA+AP creates a timing issue and the server enforces a timeout

jgcjoec, 2020-10-22:
If that was the case, wouldn't increasing the clock speed to 240 MHz improve things? It seems to occur equally as often at 80 MHz and 240 MHz.

AdityaHPatwardhan, 2020-10-22:
Hi @jgcjoec , As you are using a custom wifi-provisioning manager it is difficult to tell weather it properly closes the AP after provisiong. Can you please use the recommended wifi provisioning manager and see if the issue is still present ? The link is https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr.

jgcjoec, 2020-10-22:
@AdityaHPatwardhan - the AP needs to remain active during and after provisioning completes to report success to a web-based UI.

jgcjoec, 2020-10-22:
I did some more experimenting and when I add this single line of code (disabling the Soft AP) to my callback function for event SYSTEM_EVENT_STA_GOT_IP, the HTTP requests succeed.

esp_wifi_set_mode(WIFI_MODE_STA);

negativekelvin, 2020-10-22:
On the softap side are you also using a tls connection? Is it websocket or polled?

jgcjoec, 2020-10-22:
My web UI polls an API hosted by esp_http_server running on the AP. After credentials for the STA are supplied, the STA is started, and used to make an HTTPS (TLS 1.2) request to an AWS-hosted endpoint. 

I tried looking at the wifi_prov_mgr example as suggested, but it appears to have some sort built-in protocomm HTTP server that I unfortunately don't have the time to learn right now. I tried to bypass it by calling wifi_prov_scheme_softap_set_httpd_handle(), but I can't start my HTTP server until after the Wi-Fi AP starts (I get an "Invalid mbox" assertion in tcpip.c), but the documentation states I must call wifi_prov_scheme_softap_set_httpd_handle() before I call wifi_prov_mgr_start_provisioning(), which is before the Wi-Fi AP starts... seems like a catch 22. I saw no examples that used wifi_prov_scheme_softap_set_httpd_handle().

jgcjoec, 2020-10-22:
Ok, I think I figured out the problem....

Keep in mind my device works as follows:
1. Spin up Wi-Fi AP
2. User connects to Wi-Fi AP
3. User is redirected to a captive portal where they enter Wi-Fi credentials (among other things)
4. Device leaves Wi-Fi AP running, attempts to connect to the user selected AP (AP + STA mode)
5. Device sends HTTP request to an AWS-hosted server
6. Device reports success/failure back to the user via the captive portal.
7. Device goes to sleep or reboots (depends on configuration)

To re-cap, things were falling apart on step #5... those HTTP requests were sometimes succeeding, sometimes failing (failing more often than succeeding.

I observed that the HTTP requests ALWAYS succeed after step #7 (reboot) when the device is in STA mode only
I observed that the HTTP requests ALWAYS succeed at step #5 if the AP is explicitly turned off (i.e. putting the device in STA only mode)

My device is battery powered so naturally I am taking steps to conserve battery life - including modem sleep.

After commenting out the following line, HTTP requests succeed regardless of when they are issued:
ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MAX_MODEM));

I was able to reproduce this behavior in a stand-alone application based on the "softAP" and "station" examples under "getting_started".

I realize that power savings isn't free, but this seems like a bug to me... if the community agrees, I can post the sample code I wrote to reproduce this issue.

mahavirj, 2020-10-23:
@jgcjoec 

> I was able to reproduce this behavior in a stand-alone application based on the "softAP" and "station" examples under "getting_started".

Do you mean in either of examples (i.e. `AP` mode or `STA` mode) with enabling `esp_wifi_set_ps(WIFI_PS_MAX_MODEM)` you see problem with data transfer? Or you mean combining these examples together to create `AP + STA` mode behavior?

Anyhow, I think sample code can help us understand this better. 

Few additional points:
  - Per earlier comments, for provisioning (i.e. configuring WiFi credentials) we recommend https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html. For `SoftAP` mode, framework ensure that momentarily WiFi mode stays as `AP + STA` to attempt to connect to WiFi network and convey that status back to provisioning entity (e.g. smart phone) back. Once status is conveyed, it switches mode to pure `STA` and stays that way even after reboot (since credentials are saved in flash storage). Recommend to read more about this, but if you would like stay with your current approach then that is fine as well.

 - Power save is not supported in `AP` mode as of yet, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#ap-sleep. So if you have battery operated use-case, it is recommended to keep `AP` mod enabled for short duration only.

CC @sagb2015 @liuzfesp

jgcjoec, 2020-10-23:
@mahavirj - I combined the "softAP" and "station" examples into one, added my HTTP posting code and was able to reproduce the problem at will by simply calling esp_wifi_set_ps(WIFI_PS_MAX_MODEM). I noticed that with WIFI_PS_MAX_MODEM enabled, AP power consumption is unchanged (which is what I expected from reading the docs), but STA power consumption is drastically reduced - which is perfect for a battery powered device.

Regarding the provisioning examples, all of the examples I've looked at in the latest ESP-IDF release seem to focus on using a mobile app for provisioning and unfortunately, that won't work for my use-case. I need to be able to provision from a captive portal.

Here is the code that I wrote to be able to reproduce this issue. To reproduce:
1. Set EXAMPLE_STA_ESP_WIFI_SSID and EXAMPLE_STA_ESP_WIFI_PASS to the SSID and password (respectively) for your Wi-Fi network
2. Compile and flash (sdkconfig.txt attached) using ESP-IDF 4.1. Other versions may exhibit similar behavior, but I only tested this example on ESP-IDF 4.1
3. Connect to the "SoftAP" AP from your PC. Password is 12345678
4. Observe occasional HTTP request failures. Failures will be in RED, successes will be in GREEN. Just eye-balling it, appears that roughly 50% of them fail for me.
5. Comment out line 192: esp_wifi_set_ps(WIFI_PS_MAX_MODEM);
6. Repeat steps 2 and 3
7. Observe the HTTP request failures no longer occur, or occur FAR less frequently

```cpp
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_http_client.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_STA_ESP_WIFI_SSID      	"<YOUR WIFI SSID>"
#define EXAMPLE_STA_ESP_WIFI_PASS      	"<YOUR WIFI PASSWORD>"
#define EXAMPLE_STA_ESP_MAXIMUM_RETRY  	5
#define EXAMPLE_AP_ESP_WIFI_SSID      	"SoftAP"
#define EXAMPLE_AP_ESP_WIFI_PASS      	"12345678"
#define EXAMPLE_AP_ESP_WIFI_CHANNEL   	1
#define EXAMPLE_AP_MAX_STA_CONN      	4
#define URL								"https://httpbin.org/post"
#define JSON							"{\"FirstName\":\"foo\",\"LastName\":\"bar\",\"PhoneNumber\":\"1112223333\",\"EmailAddress\":\"someone@somewhere.com\",\"NickName\":\"test\",\"DeviceAddress\":\"25:2f:2b:e6:1a:51\",\"DeviceName\":\"Temperature Sensor\",\"ModelName\":\"Temperature Sensor V1\"}"

#define MAX_RETRY_COUNT					20

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char* TAG = "wifi station";

static int s_retry_num = 0;

static esp_http_client_handle_t http_client;

esp_err_t _http_event_handler(esp_http_client_event_t* evt)
{
[... 211 lines cut ...]
            EXAMPLE_STA_ESP_WIFI_SSID, EXAMPLE_STA_ESP_WIFI_PASS);
    }
    else
    {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_ap();
}
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5429458/sdkconfig.txt)

mahavirj, 2020-10-27:
@jgcjoec Thanks for providing sample application, can confirm the issue for `release/v4.1`. I have forwarded this internally to our WiFi team and they shall update here.

jgcjoec, 2020-10-27:
Excellent - thank you!

YouDONG-ESP, 2020-10-30:
Hi @jgcjoec, seems like this issue is related to lwip timeout. Its value is 5 sec, set it to 10 sec is a workaround.

jgcjoec, 2020-10-30:
@YouDONG-ESP - would you mind indicating which specific setting in sdkconfig you're referring to?

YouDONG-ESP, 2020-11-02:
@jgcjoec It's not in sdkconfig, you can add `client->timeout_ms 10000;` at the beginning of  esp_http_client_perform().

jgcjoec, 2020-11-02:
@YouDONG-ESP - thank you, I will give that a try. Just to be clear, that is a work-around for this issue and not intended to be a permanent fix, correct?

YouDONG-ESP, 2020-11-02:
@jgcjoec yeah, just did a packet capture, it shows that SSL handshake will take more than 5 seconds when it's in AP STA coexist mode and  max power save is enabled. Could you check the value of your AP's DTIM? You can do a packet capture and it's in the beacon frame.

jgcjoec, 2020-11-02:
@YouDONG-ESP - I checked my wireless settings (I use Unifi AC Pro access points, if that is relevant) and DTIM is set to the default value of 3.

jgcjoec, 2020-11-02:
@YouDONG-ESP  - I just tested setting timeout_ms to 10000 in the example code above and that certainly did improve things, but I am still seeing failures. If I had to estimate, maybe 10-15% of the fail.

YouDONG-ESP, 2020-11-05:
Hi @jgcjoec, seems like this issue can't be reproduced again, is it the same on your side?

jgcjoec, 2020-11-09:
@YouDONG-ESP - sorry for the late reply. I was out of town.

I just re-ran the scenario above titled "timeout_ms set to the default value and WIFI_PS_MAX_MODEM enabled" and I still experienced a 10% failure rate.

jgcjoec, 2020-11-17:
Please let me know if anyone needs me to do any further testing

YouDONG-ESP, 2020-11-18:
@jgcjoec can you provide me a packet capture for this issue?

jgcjoec, 2020-11-18:
@YouDONG-ESP - please see attached trace where there were 2 failures. Please change the txt extension to pcap.

Thanks,

- Joe

[_tmp_network-tests_ed8c71f9-efbe-4c56-8274-97c645fe99aa.txt](https://github.com/espressif/esp-idf/files/5561497/_tmp_network-tests_ed8c71f9-efbe-4c56-8274-97c645fe99aa.txt)

YouDONG-ESP, 2020-11-19:
@jgcjoec Thanks for your help, we have found the root casuse. Basically it's because our STA received beacon frames that shouldn't be received while sleeping. We are discusting solutions. 
By the way, In AP/STA coex mode, esp_wifi_set_ps() won't safe any power because RF is always open, so why not just turn it off？

jgcjoec, 2020-11-20:
@YouDONG-ESP - thanks, great news! I have a device that is briefly in AP & STA mode, then once provisioned, goes into STA only mode. esp_wifi_set_ps(WIFI_PS_MAX_MODEM) is used to save power when in STA mode only - power savings is fairly significant, too.

negativekelvin, 2020-11-21:
@YouDONG-ESP wifi driver should be smart enough to know if ap is active then don't let sta sleep if there is no advantage.

@jgcjoec i think he means call esp_wifi_set_ps(WIFI_PS_MAX_MODEM) after you switch to sta only mode

jgcjoec, 2020-11-24:
@YouDONG-ESP, @negativekelvin - I could apply the work-around you suggest, but it would require me to modify a 3rd party component that I use for wifi provisioning (https://github.com/tonyp7/esp32-wifi-manager).

closed 2021-01-15 by espressif-bot