espressif/esp-idf #6008: issues with building on current master

chegewara, 2020-10-21:
Hi,
ive been using master commit from august for some time and today i decided to pull most recent master. After pulling it i had to call `./install.sh` because of this error which is annoying but i can live with it (it appears for few months already):
```
chegewara@ubuntu:~/upwork/othernet/as-component$ . ~/esp/esp-idf/export.sh 
Setting IDF_PATH to '/home/chegewara/esp/esp-idf'
Adding ESP-IDF tools to PATH...
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/home/chegewara/esp/esp-idf/install.sh' to install it.
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/home/chegewara/esp/esp-idf/install.sh' to install it.
ERROR: tool xtensa-esp32s3-elf has no installed versions. Please run '/home/chegewara/esp/esp-idf/install.sh' to install it.
```

The main problem is that after installing toolchains i am having this issue:
```
chegewara@ubuntu:~/upwork/othernet/as-component$ idf.py set-target esp32s2
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/chegewara/upwork/othernet/as-component/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s2, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/chegewara/upwork/othernet/as-component/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /home/chegewara/upwork/othernet/as-component"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/chegewara/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/chegewara/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/chegewara/esp/esp-idf/components/nimble does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
-- Check for working C compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++
-- Check for working CXX compiler: /home/chegewara/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file /home/chegewara/upwork/othernet/as-component/sdkconfig
Traceback (most recent call last):
  File "/home/chegewara/esp/esp-idf/tools/kconfig_new/confgen.py", line 27, in <module>
    import kconfiglib
ImportError: bad magic number in 'kconfiglib': b'\x03\xf3\r\n'
CMake Error at /home/chegewara/esp/esp-idf/tools/cmake/kconfig.cmake:228 (message):
  Failed to run confgen.py
  (/home/chegewara/.espressif/python_env/idf4.3_py3.8_env/bin/python;/home/chegewara/esp/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/home/chegewara/esp/esp-idf/Kconfig;--sdkconfig-rename;/home/chegewara/esp/esp-idf/sdkconfig.rename;--config;/home/chegewara/upwork/othernet/as-component/sdkconfig;--env-file;/home/chegewara/upwork/othernet/as-component/build/config.env).
  Error 1
Call Stack (most recent call first):
  /home/chegewara/esp/esp-idf/tools/cmake/build.cmake:442 (__kconfig_generate_config)
  /home/chegewara/esp/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/chegewara/upwork/othernet/as-component/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

Any help with this please, @projectgus 

```
commit 357a2776032299b8bc4044900a8f1d6950d7ce89 (HEAD -> master, origin/master, origin/HEAD)
Merge: b62944b6b 5cdb973ee
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Tue Oct 20 00:43:49 2020 +0800

    Merge branch 'bugfix/codeowners_esp_pm' into 'master'
    
    gitlab: add esp_pm component to CODEOWNERS
    
    See merge request espressif/esp-idf!10844
```

Thanks

[CMakeOutput.log](https://github.com/espressif/esp-idf/files/5413517/CMakeOutput.log)

chegewara, 2020-10-21:
I just realized it may be issue with `kconfiglib`:
`ImportError: bad magic number in 'kconfiglib': b'\x03\xf3\r\n'`

but i dont know what to do to fix it. Any help is appreciated.
I am working on:
ubuntu 20.04

chegewara, 2020-10-21:
I found this post which helped me to solve problem:
https://github.com/Miserlou/Zappa/issues/854#issuecomment-313694177

closed 2020-10-21 by chegewara

projectgus, 2020-10-21:
I'm glad you found a solution, @chegewara .

Unfortunately sometimes changing Python version can cause breakage in this way. It shouldn't happen in newer ESP-IDF versions, we made some changes so the cache files are no longer generated.

We have also added a command `idf.py python-clean` to simplify this clean-up step. There is some more info about it here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#possible-issues

chegewara, 2020-10-21:
Thanks for update. I am using python3 for months, but im guessing esp-idf is migrating to python3 now and it was the issue.

william-ferguson-au, 2020-12-12:
Thanks @projectgus this has been driving me nuts for hours.
Out of curiosity, where are the defective .pyc files located?

projectgus, 2020-12-13:
@william-ferguson-au Sorry for the hassle! In the case of this specific error the file is probably `tools/kconfig_new/kconfiglib.pyc` but it's possible for other files to trigger the same problem.

By default Python 2 builds .pyc bytecode files in the same directory as the .py files, so probably it is in one of the same directories as a Python file or former Python file. If you get an error like `ImportError: bad magic number in 'modulename'` then you can search for `modulename.pyc`, but the `idf.py python-clean` command should take care of them all.