espressif/esp-idf #6009: Trying to get ifkey in "on_ppp_changed" through NETIF_PPP_STATUS event handler causes crash

teekaytk, 2020-10-21:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-472-gcf056a7d0
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description

I am trying to get the ifkey in the on_ppp_changed which seems to receive it from the even as eventdata. 
### Expected Behavior
I expect it to get the netif ifkey
### Actual Behavior
crashes with guru med
### Steps to reproduce

1. step1
Build pppos app with bg96 or sim800 modem
2. 
change on_ppp_changed to

```
static void on_ppp_changed(void *arg, esp_event_base_t event_base,
                           int32_t event_id, void *event_data)
{
    esp_netif_t *netif = event_data;

    const char *ifkey = esp_netif_get_ifkey(netif);
    if(strcmp(ifkey, "PPP_DEF") == 0)
     {
          printf ("%s",ifkey)
     }
}
```

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x40001277  PS      : 0x00060130  A0      : 0x800db6c5  A1      : 0x3ffc1000  
A2      : 0x4008603c  A3      : 0x3f4051e4  A4      : 0x00060120  A5      : 0x3ffc1070  
0x4008603c: prvGetCurMaxSizeByteBuf at /opt/esp/idf/components/esp_ringbuf/ringbuf.c:716

A6      : 0x00000103  A7      : 0x3ffbb434  A8      : 0x80086bd4  A9      : 0x3ffc0fd0  
A10     : 0x00000003  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x00060123  
A14     : 0x00000001  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4008603c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
0x4008603c: prvGetCurMaxSizeByteBuf at /opt/esp/idf/components/esp_ringbuf/ringbuf.c:716

Backtrace:0x40001274:0x3ffc1000 0x400db6c2:0x3ffc1010 0x40166d7d:0x3ffc1030 0x40166822:0x3ffc1060 0x401668fd:0x3ffc10a0 0x400868c1:0x3ffc10c0
0x400db6c2: on_ppp_changed at /project/build/../main/app_pppos.c:145

0x40166d7d: handler_execute at /opt/esp/idf/components/esp_event/esp_event.c:145

0x40166822: esp_event_loop_run at /opt/esp/idf/components/esp_event/esp_event.c:582 (discriminator 3)

0x401668fd: esp_event_loop_run_task at /opt/esp/idf/components/esp_event/esp_event.c:115

0x400868c1: vPortTaskWrapper at /opt/esp/idf/components/freertos/xtensa/port.c:169
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

david-cermak, 2020-10-22:
Hi @teekaytk 

Thank you for reporting this issue. There's a bug in the PPP netif event reporting (the event data contains the actual netif object content instead of it's pointer). Please see the below patch fixing this issue:
[0001-esp-netif-Fix-PPP-netif-event-posting-to-include-esp.patch.txt](https://github.com/espressif/esp-idf/files/5423962/0001-esp-netif-Fix-PPP-netif-event-posting-to-include-esp.patch.txt)

teekaytk, 2020-10-22:
Thanks @david-cermak 
This worked!

Alvin1Zhang, 2020-10-23:
Reopen until the fix is available on GitHub, when the issue will be automatically updated.

closed 2020-11-03 by espressif-bot