espressif/esp-idf #6011: CERTIFICATE_VERIFY_FAILED error on install on MacOSX Python 3.8

mbuckaway, 2020-10-21:
System: Mac OSX 10.15.7
Python: Python 3.8.1 (updated to 3.8.1 from brew) and Python 3.7.2

Following the instructions, I used the $IDF_PATH/install.sh, and I got an ssl error:
```
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "idf_tools.py", line 1488, in main
    action_func(args)
  File "idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError
```

Following the code, I ran the same thing using Python 3.7, and got the same error calling idf_tools.py directly:
```
python3.7 idf_tools.py install
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz to /Users/mark/.espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "idf_tools.py", line 1488, in main
    action_func(args)
  File "idf_tools.py", line 1181, in action_install
    tool_obj.download(tool_version)
  File "idf_tools.py", line 598, in download
    raise DownloadError()
__main__.DownloadError
```

While one could argue the website should have a valid SSL cert, I managed to fix the code to ignore the bad cert. The patch is below:

```
diff --git a/tools/idf_tools.py b/tools/idf_tools.py
index cb88b7223..d54b6ed04 100755
--- a/tools/idf_tools.py
+++ b/tools/idf_tools.py
@@ -54,6 +54,7 @@ import errno
 import shutil
 import functools
 import copy
+import ssl
 from collections import OrderedDict, namedtuple

 try:
@@ -576,6 +577,7 @@ class IDFTool(object):
                 return

         downloaded = False
+        ssl._create_default_https_context = ssl._create_unverified_context
         for retry in range(DOWNLOAD_RETRY_COUNT):
             local_temp_path = local_path + '.tmp'
             info('Downloading {} to {}'.format(archive_name, local_temp_path))
```

The SSL cert should be ignored by default as for this app, we really don't care if it's not valid.

This is a minor issue, but it makes it hard to install the IDF otherwise. I suspect the urlretrieve changed somewhere along the way.

Alvin1Zhang, 2020-10-22:
Thanks for reporting, we will look into.

igrr, 2020-10-22:
@mbuckaway Just to check, have you done the certificate installation step after installing Python? E.g. https://stackoverflow.com/questions/41691327/ssl-sslerror-ssl-certificate-verify-failed-certificate-verify-failed-ssl-c/41692664#41692664.

igrr, 2020-10-22:
Linking https://github.com/espressif/esp-idf/issues/4081 as it contains some other possible causes which can lead to this error.

mbuckaway, 2020-10-22:
No...and had I googled the issue a bit farther, I would have found the problem. Ran the Cert install command, and the installed certs were updated, re=installed ESP-IDF and it worked correctly. Thanks for pointing this out. About the only thing I might suggest is a FAQ item in the docs regarding installation on MacOSX.

aknabi, 2021-01-16:
Fyi Got it to work by running /Applications/Python 3.9/Install Certificates.command... based on (https://stackoverflow.com/questions/41691327/ssl-sslerror-ssl-certificate-verify-failed-certificate-verify-failed-ssl-c/41692664#41692664)

Still this whole toolchain makes me feel that ESP32 is more for people that like to play deal with configurations and tools hassles than actually on embedded projects. I've been screwing around for a month trying to get a setup going with a eep-prog... tried with the Mac, even bought a windows machines... did get a debug session running once, but somehow it went sideways. Maybe someone should make and update a preconfigured VM for a specific board/debug combo for folks that actually want to program the ESP32

mbuckaway, 2021-01-16:
I don't agree with this assessment. Other than the cert issue when moving to a newer Python, the ESP-IDF has been flawless to use and develop code. I use VS Code for an editor, and usually the command line to build. On occasion, use the same setup on a linux laptop. It just works. Support for external components has improved greatly to the point, for me, I can build ESP32 devices in a few hours including using modbus and homekit.

mbuckaway, 2021-01-16:
This issue is closed, btw. Long since resolved.

closed 2021-01-16 by mbuckaway