espressif/esp-idf #6012: Can not change UART1 pins.

utkutpcgl, 2020-10-22:
Note: This might be considered a feature request rather than a bug.
## Environment
- Development Kit:     ESP32-LyraT
- Module or chip used:  ESP32-WROVER
- IDF version 4.1
- Build System:       idf.py
- Compiler version   xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:       USB

## Problem Description
I can not change the dedicated uart tx-rx pins with the following function ( even if the signalling speed is low - below 40 MHz):
    uart_set_pin(UART_NUM_1, T1XD_PIN, R1XD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
T1XD_PIN is 13 
R1XD_PIN is 15
Some other pins did not work either.
### Expected Behavior
The driver should automatically route the signals according to the GPIO matrix. 
### Actual Behavior
However, there is no signal in the MTDO(15), MTCK(13) pins (default JTAG functions in IO_MUX).

### Steps to reproduce
Replace the pin numbers in esp-idf/examples/peripherals/uart/uart_async_rxtxtasks project with the pins I used.

### Code to reproduce this issue
```
/* UART asynchronous example, that uses separate RX and TX tasks

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_log.h"
#include "driver/uart.h"
#include "string.h"
#include "driver/gpio.h"

static const int RX_BUF_SIZE = 1024;

#define T1XD_PIN (13) //35
#define R1XD_PIN (15) //34
#define CTS (31) // WRONG! TODO
#define RTS (36)

void init(void) {
    static const char *INIT = "Init";
    //gpio_set_direction(TXD_PIN, GPIO_MODE_OUTPUT);
    const uart_config_t uart_config = {
        .baud_rate = 9600, //57600 115200
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_APB,
    };
    //ESP_LOGI(INIT,"hey %d ", 1);
    // We won't use a buffer for sending data.
    
    uart_param_config(UART_NUM_1, &uart_config); // UART_NUM_1
    uart_set_pin(UART_NUM_1, T1XD_PIN, R1XD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE); //UART_PIN_NO_CHANGE UART_NUM_1
    uart_driver_install(UART_NUM_1, RX_BUF_SIZE * 2, 0, 0, NULL, 0); //UART_NUM_0, UART_NUM_1
}

int sendData(const char* logName, const char* data)
{
    const int len = strlen(data);
    const int txBytes = uart_write_bytes(UART_NUM_1, data, len); // UART_NUM_1
    ESP_LOGI(logName, "Wrote %d bytes", txBytes);
    //printf("hey");
    return txBytes;
}

static void tx_task(void *arg)
{
    //printf("hey");
    static const char *TX_TASK_TAG = "TX_TASK";
    esp_log_level_set(TX_TASK_TAG, ESP_LOG_INFO);
    while (1) {
        sendData(TX_TASK_TAG, "Hello world");
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
}

static void rx_task(void *arg)
{
    int length = 0;
    static const char *RX_TASK_TAG = "RX_TASK";
    esp_log_level_set(RX_TASK_TAG, ESP_LOG_INFO);
    uint8_t* data = (uint8_t*) malloc(RX_BUF_SIZE+1);
    //printf("heyaaaaa");
    //ESP_LOGI(RX_TASK_TAG,"hey %d ", 1);
    while (1) 
    {
        //ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0, (size_t*)&length));
        //printf("heyaaaaa");
        const int rxBytes = uart_read_bytes(UART_NUM_1, data, RX_BUF_SIZE, 1000 / portTICK_RATE_MS ); // RX_BUF_SIZE 1000 / portTICK_RATE_MS UART_NUM_1
        if (rxBytes > 0) {
            ESP_LOGI(RX_TASK_TAG, "READ len is %d ", length);
            data[rxBytes] = 0;
            ESP_LOGI(RX_TASK_TAG, "Read %d bytes: '%s'", rxBytes, data);
            ESP_LOG_BUFFER_HEXDUMP(RX_TASK_TAG, data, rxBytes, ESP_LOG_INFO);
            //uart_flush(UART_NUM_0);
        }
    }
    free(data);
}

void app_main(void)
{   
    static const char *MAIN = "MAIN";
    ESP_LOGI(MAIN,"hey %d ", UART_NUM_1);
    //printf("hey");

    init();

    xTaskCreate(rx_task, "uart_rx_task", 1024*2, NULL, 3, NULL); //configMAX_PRIORITIES
    //ESP_LOGI(MAIN,"hey %d ", 1);
    xTaskCreate(tx_task, "uart_tx_task", 1024*2, NULL, 2, NULL); //configMAX_PRIORITIES-1
}
```

utkutpcgl, 2020-10-22:
I noticed that GPIO matrix pin muxing is done automatically (also for the method uart_set_pin). I guess my problem was that JTAG pins were inactive by default. To activate the dip switch has to be changed. I used pins 25 and 26 which do not have default pins soldered to them, then it worked. However, it would be nice if there were more warnings about the pin activation in the documentation. Also, a GPIO example could reduce a lot of effort for newbies like me. Thanks anyway.

versamodule, 2020-10-22:
There is and example. 
[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo](url)

utkutpcgl, 2020-10-23:
Maybe it is only me who can not understand that any pin (except some) can be assigned as UART TX and RX from the example you shared @versamodule :D

negativekelvin, 2020-10-23:
This is specific to lyra-t board, not sure it is practical to cover all possible gpio gotchas on various hardware within the examples.  But should be mentioned here at least https://docs.espressif.com/projects/esp-adf/en/latest/get-started/get-started-esp32-lyrat.html

utkutpcgl, 2020-11-25:
I could not find it, but no big issue. Thanks.

closed 2020-11-25 by utkutpcgl