espressif/esp-idf #6013: ppos_client example running in a loop crashes - failed to set working mode

kannannair084, 2020-10-22:
## Environment

- Development Kit:      Internal Prototype
- Module or chip used: |ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):   v4.2-beta1-227-gf0e87c933
- Build System:         [idf.py]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows|
- (Windows only) environment type: [ESP Command Prompt|
- Using an IDE?: [No]
- Power Supply: [external 3.3V|

## Problem Description
i modified the ppos_client example to run in a loop.
Not able to run  pppos_client in a loop of init and de-init to mimic a dropping connection and reconnection.
problem is, failed to send set working mode to command mode and crashed.

### Expected Behavior
To be able to connect and disconnect in a loop.

### Actual Behavior
crashing
### Steps to reproduce

run it

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "mqtt_client.h"
#include "esp_modem.h"
#include "esp_modem_netif.h"
#include "esp_log.h"
#include "sim800.h"
#include "bg96.h"
#include "driver/gpio.h"

#define BROKER_URL "mqtt://mqtt.eclipse.org"

static const char *TAG = "pppos_example";
static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
static const int STOP_BIT = BIT1;
static const int GOT_DATA_BIT = BIT2;

static void modem_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    switch (event_id) {
    case ESP_MODEM_EVENT_PPP_START:
        ESP_LOGI(TAG, "Modem PPP Started");
        break;
    case ESP_MODEM_EVENT_PPP_STOP:
        ESP_LOGI(TAG, "Modem PPP Stopped");
        xEventGroupSetBits(event_group, STOP_BIT);
        break;
    case ESP_MODEM_EVENT_UNKNOWN:
        ESP_LOGW(TAG, "Unknow line received: %s", (char *)event_data);
        break;
    default:
        break;
    }
}

[... 310 lines cut ...]
Backtrace:0x40085013:0x3ffb71f0 0x40085701:0x3ffb7210 0x4008959a:0x3ffb7230 0x40083ec3:0x3ffb72a0 0x400d6fc2:0x3ffb72c0 0x400d5476:0x3ffb73f0 0x40085709:0x3ffb7420
0x40085013: panic_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/panic.c:330

0x40085701: esp_system_abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_system/system_api.c:106

0x4008959a: abort at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/newlib/abort.c:46

0x40083ec3: _esp_error_check_failed at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp_common/src/esp_err.c:46

0x400d6fc2: app_main at c:\olf\development\firmware\ppos\pppos_client\build/../main/pppos_client_main.c:298 (discriminator 1)

0x400d5476: main_task at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/esp32/cpu_start.c:600 (discriminator 2)

0x40085709: vPortTaskWrapper at C:/OLF/Development/Firmware/ESP-IDF/esp-idf_4_2/components/freertos/xtensa/port.c:143
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5422916/sdkconfig.txt)

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
[pppos_client.elf.txt](https://github.com/espressif/esp-idf/files/5422903/pppos_client.elf.txt)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-10-23:
Thanks for reporting and sorry for the trouble incurred, we will look into.

david-cermak, 2020-10-23:
@kannannair084 Thanks again for the report. This is a recognized problem which needs a fix, there seems to be a race condition in deinitializing the network and switching back to the command mode.

As a workaround, as seen you use gpio's to control pwr and reset, I might suggest using these gpio's to power-down the module after ignoring the failure from
```
E (52752) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (52752) bg96: bg96_set_working_mode(298): send command failed
```
by
``` c
        /* Exit PPP mode */
        esp_modem_stop_ppp(dte);
```

Alvin1Zhang, 2020-12-22:
Thanks for reporting and sorry for slow turnaround, we have a fix under internal reviewing now, and the issue will close automatically once the fix is available on GitHub, thanks.

TahirShaik, 2020-12-25:
Is this issue resolved?????

closed 2021-01-05 by espressif-bot